Koenkk/zigbee2mqtt #5159: Adding a device that reports "unknown" model and manufacturer

azzamatazz, 2020-12-02:
Apologies if this is the wrong place to publish this question! 

I bought a temp/humidity sensor off ebay that looks exactly like this device:
https://www.zigbee2mqtt.io/devices/TS0201.html

.. but it is not.

It is displaying unknown/undefined make/model in the logs and Z2MA.

**zigbee2mqtt             | Zigbee2MQTT:debug 2020-11-27 04:02:43: Received Zigbee message from 'sensor_climate_display_dining', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2780}' from endpoint 1 with groupID 0**
zigbee2mqtt             | Zigbee2MQTT:warn  2020-11-27 04:02:43: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt             | Zigbee2MQTT:warn  2020-11-27 04:02:43: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
**zigbee2mqtt             | Zigbee2MQTT:debug 2020-11-27 04:02:44: Received Zigbee message from 'sensor_climate_display_dining', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":7730}' from endpoint 1 with groupID 0**
zigbee2mqtt             | Zigbee2MQTT:warn  2020-11-27 04:02:44: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt             | Zigbee2MQTT:warn  2020-11-27 04:02:44: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html._
**zigbee2mqtt             | Zigbee2MQTT:debug 2020-11-27 04:02:45: Received Zigbee message from 'Received Zigbee message from 'sensor_climate_display_dining', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200,"batteryVoltage":32}' from endpoint 1 with groupID 0**

It seems like it already has the required converters in fromZigbee.js:

    temperature: {
        cluster: 'msTemperatureMeasurement',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const temperature = parseFloat(msg.data['measuredValue']) / 100.0;
            const property = postfixWithEndpointName('temperature', msg, model);
            return {[property]: calibrateAndPrecisionRoundOptions(temperature, options, 'temperature')};
        },
    },

    humidity: {
        cluster: 'msRelativeHumidity',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const humidity = parseFloat(msg.data['measuredValue']) / 100.0;

            // https://github.com/Koenkk/zigbee2mqtt/issues/798
            // Sometimes the sensor publishes non-realistic vales, it should only publish message
            // in the 0 - 100 range, don't produce messages beyond these values.
            if (humidity >= 0 && humidity <= 100) {
                return {humidity: calibrateAndPrecisionRoundOptions(humidity, options, 'humidity')};
            }
        },
    },

    battery: {
        cluster: 'genPowerCfg',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const payload = {};
            if (msg.data.hasOwnProperty('batteryPercentageRemaining')) {
                // Some devices do not comply to the ZCL and report a
                // batteryPercentageRemaining of 100 when the battery is full (should be 200).
                const dontDividePercentage = model.meta && model.meta.battery && model.meta.battery.dontDividePercentage;
                let percentage = msg.data['batteryPercentageRemaining'];
                percentage = dontDividePercentage ? percentage : percentage / 2;
                payload.battery = precisionRound(percentage, 2);
            }

I'm not sure of the way forward to fix this, by either creating an model/vendor "undefined" in devices.js cloned off the existing TuYa device (which seems a bit of a crude hack):

        zigbeeModel: ['TS0201'],
        model: 'TS0201',
        vendor: 'TuYa',
        description: 'Temperature & humidity sensor with display',
        fromZigbee: [fz.battery, fz.temperature, fz.humidity],
        toZigbee: [],
        exposes: [e.battery(), e.temperature(), e.humidity()],
    },

Or is there a way I can deal with it using a Manual Home Assistant configuration? Or is there some configuration override I can do in Z2M to force the device to present as another known existing model in Z2M?

Thanks in advance for any help.

Cheers,
Azza

Koenkk, 2020-12-02:
Can you try to repair the device by first removing from https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove and pair after that?

AzzaHQ, 2020-12-03:
Hi Koen,
Thank you very very much for taking the time to respond!
I tried removing both devices in question and re-pairing them, but they still pair with the same undefined vendor/model attributes.

Koenkk, 2020-12-03:
Can you share the data/database.db entry of this device?

AzzaHQ, 2020-12-03:
This is now coming from my personal email

=====================================

Most certainly.... see cat'd output below.

Please know, I am most honored to have your help in my situation most 
noble sir !!

IDs 12 and 29 (highlighted in bold)....

I imagine if you had a $1 for every time you have trawled through these 
logs you would be a very very very rich man! Haha !!

zfs@zfs-server:~/docker/zigbee2mqtt/data$ cat !$
cat database.db
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b000be8726d","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281],"outClusterList":[1280,1282],"clusters":{},"binds":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null}
{"id":2,"type":"Router","ieeeAddr":"0x00124b0014d9cae2","nwkAddr":63521,"manufId":4447,"manufName":"LUMI","powerSource":"Mains 
(single 
phase)","modelId":"lumi.router","epList":[8],"endpoints":{"8":{"profId":260,"epId":8,"devId":256,"inClusterList":[0,6],"outClusterList":[0,6],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.router","manufacturerName":"LUMI","powerSource":1,"zclVersion":1,"hwVersion":1,"dateCode":"2018-09-26"}},"genOnOff":{"attributes":{"onOff":0}},"genBinaryValue":{"attributes":{"description":"27723/0x00158D00024DEB98","inactiveText":"CHILD_RFD","presentValue":99,"relinquishDefault":2,"minimumOffTime":1}}},"binds":[],"meta":{}}},"hwVersion":1,"dateCode":"2018-09-26","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1607010255822}
{"id":3,"ieeeAddr":"0x00124b001949fe7d","nwkAddr":15414,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1606744159875}
{"id":4,"type":"Router","ieeeAddr":"0x00124b001949fe7d","nwkAddr":25453,"manufId":4447,"manufName":"LUMI","powerSource":"Mains 
(single 
phase)","modelId":"lumi.router","epList":[8],"endpoints":{"8":{"profId":260,"epId":8,"devId":256,"inClusterList":[0,6],"outClusterList":[0,6],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.router","manufacturerName":"LUMI","powerSource":1,"zclVersion":1,"hwVersion":1,"dateCode":"2018-09-26"}},"genOnOff":{"attributes":{"onOff":0}},"genBinaryValue":{"attributes":{"description":"1839/0x00158D00024DE837","inactiveText":"CHILD_RFD","presentValue":46,"relinquishDefault":3,"minimumOffTime":0}}},"binds":[],"meta":{}}},"hwVersion":1,"dateCode":"2018-09-26","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1607010235217}
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d00024df5ef","nwkAddr":43097,"manufId":4151,"manufName":"LUMI","powerSource":"DC 
Source","modelId":"lumi.ctrl_neutral2","epList":[1,2,3,4,5,6,8],"endpoints":{"1":{"profId":260,"epId":1,"devId":6,"inClusterList":[0,3,1,2,25,10],"outClusterList":[0,10,25],"clusters":{"genBasic":{"attributes":{"65281":{"1":3300,"3":33,"5":104,"8":9749,"10":25453,"100":0,"101":0,"110":0,"111":0,"153":0,"155":0},"manufacturerName":"LUMI","powerSource":4,"zclVersion":0,"stackVersion":2,"hwVersion":38,"dateCode":"11-25-2017"}}},"binds":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[16,6,4,5],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"61440":44587264,"onOff":0}}},"binds":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[16,6,4,5],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"61440":61364513,"onOff":0}}},"binds":[],"meta":{}},"4":{"epId":4,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"5":{"epId":5,"inClusterList":[],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"meta":{}},"6":{"epId":6,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"8":{"epId":8,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":38,"dateCode":"11-25-2017","zclVersion":0,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1607010232629}
[... cut ...] ":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":3045},{"elmType":33,"elmVal":424},{"elmType":36,"elmVal":[0,0]},{"elmType":33,"elmVal":22996},{"elmType":32,"elmVal":90}],"modelId":"lumi.sensor_motion","appVersion":11}},"msOccupancySensing":{"attributes":{"occupancy":1}}},"binds":[],"meta":{}}},"appVersion":11,"interviewCompleted":true,"meta":{},"lastSeen":1606896315462}zfs@zfs-server:~/docker/zigbee2mqtt/data$ 

Thank you Koen! I am most honoured by your assistance !

Cheers,

Aaron

Koenkk, 2020-12-03:
Try updating the entry in devices.js of TS0201 to

```js
{
    fingerprint: [
        {type: 'EndDevice', manufacturerID: 4098, endpoints: [
            {ID: 1, inputClusters: [], outputClusters: []},
        ]},
    ],
    zigbeeModel: ['TS0201'],
    model: 'TS0201',
    vendor: 'TuYa',
    description: 'Temperature & humidity sensor with display',
    fromZigbee: [fz.battery, fz.temperature, fz.humidity],
    toZigbee: [],
    exposes: [e.battery(), e.temperature(), e.humidity()],
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

AzzaHQ, 2020-12-04:
You're a bloody champion !!!

It works !!

Thanks Koen !

Passes correctly through to Home Assistant too !

Thanks again Koen.

Cheers,

Aaron

AzzaHQ, 2020-12-04:
You're a bloody champion !!!

It works !!

Thanks Koen !
 
[image]

Passes correctly through to Home Assistant too !
[image]

Thanks again Koen.

Cheers,
Aaron

closed 2020-12-04 by AzzaHQ

uowis, 2023-04-05:
i have device with no model ID and no manufacter name too : 
{"id":32,"type":"Router","ieeeAddr":"0xa4c13809602682ac","nwkAddr":41195,"manufId":4417,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"manuSpecificBosch":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1680703373514,"defaultSendRequestWhen":"immediate"}
how i try to make new external converter ?