Koenkk/zigbee2mqtt #5160: QBKG12LM \ QBKG03LM non-existing "state": "OFF"

VOVAiS, 2020-12-02:
## What happened
I get from switches (0x00158d00024862df) a non-existent topic "state": "OFF"

## What did you expect to happen
Start Zigbee2MQTT
LOG [https://pastebin.com/ksTxhTag](url)

## How to reproduce it (minimal and precise)
 Just wait and this topic will appear

## Debug info 
Zigbee2MQTT version: 1.16.1-dev comit 9ead1a24 
Adapter hardware: CC2531
Adapter firmware version: 20190619

VOVAiS, 2021-01-02:
there is also a non-existent topic in the logs #5443

Koenkk, 2021-01-02:
If instead of `state` you would expect `state_left`/`state_right`. This happens when you control it via `zigbee2mqtt/0x00158d00024862df/set` `{"state": "ON"}` (so no `{"state_left": "ON"}`). When controlling it via `state` it will control the first endpoint (probably left). You can remove it manually from the `data/state.json`.

VOVAiS, 2021-01-02:
no I do not give commands to the topic zigbee2mqtt / 0x00158d00024862df / set

in fresh logs you can look at 0x00158d0002235cb3

> info  2021-01-02 13:37:35: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"2dc9016f","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20201010,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0x00124b00060fd6da","panID":6755},"permit_join":false,"version":"1.17.0-dev"}'
debug 2021-01-02 13:37:46: Received Zigbee message from '0x00158d0002235cb3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 4 with groupID 0
info  2021-01-02 13:37:46: MQTT publish: topic 'zigbee2mqtt/0x00158d0002235cb3', payload '{"click":"left","linkquality":78,"state_left":"ON","state_right":"OFF"}'
debug 2021-01-02 13:37:47: Received MQTT message on 'zigbee2mqtt/0x00158d0002235cb3/left/set/state' with data 'OFF'
debug 2021-01-02 13:37:47: Publishing 'set' 'state' to '0x00158d0002235cb3'
debug 2021-01-02 13:37:47: Received MQTT message on 'zigbee2mqtt/0x00158d0002235cb3/right/set/state' with data 'OFF'
debug 2021-01-02 13:37:47: Publishing 'set' 'state' to '0x00158d0002235cb3'
info  2021-01-02 13:37:47: MQTT publish: topic 'zigbee2mqtt/0x00158d0002235cb3', payload '{"linkquality":94,"state_left":"OFF","state_right":"OFF"}'
debug 2021-01-02 13:37:47: Received Zigbee message from '0x00158d0002235cb3', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440518,"onOff":0}' from endpoint 2 with groupID 0
info  2021-01-02 13:37:47: MQTT publish: topic 'zigbee2mqtt/0x00158d0002235cb3', payload '{"linkquality":97,"state_left":"OFF","state_right":"OFF"}'
info  2021-01-02 13:37:47: MQTT publish: topic 'zigbee2mqtt/0x00158d0002235cb3', payload '{"linkquality":99,"state_left":"OFF","state_right":"OFF"}'
debug 2021-01-02 13:37:47: Received Zigbee message from '0x00158d0002235cb3', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440519,"onOff":0}' from endpoint 3 with groupID 0
info  2021-01-02 13:37:47: MQTT publish: topic 'zigbee2mqtt/0x00158d0002235cb3', payload '{"linkquality":99,"state_left":"OFF","state_right":"OFF"}'
debug 2021-01-02 13:38:04: Received Zigbee message from '0x00158d0002235cb3', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2021-01-02 13:38:08: Received MQTT message on 'zigbee2mqtt/Spal/set' with data 'ON'
debug 2021-01-02 13:38:08: Publishing 'set' 'state' to 'Spal'
info  2021-01-02 13:38:08: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffeff3d06', payload '{"state":"ON"}'
info  2021-01-02 13:38:08: MQTT publish: topic 'zigbee2mqtt/Spal', payload '{"state":"ON"}'
info  2021-01-02 13:38:08: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffeff2e15', payload '{"state":"ON"}'
debug 2021-01-02 13:38:12: Received MQTT message on 'zigbee2mqtt/Spal/set' with data 'OFF'
debug 2021-01-02 13:38:12: Publishing 'set' 'state' to 'Spal'
info  2021-01-02 13:38:12: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffeff3d06', payload '{"state":"OFF"}'
info  2021-01-02 13:38:12: MQTT publish: topic 'zigbee2mqtt/Spal', payload '{"state":"OFF"}'
info  2021-01-02 13:38:12: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffeff2e15', payload '{"state":"OFF"}'
debug 2021-01-02 13:38:13: Received MQTT message on 'zigbee2mqtt/0x00158d0002166ebe/set/state' with data 'ON'
debug 2021-01-02 13:38:13: Publishing 'set' 'state' to '0x00158d0002166ebe'
info  2021-01-02 13:38:13: MQTT publish: topic 'zigbee2mqtt/0x00158d0002166ebe', payload '{"linkquality":57,"state":"ON"}'
debug 2021-01-02 13:38:14: Received Zigbee message from '0x00158d0002166ebe', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440522,"onOff":1}' from endpoint 1 with groupID 0
info  2021-01-02 13:38:14: MQTT publish: topic 'zigbee2mqtt/0x00158d0002166ebe', payload '{"linkquality":57,"state":"ON"}'
debug 2021-01-02 13:38:14: Received Zigbee message from '0x00158d0002166ebe', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2021-01-02 13:38:14: MQTT publish: topic 'zigbee2mqtt/0x00158d0002166ebe', payload '{"linkquality":57,"state":"ON"}'
debug 2021-01-02 13:38:27: Received Zigbee message from '0x00158d0002d7a022', type 'attributeReport', cluster 'genBasic', data '{"65281":{"100":0,"101":0,"149":0.1392333060503006,"152":0,"154":0,"3":32,"5":127,"7":"0x0000000000000000","8":4385,"9":1538}}' from endpoint 1 with groupID 0
info  2021-01-02 13:38:27: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d7a022', payload '{"consumption":0.14,"linkquality":94,"power":0,"temperature":32}'
debug 2021-01-02 13:39:00: Received MQTT message on 'zigbee2mqtt/0x00158d0002235cb3/left/set/state' with data 'ON'
debug 2021-01-02 13:39:00: Publishing 'set' 'state' to '0x00158d0002235cb3'
debug 2021-01-02 13:39:00: Received MQTT message on 'zigbee2mqtt/0x00158d0002235cb3/right/set/state' with data 'ON'
debug 2021-01-02 13:39:00: Publishing 'set' 'state' to '0x00158d0002235cb3'
info  2021-01-02 13:39:01: MQTT publish: topic 'zigbee2mqtt/0x00158d0002235cb3', payload '{"linkquality":81,"state_left":"ON","state_right":"OFF"}'
debug 2021-01-02 13:39:01: Received Zigbee message from '0x00158d0002235cb3', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440523,"onOff":1}' from endpoint 2 with groupID 0
[... cut ...]
info  2021-01-02 13:41:53: MQTT publish: topic 'zigbee2mqtt/0x00158d0002235cb3', payload '{"linkquality":84,"state":"ON","state_left":"OFF","state_right":"OFF","temperature":26}'
debug 2021-01-02 13:41:57: Not saving state
debug 2021-01-02 13:42:11: Received Zigbee message from '0x00158d0002235cb3', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2021-01-02 13:43:13: Received Zigbee message from '0x00158d0002235cb3', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0

Koenkk, 2021-01-02:
Thanks, I see where the issue is coming from now.

Can you provide the same logging (so no `state` at start of z2m and wait until `state` is published) with first the left being OFF and right being ON. And a second log with the left being ON and right being OFF.

VOVAiS, 2021-01-02:
> info  2021-01-02 14:15:34: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"2dc9016f","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20201010,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0x00124b00060fd6da","panID":6755},"permit_join":false,"version":"1.17.0-dev"}'
debug 2021-01-02 14:15:39: Received MQTT message on 'zigbee2mqtt/0x00158d0002235cb3/left/set/state' with data 'ON'
debug 2021-01-02 14:15:39: Publishing 'set' 'state' to '0x00158d0002235cb3'
info  2021-01-02 14:15:39: MQTT publish: topic 'zigbee2mqtt/0x00158d0002235cb3', payload '{"linkquality":78,"state_left":"ON"}'
debug 2021-01-02 14:15:39: Received Zigbee message from '0x00158d0002235cb3', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440515,"onOff":1}' from endpoint 2 with groupID 0
info  2021-01-02 14:15:39: MQTT publish: topic 'zigbee2mqtt/0x00158d0002235cb3', payload '{"linkquality":78,"state_left":"ON"}'
debug 2021-01-02 14:16:17: Received Zigbee message from '0x00158d0002235cb3', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2021-01-02 14:16:48: Received Zigbee message from '0x000d6f0013153342', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":16}' from endpoint 1 with groupID 0
debug 2021-01-02 14:16:48: No converter available for 'HS2WD-E' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":16}'
debug 2021-01-02 14:17:19: Received Zigbee message from '0x00158d0002235cb3', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2021-01-02 14:18:15: Received Zigbee message from '0x00158d0002166ebe', type 'attributeReport', cluster 'genBasic', data '{"65281":{"100":1,"149":12.08336067199707,"152":0,"154":0,"3":31,"5":98,"7":"0x0000000000000000","8":4383,"9":1281}}' from endpoint 1 with groupID 0
info  2021-01-02 14:18:15: MQTT publish: topic 'zigbee2mqtt/0x00158d0002166ebe', payload '{"consumption":12.08,"linkquality":68,"power":0,"state":"ON","temperature":31}'
debug 2021-01-02 14:18:21: Received Zigbee message from '0x00158d0002235cb3', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2021-01-02 14:19:23: Received Zigbee message from '0x00158d0002235cb3', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2021-01-02 14:19:28: Received Zigbee message from '0x00158d00024862df', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2021-01-02 14:19:28: MQTT publish: topic 'zigbee2mqtt/0x00158d00024862df', payload '{"linkquality":78,"state_left":"OFF"}'
debug 2021-01-02 14:19:35: Received Zigbee message from '0x00158d0002235cb3', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3300,"10":0,"100":1,"101":0,"110":1,"111":0,"153":0,"155":0,"3":27,"5":115,"8":9749}}' from endpoint 1 with groupID 0
info  2021-01-02 14:19:35: MQTT publish: topic 'zigbee2mqtt/0x00158d0002235cb3', payload '{"linkquality":78,"state":"ON","state_left":"ON","temperature":27}'
debug 2021-01-02 14:19:42: Received Zigbee message from '0x00158d0002d7a022', type 'attributeReport', cluster 'genBasic', data '{"65281":{"100":0,"101":0,"149":0.1392333060503006,"152":0,"154":0,"3":32,"5":127,"7":"0x0000000000000000","8":4385,"9":1538}}' from endpoint 1 with groupID 0
info  2021-01-02 14:19:42: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d7a022', payload '{"consumption":0.14,"linkquality":92,"power":0,"temperature":32}'
debug 2021-01-02 14:19:59: Received Zigbee message from '0x00158d0002d7a022', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2021-01-02 14:19:59: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d7a022', payload '{"consumption":0.14,"linkquality":94,"power":0,"state_left":"OFF","temperature":32}'
debug 2021-01-02 14:20:09: Received Zigbee message from '0x00158d0002166ebe', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2021-01-02 14:20:09: MQTT publish: topic 'zigbee2mqtt/0x00158d0002166ebe', payload '{"consumption":12.08,"linkquality":68,"power":0,"state":"ON","temperature":31}'
debug 2021-01-02 14:20:25: Not saving state
debug 2021-01-02 14:20:25: Received Zigbee message from '0x00158d0002235cb3', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2021-01-02 14:21:27: Received Zigbee message from '0x00158d0002235cb3', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
info  2021-01-02 14:22:19: MQTT publish: topic 'zigbee2mqtt/0x00158d0002235cb3/availability', payload 'offline'
info  2021-01-02 14:22:19: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d7a022/availability', payload 'offline'
info  2021-01-02 14:22:19: MQTT publish: topic 'zigbee2mqtt/0x00158d00024862df/availability', payload 'offline'
info  2021-01-02 14:22:19: MQTT publish: topic 'zigbee2mqtt/0x00158d00023a9295/availability', payload 'offline'
info  2021-01-02 14:22:19: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a8ab02/availability', payload 'offline'
info  2021-01-02 14:22:19: MQTT publish: topic 'zigbee2mqtt/0x000d6f0013153342/availability', payload 'offline'
info  2021-01-02 14:22:19: MQTT publish: topic 'zigbee2mqtt/0x00158d00023a9278/availability', payload 'offline'
[... cut ...]
info  2021-01-02 14:22:19: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe783070/availability', payload 'offline'
info  2021-01-02 14:22:19: MQTT publish: topic 'zigbee2mqtt/0x0004740000a18022/availability', payload 'offline'
info  2021-01-02 14:22:19: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2abca2/availability', payload 'offline'
debug 2021-01-02 14:22:19: Not saving state
info  2021-01-02 14:22:19: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
info  2021-01-02 14:22:19: Disconnecting from MQTT server
info  2021-01-02 14:22:20: zigbee-herdsman stopped

VOVAiS, 2021-01-02:
> info  2021-01-02 14:24:27: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"2dc9016f","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20201010,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0x00124b00060fd6da","panID":6755},"permit_join":false,"version":"1.17.0-dev"}'
debug 2021-01-02 14:24:28: Received Zigbee message from '0x00158d00024862df', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2021-01-02 14:24:28: MQTT publish: topic 'zigbee2mqtt/0x00158d00024862df', payload '{"linkquality":81,"state_left":"OFF"}'
debug 2021-01-02 14:24:33: Received Zigbee message from '0x00158d0002235cb3', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2021-01-02 14:24:35: Received Zigbee message from '0x00158d00024862df', type 'attributeReport', cluster 'genBasic', data '{"65281":{"100":0,"101":0,"149":397.25885009765625,"152":0,"154":0,"3":27,"5":77,"7":"0x0000000a7558257d","8":4383,"9":1024}}' from endpoint 1 with groupID 0
info  2021-01-02 14:24:35: MQTT publish: topic 'zigbee2mqtt/0x00158d00024862df', payload '{"consumption":397.26,"linkquality":81,"power":0,"state_left":"OFF","temperature":27}'
debug 2021-01-02 14:24:38: Received MQTT message on 'zigbee2mqtt/0x00158d0002235cb3/right/set/state' with data 'ON'
debug 2021-01-02 14:24:38: Publishing 'set' 'state' to '0x00158d0002235cb3'
info  2021-01-02 14:24:38: MQTT publish: topic 'zigbee2mqtt/0x00158d0002235cb3', payload '{"linkquality":78,"state_right":"ON"}'
debug 2021-01-02 14:24:38: Received Zigbee message from '0x00158d0002235cb3', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440515,"onOff":1}' from endpoint 3 with groupID 0
info  2021-01-02 14:24:38: MQTT publish: topic 'zigbee2mqtt/0x00158d0002235cb3', payload '{"linkquality":81,"state_right":"ON"}'
debug 2021-01-02 14:24:54: Received Zigbee message from '0x00158d0002235cb3', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3300,"10":0,"100":0,"101":1,"110":0,"111":1,"153":0,"155":0,"3":27,"5":115,"8":9749}}' from endpoint 1 with groupID 0
info  2021-01-02 14:24:54: MQTT publish: topic 'zigbee2mqtt/0x00158d0002235cb3', payload '{"linkquality":81,"state":"OFF","state_right":"ON","temperature":27}'
debug 2021-01-02 14:24:59: Received Zigbee message from '0x00158d0002d7a022', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2021-01-02 14:24:59: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d7a022', payload '{"linkquality":94,"state_left":"OFF"}'
debug 2021-01-02 14:25:09: Received Zigbee message from '0x00158d0002166ebe', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2021-01-02 14:25:09: MQTT publish: topic 'zigbee2mqtt/0x00158d0002166ebe', payload '{"linkquality":65,"state":"ON"}'
debug 2021-01-02 14:25:35: Received Zigbee message from '0x00158d0002235cb3', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
debug 2021-01-02 14:26:37: Received Zigbee message from '0x00158d0002235cb3', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
info  2021-01-02 14:26:43: MQTT publish: topic 'zigbee2mqtt/0x00158d0002235cb3/availability', payload 'offline'
info  2021-01-02 14:26:43: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d7a022/availability', payload 'offline'
info  2021-01-02 14:26:43: MQTT publish: topic 'zigbee2mqtt/0x00158d00024862df/availability', payload 'offline'
info  2021-01-02 14:26:43: MQTT publish: topic 'zigbee2mqtt/0x00158d00023a9295/availability', payload 'offline'
info  2021-01-02 14:26:43: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a8ab02/availability', payload 'offline'
info  2021-01-02 14:26:43: MQTT publish: topic 'zigbee2mqtt/0x000d6f0013153342/availability', payload 'offline'
info  2021-01-02 14:26:43: MQTT publish: topic 'zigbee2mqtt/0x00158d00023a9278/availability', payload 'offline'
info  2021-01-02 14:26:43: MQTT publish: topic 'zigbee2mqtt/0x00158d0002166ebe/availability', payload 'offline'
info  2021-01-02 14:26:43: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffeff3d06/availability', payload 'offline'
info  2021-01-02 14:26:43: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffeff2e15/availability', payload 'offline'
info  2021-01-02 14:26:43: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe783070/availability', payload 'offline'
info  2021-01-02 14:26:43: MQTT publish: topic 'zigbee2mqtt/0x0004740000a18022/availability', payload 'offline'
info  2021-01-02 14:26:43: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2abca2/availability', payload 'offline'
debug 2021-01-02 14:26:43: Not saving state
info  2021-01-02 14:26:43: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
info  2021-01-02 14:26:43: Disconnecting from MQTT server
info  2021-01-02 14:26:44: zigbee-herdsman stopped

Koenkk, 2021-01-02:
Fixed.

Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

VOVAiS, 2021-01-02:
Thank you. It works

closed 2021-01-02 by VOVAiS