Koenkk/zigbee2mqtt #5162: Tuya Switch TS0012 (No Neutral) - Doesn't Repair

MUzair030, 2020-12-02:
Hello there, i'm using a Tuya Switch 2 Gang TS0012 (with No Neutral). It works perfectly fine but i'm facing a critical issue that, when I cut off the power of the switch it doesn't pair back (reconnect) with the CC2531 Zigbee adapter (as most devices auto announce themselves) until, I reset the device again!

Do anyone know how to solve that issue?

junaidhoocontrols, 2020-12-28:
Hello, I am having the same issue. I have around 10 of these switches installed in my home and none of them come back online after a power loss. I have to manually repair them again 1 by 1. This is a one of its kind problem that is not faced by any other type of devices i.e. Dimmers, motions sensors, Aqara D1 switches etc. Does anyone else had a similar issue with any other device or has come across a solution?

MJunaidAlam, 2020-12-28:
Hello, I am having the same issue. I have around 10 of these switches (TS0011, TS0012, TS0013 mixed) installed in my home and none of them come back online after a power loss. I have to manually repair them again 1 by 1. This is a one of its kind problem that is not faced by any other type of devices i.e. Dimmers, motions sensors, Aqara D1 switches etc. Does anyone else had a similar issue with any other device or has come across a solution?

MJunaidAlam, 2020-12-29:
### How to reproduce the bug.

Pair the device with zigbee2mqtt. check everything is working fine then cut the power of switch (just the switch and not the coordinator or any other devices). Power the switch back on.

### Debug log
Here is the debug log. Similar messages appear after I cut power of the switch (TS0011 or TS0012 os TS0013) and connect back the power.

```
Zigbee2MQTT:debug 2020-12-29 12:33:39: Received MQTT message on 'zigbee2mqtt/0x60a423fffe93dd23/set' with data '{"state_left":"ON"}'
Zigbee2MQTT:debug 2020-12-29 12:33:39: Publishing 'set' 'state' to '0x60a423fffe93dd23'
Dec 29 12:34:22 hoo sudo[859]: Zigbee2MQTT:error 2020-12-29 12:34:22: Publish 'set' 'state' to '0x60a423fffe93dd23' failed: 'Error: Command 0x60a423fffe93dd23/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'
Dec 29 12:34:22 hoo sudo[859]: Zigbee2MQTT:debug 2020-12-29 12:34:22: Error: Command 0x60a423fffe93dd23/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC transaction expired' (240))
at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:339:27)
at Generator.next (<anonymous>)
at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)
Zigbee2MQTT:info  2020-12-29 12:34:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x60a423fffe93dd23' failed: 'Error: Command 0x60a423fffe93dd23/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'","meta":{"friendly_name":"0x60a423fffe93dd23"},"type":"zigbee_publish_error"}'
Data request failed with error: 'MAC transaction expired' (240)Dec 29 12:35:48 hoo sudo[859]: Zigbee2MQTT:debug 2020-12-29 Saving state to file /opt/zigbee2mqtt/data/state.json
```

### Network map (Graphviz)

[image]

Obervations: 

1. If I cut power of the coordinator and all the devices are powered up, devices work fine after the zigbee2mqtt comes back on
2. All the other devices keep working fine other than the 3 devices mentioned above after power cut.

MJunaidAlam, 2020-12-29:
Further digging deep into the issue I sniffed the packets. I paired two devices, one of them worked fine and the other one is the problematic one. I noticed that after powering back on the two devices, the problematic device broadcasts beacon request on IEEE 802.15.4 (red underlined) just once and does not get any response from the coordinator, while the one that works fine broadcasts on zigbee and keeps doing it, until the coordinator responds which it eventually does. 
[image]

closed 2021-02-06 by github-actions[bot]