Koenkk/zigbee2mqtt #5163: Zigbee device is not showed in HA

tpbillund, 2020-12-02:
zigbee2mqtt add-on version (if edge, please report commit hash):
NODE_VERSION | 12.19.0
YARN_VERSION | 1.22.5

Operating environment (HassOS, Virtual Machine, Device/platform running Home Assistant):
Docker 18.09.0-0513 on Synology DS1618+ with DSM 6.2.3-25426 Update 2.
Home Assistant Core 0.119.0.dev20201127
Supervisor 2020.11.0
HACS 1.6.2

Description of problem:
I have installed zigbee2mqtt and MQTT, when i add devices zigbee is detecting em and add em too MQTT
So the data from the devices is ok, but when i then open HA then I can't find em as sensor, can someone help me?
Say if you need some files provided...

## Information about the device + link
AQARA https://www.aqara.com/us/temperature_humidity_sensor.html

## data/database.db entry of the device
(This is some data from my log inside portainer, for zigbee2mqtt)

Zigbee2MQTT:info  2020-11-28 08:40:54: MQTT publish: topic 'zigbee2mqtt/Temp_stue', payload '{"battery":100,"device":{"applicationVersion":5,"dateCode":"20191205","friendlyName":"Temp_stue","hardwareVersion":30,"ieeeAddr":"0x00158d00030b7984","manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","networkAddress":1715,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"humidity":48.52,"linkquality":68,"pressure":1018.6,"temperature":22.15,"voltage":3015}'

Zigbee2MQTT:info  2020-11-28 08:40:54: MQTT publish: topic 'zigbee2mqtt/Temp_stue', payload '{"battery":100,"device":{"applicationVersion":5,"dateCode":"20191205","friendlyName":"Temp_stue","hardwareVersion":30,"ieeeAddr":"0x00158d00030b7984","manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","networkAddress":1715,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"humidity":48.52,"linkquality":68,"pressure":1018,"temperature":22.15,"voltage":3015}'

insipiens, 2020-12-02:
In HA’s mqtt integration have you set the discovery in system options?

[image]

tpbillund, 2020-12-02:
> 
> 
> In HA’s mqtt integration have you set the discovery in system options?
> 
> [image]
> 
> The other setting which you probably do have is in the Z2m configuration.yaml file...this wouldn’t be directly responsible for your problem though
> 
> [image]

Hi and thx. for reply :-)
The MQTT integration... NO.
I get this question:
Do you want to configure Home Assistant to connect to the MQTT broker provided by the hass.io extension Mosquitto broker?
If I then click "Send" then i keep getting this error: "Failed to connect to broker".

So can't get past this step.

The Z2M configuration.yaml i have this in it:
`homeassistant: true
permit_join: false
# permit_join er true når nye enheder skal tilføjes.
mqtt:
  base_topic: zigbee2mqtt
  user: petersen
  password: xxxxx
  server: 'mqtt://172.25.xxx.xxx:1883'
  client_id: zigb2mqtt
  include_device_information: true
serial:
  port: /dev/ttyACM0
  disable_led: false
devices:
  '0x00158d00054068e9':
    friendly_name: Temp_bad
  '0x00158d00030b7984':
    friendly_name: Temp_stue`

Hope that can help, guide me :-)

insipiens, 2020-12-02:
The integration should point to the same broker ip as shown in your z2m config, including username and password (you might have to add that broker user as a user in HA).

If you reinstall the mqtt integration it should ask you for ip and user credentials as part of the installation process.

tpbillund, 2020-12-03:
Okay let me try again... bc there must be something that i do wrong...

**##Portainer##**
Portainer shows that my "Zigbee2MQTT" container is running and is logging data from my two devices.
`
Zigbee2MQTT:info  2020-12-03 17:55:05: MQTT publish: topic 'zigbee2mqtt/Temp_stue', payload '{"battery":100,"device":{"applicationVersion":5,"dateCode":"20191205","friendlyName":"Temp_stue","hardwareVersion":30,"ieeeAddr":"0x00158d00030b7984","manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","networkAddress":1715,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"humidity":48.35,"pressure":1013.7,"temperature":22.29,"voltage":3015}'

Zigbee2MQTT:info  2020-12-03 18:00:22: MQTT publish: topic 'zigbee2mqtt/Temp_bad', payload '{"battery":97,"device":{"applicationVersion":5,"friendlyName":"Temp_bad","ieeeAddr":"0x00158d00054068e9","manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","networkAddress":52912,"powerSource":"Battery","type":"EndDevice"},"humidity":38.44,"linkquality":123,"pressure":988.9,"temperature":24.2,"voltage":2995}'
`
Portainer also show that the container "addon_core_mosquitto" is running.
ACM is running manual network port publishing host:1884-1885 --> container:1883-1884 (if host is 1883-1884, then ACM will not start, bc 1883 is in use).

**##Docker##**
On my synology with the ip 172.25.100.201 I have the folder "Docker" and inside that i have "zigbee2mqtt" folder and inside that i have "Data" folder.
Here I have a configuration.yaml file, that have this code inside (The devices is automatic added):
`
homeassistant: true
permit_join: false 
mqtt:
  base_topic: zigbee2mqtt
  user: mqtt
  password: xxxxxxx
  server: 'mqtt://172.25.100.201:1883'
  client_id: zigb2mqtt
  include_device_information: true
serial:
  port: /dev/ttyACM0
  disable_led: false
devices:
  '0x00158d00054068e9':
    friendly_name: Temp_bad
  '0x00158d00030b7984':
    friendly_name: Temp_stue
`

**##Home Assistant##**
In HA i have added an user "mqtt" with password.
Supervisor shows that "Mosquitto broker" runs and is using this configuration:
`
logins:
  - username: mqtt
    password: xxxxxx
anonymous: false
customize:
  active: false
  folder: mosquitto
certfile: fullchain.pem
keyfile: privkey.pem
require_certificate: false
`
Log is in Mosquitto broker is giving these errors:
[17:55:23] ERROR: Got unexpected response from the API: System is not ready with state: setup
[17:55:23] INFO: Setup mosquitto configuration
[17:55:23] WARNING: SSL not enabled - No valid certs found!
[17:55:23] INFO: Found local users inside config
[17:55:23] INFO: Initialize Hass.io Add-on services
[17:55:23] ERROR: Can't setup Hass.io service mqtt
[17:55:23] INFO: Initialize Home Assistant discovery
[17:55:23] ERROR: Can't setup Home Assistant discovery mqtt
[17:55:23] INFO: Start Mosquitto daemon
1607014523: Error: Unable to open include_dir '/share/mosquitto'.
1607014523: Error found at /etc/mosquitto.conf:29.

Inside HA configuration --> integration, MQTT is running, but is not showing anything inside "configuration".
Inside HA /config/configuration.yaml I have this code:
`
mqtt:
  broker: core-mosquitto # This will have to be your mqtt broker, Mosquitto addon is recommended.
  discovery: true
` 

**## MQTT Integration ##**
If I here add:
Broker: core-mosquitto
Port: 1883
Username: mqtt
Password: xxxxxxxx

Then i get "Failed to connect".

So I can't see my devices inside HA, but zigbee2mqtt container, do log info from em.
What to do?

insipiens, 2020-12-03:
Are your container’s ports mapped to the host?

tpbillund, 2020-12-04:
> 
> 
> Are your container’s ports mapped to the host?

Hi
Can you explain what you mean by "mapped to host" ?

insipiens, 2020-12-04:
I’m not a docker user but as I understand it only when a container has exposed a device or port can it be seen by applications running on the host, or other containers.

It doesn’t sound like this issue is probably a Z2M problem, maybe you should raise a ticket on the mqtt or ha sites as appropriate? 

To confirm that, you could listen to the zigbee2mqtt topic within your mqtt broker, or use mqtt-explorer.  http://mqtt-explorer.com/

biscuit-316, 2021-01-01:
MQTT discovery doesn't work for me either. But if I add my light via the "Manual Home Assistant configuration" my light does work.

does the sensor work for you if you add it yourself? 
[https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html](https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html)

closed 2021-02-09 by github-actions[bot]