Koenkk/zigbee2mqtt #5164: Support for Lidl / LivarnoLux RGB+CCT light models: GU10, E14, E27

maichai, 2020-12-02:
## Information about the device + link

There are 3 RGB+CCT models sold by Lidl under the brand LivarnoLux currently:

GU10 RGB+CCT: Model is TS0505A by _TZ3000_kdpxju99.
E27 RGB+CCT: Model is TS0505A by _TZ3000_dbou1ap4
E14 RGB+CCT: Model is TS0505A by _TZ3000_odygigth.

## data/database.db entry of the device

GU10 RGB+CCT:

{"id":redacted,"type":"Router","ieeeAddr":"redacted","nwkAddr":redacted"manufId":4098,"manufName":"_TZ3000_kdpxju99","powerSource":"Mains (single phase)","modelId":"TS0505A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505A","manufacturerName":"_TZ3000_kdpxju99","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1606914174834}

E27 RGB+CCT:

{"id":redacted,"type":"Router","ieeeAddr":"redacted","nwkAddr":redacted,"manufId":4098,"manufName":"_TZ3000_dbou1ap4","powerSource":"Mains (single phase)","modelId":"TS0505A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505A","manufacturerName":"_TZ3000_dbou1ap4","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1606915778403}

E14 RGB+CCT:

{"id":redacted,"type":"Router","ieeeAddr":"redacted","nwkAddr":redacted,"manufId":4098,"manufName":"_TZ3000_odygigth","powerSource":"Mains (single phase)","modelId":"TS0505A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505A","manufacturerName":"_TZ3000_odygigth","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1606916481672}

Koenkk, 2020-12-02:
Can you provide the models for all these device? (should start with HG, e.g. for the plug it's HG06337)

sjorge, 2020-12-02:
1/3 already seems to be support, https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L15364

Anjerlaan, 2020-12-02:
@Koenkk, you already did one of them:
Distinguish HG06106B from TS0505A. Koenkk/zigbee2mqtt #5109 at
https://github.com/Koenkk/zigbee-herdsman-converters/commit/ea8bf9ba8866454dde46a3297efffc977d538602

These are the model names, And I think the vendor is not Silvercrest (as you wrote at vendor name) but Livarno lux according to the lidl site and the manual 
[image]
https://www.lidl-service.com/cps/rde/xchg/SID-9DC000A7-0D2C39EB/lsp/hs.xsl/product.html?id=5019441699&title=LED-lamp&count=1

Koenkk, 2020-12-02:
Since Lidl uses 3 different names (silvercrest, livarno lux and Melinera) I've changed the vendor to "Lidl" and put the name in front of description, e.g. Livarno Lux E14 candle RGB, Melinera smart LED string lights, Silvercrest smart motion sensor. Now Lidl devices are better to recognise.  

Assuming this can be closed now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-02 by Koenkk

Anjerlaan, 2020-12-02:
Should the line: "vendor: 'Silvercrest'" of the HG06106B not be removed? now there are 2 vendors

```
{
        fingerprint: [{modelID: 'TS0505A', manufacturerName: '_TZ3000_odygigth'}],
        model: 'HG06106B',
        vendor: 'Silvercrest',
        description: 'E14 candle',
        vendor: 'Lidl',
        description: 'Livarno Lux E14 candle RGB',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
    },
    {
        fingerprint: [{modelID: 'TS0505A', manufacturerName: '_TZ3000_kdpxju99'}],
        model: 'HG06106A',
        vendor: 'Lidl',
        description: 'Livarno Lux GU10 spot RGB',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
    },
    {
        fingerprint: [{modelID: 'TS0505A', manufacturerName: '_TZ3000_dbou1ap4'}],
        model: 'HG06106C',
        vendor: 'Lidl',
        description: 'Livarno Lux E27 bulb RGB',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
    },
```

Koenkk, 2020-12-02:
@Anjerlaan you are looking at a diff view (https://github.com/Koenkk/zigbee-herdsman-converters/commit/13be3d24088412ae48dfce68879dcab8310114fd#diff-5520dd49755e0a353d3857b8963a91cb4b8d87a01b6b8eb7bb6cce1c180eac83R15375) red is removed, green is added. (so there is only one of each)

Anjerlaan, 2020-12-02:
Sorry your'e right....red and green...I must be color blind ;-)

maichai, 2020-12-02:
The lights work in principle with m2z 1.16.2 - BUT - it would be really nice if the transition time that is provided with the command to change the color/brightness would also be used. Currently there is a short transition of 1-2s to fade from the current brightness/color to the now newly set brightness/color. However, if the command contains a transition time, then the fade should take as long as specified. Can this be done or doe these lights not support that?

Robbie-65, 2020-12-03:
Hi,
I would like to connect the lights to my Vera Plus.
I can connect them, but can only put the lights on and off. 
Not dimm or control the color temperature.
I probably need a correct XML and Json file in my Vera Plus, because the lights are not currently supported.
Who can help me with this?

Rob

goliath888, 2020-12-06:
I saw in the dev branch that there is already support for this devices.

@Koenkk, would be possible to add also support for the GU10 non-RGB?

I think it would be just as RGB version but without the colorxy part.

It is recognized as [this ](https://www.zigbee2mqtt.io/devices/TS0502A.html)device, but it should be [this](https://zigbeealliance.org/es/zigbee_products/lidl-home-bombilla-led-inteligente-3/) one.

This is in my database:

{"id":22,"type":"Router","ieeeAddr":"0x588e81fffef18e00","nwkAddr":38633,"manufId":4098,"manufName":"_TZ3000_el5kt5im","powerSource":"Mains (single phase)","modelId":"TS0502A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0502A","manufacturerName":"_TZ3000_el5kt5im","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1607248360777}

Something like this?

    {
        fingerprint: [{modelID: 'TS0502A', manufacturerName: '_TZ3000_el5kt5im'}],
        model: 'HG06492A',
        vendor: 'Lidl',
        description: 'Livarno Lux GU10',
        extend: generic.light_onoff_brightness_colortemp,
    },

Thanks for your help.

Koenkk, 2020-12-06:
@goliath888 responded to the HA part in the other issue, added this definition.

slima, 2022-08-15:
Is possible to add power_on_behavior on this device?