Koenkk/zigbee2mqtt #5165: HEIMAN HS1SA Smoke detector becomes unavailable

groenmarsmannetje, 2020-12-02:
## What happened
I have 3 of these smoke detectors and they become unavailable after a day. I have availibity detection turned on. The strange thing is that earlier (months ago) these devices at least send a message once every 24 hours, but that isn’t happening anymore. The last seen timestamp remains unchanged for days and weeks. They are still connected because when I press the test button the smoke sensor will turn on. 

## What did you expect to happen
At least they should come out of deep sleep more frequently and a message should be triggered. That is what I actually saw happening with these devices. But probably that was with different stick, firmware and zigbee2mqtt version.

Of course I am aware that I could block these devices from availability check, but now I have to physically test them each week to be sure that they are still connected, because I have no information that they actually are.

## How to reproduce it (minimal and precise)
Connect the device and leave it unharmed for a couple of days. See if last seen timestamp updates and maybe also linkquality level.

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: zzh
Adapter firmware version: 20200805

Anjerlaan, 2020-12-02:
Are they disconnected from Z2M network too? (create network map to see or in log search for error: no network route)
Maybe you should update ZZH to latest firmware:
1. First (force) remove the smoke devices
2. Update ZZH firmware to "20201026"
3. Pair the devices

I had similar problems with the Xiaomi Honeywell Smoke detector devices
Read this thread #4285

groenmarsmannetje, 2020-12-02:
Thank you for sharing this. I have done the steps that you described. Let’s see how it behaves next couple of days.

groenmarsmannetje, 2020-12-06:
I can confirm that they now stay connected and I see a linkquality and last seen update approximately once per day again. Issue closed.

closed 2020-12-06 by groenmarsmannetje