Koenkk/zigbee2mqtt #5166: Lidl / Silvercrest power plug (1 socket)

maichai, 2020-12-02:
## Information about the device + link

Lidl / Silvercrest power plug (1 socket)

## data/database.db entry of the device

{"id":redacted,"type":"Router","ieeeAddr":"redacted","nwkAddr":redacted,"manufId":4098,"manufName":"_TZ3000_kdi2o9m6","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_kdi2o9m6","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1606918875020}

Koenkk, 2020-12-02:
Device is already supported in zigbee2mqtt 1.16.2

closed 2020-12-02 by Koenkk

maichai, 2020-12-02:
just upgraded m2z and can confirm that it works fine.