Koenkk/zigbee2mqtt #5167: Lidl / Silvercrest motion sensor

maichai, 2020-12-02:
## Information about the device + link

Lidl / Silvercrest motion sensor, battery driven

## data/database.db entry of the device

{"id":redacted,"type":"EndDevice","ieeeAddr":"redacted","nwkAddr":redacted,"manufId":4619,"manufName":"_TZ1800_fcdjzz3s","powerSource":"Battery","modelId":"TY0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TY0202","manufacturerName":"_TZ1800_fcdjzz3s","powerSource":3,"zclVersion":1,"appVersion":65,"stackVersion":2,"hwVersion":1,"dateCode":"2019.12.12"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b001ca1b526","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[],"meta":{}}},"appVersion":65,"stackVersion":2,"hwVersion":1,"dateCode":"2019.12.12","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1606920279404}

sjorge, 2020-12-02:
Already supported in dev: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L15286

Koenkk, 2020-12-02:
Device is already support in 1.16.2

closed 2020-12-02 by Koenkk

maichai, 2020-12-02:
Confirmed to work fine.

Fridolin34, 2020-12-03:
Sorry, but this is still not working for me. Updated everything, tried stable and edge version, but it simply doesn't work.

I've bought a couple of Silvercrest products today and although the icons shown in the list/map is not making sense, all other devices have been nicely recognized and can be managed without any problems.

Just the motion sensor is not working.

nightscoutteun, 2021-01-05:
What error do you got?

Fridolin34, 2021-01-05:
Thanks for the reply. My problem is gone. I've set up everything freshly from scratch and now it works.

platte-73, 2021-02-01:
I have also a problem with the lidl smart home sensor.

I use domoticz with synology and when I want to pair the sensor there are 5 devices showing. One of them has a motion sensor icon and I add that device. But when I go to the switches tab it is there but it wont update his state. Last time update is when I paired it

Am I doing something wrong. I copied and past the new found devices

		976	Zigbee2MQTT	0x680ae2fffe01f250_signal	2	Zigbee2MQTT - 0x680ae2fffe01f250 (Link Quality)	General	Custom Sensor	70	3	-	   	2021-02-01 20:44:17
		984	Zigbee2MQTT	api_transport	255	Zigbee2MQTT - Zigbee2MQTT API Transport	General	Text		-	-	   	2021-02-01 20:34:47
		975	Zigbee2MQTT	0x588e81fffee93df6_light	8	Lamp Zigbee	Color Switch	RGBWW	On	3	-	   	2021-02-01 20:03:04
		977	Zigbee2MQTT	0xbc33acfffe06829e_motion	3	Sensor Zigbee	Light/Switch	Switch	Off	-	-	   	2021-02-01 20:03:04
		978	Zigbee2MQTT	0xbc33acfffe06829e_lowbtr	4	Zigbee2MQTT - 0xbc33acfffe06829e (Low Battery)	Light/Switch	Switch	Closed	-	-	   	2021-02-01 20:03:04
		979	Zigbee2MQTT	0xbc33acfffe06829e_tamper	5	Zigbee2MQTT - 0xbc33acfffe06829e (tamper)	Light/Switch	Switch	Closed	-	-	   	2021-02-01 20:03:04
		980	Zigbee2MQTT	0xbc33acfffe06829e_btperc	6	Zigbee2MQTT - 0xbc33acfffe06829e (Battery)	General	Percentage	0%	-	-	   	2021-02-01 20:03:04
		981	Zigbee2MQTT	0xbc33acfffe06829e_signal	7	Zigbee2MQTT - 0xbc33acfffe06829e (Link Quality)	General	Custom Sensor	0	-	-	   	2021-02-01 20:03:04
		982	Zigbee2MQTT	0x588e81fffee93df6_effec	9	Zigbee2MQTT - 0x588e81fffee93df6 (effect)	Light/Switch	Selector Switch	Off	-	-	   	2021-02-01 20:03:04
		983	Zigbee2MQTT	0x588e81fffee93df6_signal	10	Zigbee2MQTT - 0x588e81fffee93df6 (Link Quality)	General