Koenkk/zigbee2mqtt #5168: Sonoff SNZB-01 not recognized

CBismark, 2020-12-02:
## What happened
Tried to pair a Sonoff SNZB-01 to my zigbee2mqtt instance.
## What did you expect to happen
It should pair, since it is listed as a supported device
## How to reproduce it (minimal and precise)
Try to pair a Sonoff SNZB-01.

I already found out what the problem was: The SNZB-01 showed up as "modelId":"WB-01" (read out from the database) and in the devices.js, the SNZB-01 is only listed as WB01. I added WB-01 to the devices.js and then it worked. Please add the zigbee model WB-01 to the devices.js, so nobody else runs into the same problem.

Regards

CBismark

sjorge, 2020-12-02:
Already fixed in the dev branch I think https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L13348

CBismark, 2020-12-02:
wow you're quick. I wanted to check, but didn't find the devices.js in github (new to this, sorry). Anyway, it was interviewed correctly but I still can't seem to use it. no presses are recognized.

closed 2020-12-02 by CBismark

sjorge, 2020-12-02:
There are different repos...
Koenkk/zigbee2mqtt -> zigbee2mqtt, basically a mqtt to zigbee gateway
Koenkk/zigbee-herdsman -> building blocks to build a zigbee coordinator
Koenkk/zigbee-herdsman-converters -> convertors to translate attributes to zigbee and back, this is where devices get added.

realchika, 2021-02-06:
Hi,  i have the same issue With my snzb01, how can i use this update? I saw the .js file but don't really know what to do With ...

CBismark, 2021-02-07:
You should be able to switch to dev branch ([https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html]) and there it would already be fixed.