Koenkk/zigbee2mqtt #5169: Gledopto Lamp (GL-S-003Z) endpoint bug

NoAlucard, 2020-12-02:
## What happened
I've six Gledoto GU10 RGBW Spots they provide three endpoints 11,12,13 the lamps pair successfull but if you send a command to the Lamp, for example change color the lamps executes the command but zigbee2mqtt gets an timeout.

After some testing i created a group (in the new webinterface) wich uses the endpoint 12 of these Lamps. If i send now a command to this group the lamp work as axpected, and zigbee also gets a answer from the lamp. (Also the state on the webif gets updated)
## What did you expect to happen
On a command to a single Lamp zigbee should get a response

Logfile of timeout: https://pastebin.com/ZEMTx00z

## How to reproduce it (minimal and precise)
Freshly pair the lamp. Send a command to the Lamp -> timeout

## Lamp Info

- Description: **Smart RGBW GU10** 
- IEEE address: **0x00124b001a066cdd**
- Network address: **0xBC71**
- Firmware build date: **20131206**
- Firmware version: **1.0.3**
- Vendor: **Gledopto**
- Model: **GL-S-003Z**
- Power source: **Mains(single phase)**
- Interview completed: **Yes**

## Debug info
Zigbee2MQTT version: **1.16.2 commit: 04c15f7**
Adapter hardware: **zig-a-zig-ah! (CC2652 Stick)**
Adapter firmware version: **20201026**
Coordinator type: **zStack3x0**
Frontend version: **0.2.100**

Koenkk, 2020-12-02:
Does changing the definition to:

```js
{
    zigbeeModel: ['GL-S-003Z'],
    model: 'GL-S-003Z',
    vendor: 'Gledopto',
    description: 'Smart RGBW GU10 ',
    extend: gledopto.light_onoff_brightness_colorxy,
    meta: {disableDefaultResponse: true},
},
```

fixes it?

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

NoAlucard, 2020-12-04:
Firstly yes, zigbee now wont show up an error, but it seems more like a workaround because the "get status" wont work either.
As i mentioned befor if the endpoint 12 is used the lamp seems to send the state as it should.

It's okay for me if you close this issue with this solution but im also willing to help to find a solution where the lamp sends a correct state

Koenkk, 2020-12-06:
Can you try with:
```js
{
    zigbeeModel: ['GL-S-003Z'],
    model: 'GL-S-003Z',
    vendor: 'Gledopto',
    description: 'Smart RGBW GU10 ',
    extend: gledopto.light_onoff_brightness_colorxy,
    endpoint: (device) => {
        return {default: 12};
    },
},
```

NoAlucard, 2020-12-06:
> 
> 
> Can you try with:
> 
> ```js
> {
>     zigbeeModel: ['GL-S-003Z'],
>     model: 'GL-S-003Z',
>     vendor: 'Gledopto',
>     description: 'Smart RGBW GU10 ',
>     extend: gledopto.light_onoff_brightness_colorxy,
>     endpoint: (device) => {
>         return {default: 12};
>     },
> },
> ```

Yep, that fixed it. Now the lamp answers the requests like a charm.
Thank you very much :)

closed 2020-12-06 by NoAlucard