Koenkk/zigbee2mqtt #5170: Trust ZYCT-202 Binding fails

Ulrar, 2020-12-02:
## What happened
I've found the Hue switches to be highly unreliable, failing to do anything on a good third of the presses.
So I've bought a Trust ZYCT-202 switch and paired it with zigbee2mqtt, in the hope that it might work a bit better.

It shows up in the web interface and in home assistant fine, and home assistant does seem to be receiving the button presses fine too.
However binding (from the zigbee2mqtt web interface) it to a light or a group of lights fails :

```
ERROR: Failed to bind cluster 'genOnOff' from '0x00158d00038e5fec' to 'kitchen_lights' (Error: Bind 0x00158d00038e5fec/1 genOnOff from '2' failed (AREQ - ZDO - bindRsp after 10000ms))
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genOnOff","from":"0x00158d00038e5fec","to":"kitchen_lights"},"type":"device_bind_failed"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":["genOnOff"],"from":"0x00158d00038e5fec/1","to":"kitchen_lights"},"error":"Failed to bind","status":"error"}'
ERROR: Failed to bind
```
I did try to mash buttons constantly during the binding, like I had to do during the interview process, but it always ends like this.

## What did you expect to happen

Bind the device to the light or group of lights to control them without going through home assistant.
Also the action_group shows in the web interface but isn't seen by home assistant. I'd rather bind different groups of lights to each but if the only way to make this work is through home assistant, it'd be very helpful to expose the action_group.

## How to reproduce it (minimal and precise)

Try to bind the On/Off cluster of that switch to a light

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: ZZH
Adapter firmware version: Haven't touched it, not sure.

Thanks !

sjorge, 2020-12-02:
Does the device support bindings? (Not all do, and I couldn't find if this one does at a quick look around)

Ulrar, 2020-12-02:
Is there an easy way to tell ?
I think it's meant to be bound directly to lights through touchlink, so I assume yes, but maybe that's the only way to do it.

sjorge, 2020-12-02:
I usually sniff and see if the device ignores or responds with an error. Which might not be doable for everyone

Ulrar, 2020-12-02:
Right I've never done that, I doubt I have the hardware for it.
The binding directly to bulbs using touchlink works great, and they get controlled by group which is very handy, but unfortunately it's not binding with devices that don't have touchlink like my kitchen LED strips. I was kind of hoping it'd bind like a hue switch to any group but oh well.
That's unfortunate, from my limited testing now it does work so much better than a hue switch !

MattWestb, 2020-12-02:
With light controlling devices is common that they only can being bounded to one device / group like the Philips HUE Dimmer Switch.
The coordinator is normally binding all possible cluster to it self the pairing devices and its necessary first unbinding the cluster/s from the coordinator and then binding them to the groupe (also normal only accepting binding to groups and not devices.

And don't forget thing keeping the remote awake then sending the command ;-)))

Ulrar, 2020-12-02:
That's not the case here, it doesn't bind to the coordinator either. The
action does work in home assistant though, even though it's not bound.

The bind menu shows all the clusters but by default they're all unticked,
and trying to bind one of them to anything times out.

MattWestb, 2020-12-02:
ZHA its lazy doing the coordinator binding is our experience so can working if doing the binding fast after pairing. The HUE was fixed in ZHA by implanting exclusive unbinding from the coordinator for some weeks ago with cluster commands.

If the on/off cluster (0x0006) was not bonded to the coordinator then you dont can see the command from the switch in Z2M. Can you see the command from the switch in Z2M ?
Is you having one default coordinator / binding groupe groupe ?
It can also being that the switch have making one own light groupe (the old IKEA motion sensor is doing that) and if yes put the lights in the "light groupe" (if you can getting it from the log ot the lighting link cluster).

milobt, 2020-12-03:
You can authorise these to work with the hue hub to control bulbs paired with the hue hub.
I had problems with Z2M controlling Gledopto bulbs, so opted to pair all bulbs with hue hub - so also have control when HA or Z2M crashes - and then any other zigbee devices and switches use Z2M.

Trust have instruction for linking this remote with the hue hub on their website: 
[image]

You can then touchlink them to bulbs and group them.
I have two now and work great, especially with allowing smooth dimming.
Hope this helps.

Ulrar, 2020-12-03:
I'm not using the hue hub, only Z2M.
But I did pair things manually using touchlink which works fine, you just
can't bind from Z2M.

Also when using it that way Z2M doesn't know about the lights changing
state, but I think I need to get reporting working to fix that.

DirkBaumeister, 2020-12-09:
I'm also facing this problem but in my case this exact same device is refusing to complete configuration. It states:
`Zigbee2MQTT:error 2020-12-09 14:07:29: Failed to configure 'Fernbedienung Mini', attempt 3 (Error: Bind 0x00158d00018559f1/1 genOnOff from '0x00124b001949c5d5/1' failed (AREQ - ZDO - bindRsp after 10000ms)`
Any ideas on this one? I mean, it is working perfectly so but it's kind of annoying to have this messages on each restart of Z2M.

timoline, 2020-12-17:
Same here

ERROR: Failed to bind cluster 'genOnOff' from 'remote/remote1' to 'outside/christmas' (Error: Bind 0x00158d0004ff5ea7/1 genOnOff from '1' failed (AREQ - ZDO - bindRsp after 10000ms))
INFO: MQTT publish: topic 'z2m/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":["genOnOff"],"from":"remote/remote1/1","to":"outside/christmas"},"error":"Failed to bind","status":"error"}'
ERROR: Failed to bind

timoline, 2020-12-17:
@Koenkk or @sjorge ny idea or solution, how can we help?

DirkBaumeister, 2020-12-17:
I kinda fixed it for me by manually adding `"meta":{"configured":2}` to the entity in the database.db. It was `"meta":{}` before my change so I'm asuming that the configuration is just not finishing.

efmuurman, 2020-12-17:
Do you also now have working buttons ?

DirkBaumeister, 2020-12-17:
My buttons were already working, even with this annoying error message. My fixed just got me getting rid of the error message on restart of z2m

timoline, 2020-12-17:
> My buttons were already working, even with this annoying error message. My fix just let me getting rid of the error message on restart of z2m

ok thx,not sure what i'm doing, but is your battery status working?

DirkBaumeister, 2020-12-17:
Nope, just getting `action`, `action_group` and `link_quality`. No `battery` so far. But beside that everything is working. I even can split my actions based on the action_group in Home Assistant.

timoline, 2020-12-17:
and binding is also working?

DirkBaumeister, 2020-12-17:
I don't think so. I guess my "fix" just suppressed the constant reconfigure of the entity and just skipped the binding part. But because it works like a charm together with my Home Assistant I have stopped trying to figure it out 😄

efmuurman, 2020-12-18:
> > My buttons were already working, even with this annoying error message. My fix just let me getting rid of the error message on restart of z2m
> 
> ok thx,not sure what i'm doing, but is your battery status working?

Can you post what is in the database file?
mine is just detected. but not recognized

closed 2021-01-25 by github-actions[bot]