Koenkk/zigbee2mqtt #5172: Ikea Tradfri Motion Sensor Issue

b-reich, 2020-12-02:
## What happened
On two new IKEA E1525/E1745

```
zigbee2mqtt    | Zigbee2MQTT:info  2020-12-02 20:29:10: MQTT publish: topic 'zigbee2mqtt/motion_indoor', payload '{"linkquality":96,"occupancy":true,"update_available":false}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-12-02 20:29:14: MQTT publish: topic 'zigbee2mqtt/motion_outdoor', payload '{"linkquality":102,"occupancy":true,"update_available":false}'
```
## What did you expect to happen
The senosor wont report battery, requested_brightness_level and requested_brightness_percent

## Debug info
Zigbee2MQTT version:1.16.2
Adapter hardware: slaesh’s CC2652RB stick 
Adapter firmware version: 20200925

sjorge, 2020-12-02:
How long ago did you pair them? I believe they only report once every 24 hours.

b-reich, 2020-12-02:
@sjorge this could be the problem. Its paired for ~1h now

sjorge, 2020-12-02:
I’d suggest waiting until tomorrow.

Or if you are really impatient, you can pull the battery, count to 10 and insert it again. The remotes seem to send the battery info right after deviceAnnouncement. (Usually for me atleast) It might also be the case for the motion sensors.

b-reich, 2020-12-03:
1 of 2 reports the battery at the moment. This sould be added here https://www.zigbee2mqtt.io/devices/E1525_E1745.html#battery-numeric.

b-reich, 2020-12-04:
Now every sensor reported a battery state

closed 2020-12-04 by b-reich