Koenkk/zigbee2mqtt #5175: Lidl / LivarnoLux lights to not use transition time provided

maichai, 2020-12-02:
## What happened

The new Lidl / LivarnoLux lights (see [support request](https://github.com/Koenkk/zigbee2mqtt/issues/5164#issue-755297003)) work in principle with m2z 1.16.2 - BUT - it would be really nice if the transition time that is provided with the command to change the color/brightness would also be used. Currently there is a short transition of 1-2s to fade from the current brightness/color to the now newly set brightness/color. However, if the command contains a transition time, then the fade should take as long as specified. Can this be done or do these lights not support that? (as a comparison you may look at the Osram/LedVance Smart+ Spot GU10 lights, they do use the fade transition time)

## What did you expect to happen

If the command to change brightness/color contains a transition time, then the fade should take as long as specified. Can this be done or do these lights not support that? (as a comparison you may look at the Osram/LedVance Smart+ Spot GU10 lights, they do use the fade transition time)

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC26X2R1
Adapter firmware version: zStack3x0

Koenkk, 2020-12-03:
Of all supported Lidl devices, only the HG06467 has no transition support (from a Zigbee2MQTT perspective). If transition doesn't work on the others it means those don't support it.

maichai, 2020-12-03:
What a pity.

And what about the effects that can be selected, but all do the same? Is that also a shortcoming of the firmware of the lights?

Koenkk, 2020-12-03:
Yes, all we send is default Zigbee cluster library, if it doesn't work the issue is in the firmwares of the lights.

closed 2020-12-04 by maichai

CypherMK, 2021-02-03:
I have this Lidl light bar: https://www.zigbee2mqtt.io/devices/14149505L_14149506L.html

I'm also trying to use 'transition' for brightness and color. But it doesn't seem to work. I guess it is the same issue as mentioned here?