Koenkk/zigbee2mqtt #5176: Support For Aurora AOne One Gang Wireless Battery Rotary Dimmer (AU-A1ZBR1GW)

ashp8i, 2020-12-03:
Code in devices.js

{
        zigbeeModel: ['1GBatteryDimmer50AU'],
        model: 'AU-A1ZBR1GW',
        vendor: 'Aurora Lighting',
        description: 'AOne One Gang Wireless Battery Rotary Dimmer',
        supports: 'action',
        fromZigbee: [fz.battery, fz.command_on, fz.command_off, fz.greenpower_on_off_switch, fz.command_step, fz.command_move_to_color_temp, fz.command_move_to_color, fz.command_stop, fz.command_move, fz.command_color_loop_set, fz.command_ehanced_move_to_hue_and_saturation, fz.tint404011_scene],
        toZigbee: [tz.light_brightness_step, tz.light_colortemp, tz.light_colortemp, tz.light_color_colortemp],
        exposes: [e.battery(), e.action(['on', 'off', 'brightness_step_up', 'brightness_step_down', 'color_temperature_move', 'color_move', 'brightness_stop', 'brightness_move_up', 'brightness_move_down', 'color_loop_set', 'enhanced_move_to_hue_and_saturation', 'recall_*', 'scene_*'])],
        ota: ota.zigbeeOTA,
        meta: {configureKey: 3},
        endpoint: (device) => {
            return {default: 1, system: 242};},
        configure: async (device, coordinatorEndpoint) => { 
        	const endpoint1 = device.getEndpoint(1); 
        	await bind(endpoint1, coordinatorEndpoint, ['genBasic', 'genIdentify','genOnOff', 'genLevelCtrl', 'lightingColorCtrl', 'genPowerCfg']);
            await configureReporting.onOff(endpoint1);
            await configureReporting.onOff.batteryPercentageRemaining(endpoint)
            await configureReporting.genLevelCtrl(endpoint1);
            const endpoint242 = device.getEndpoint(242); 
            await bind(endpoint242, coordinatorEndpoint, ['greenPower']);
        },

Zigbee2MQTT:debug 2020-12-03 00:03:38: Received Zigbee message from 'Single Rotary Dimmer Remote', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-03 00:03:38: No converter available for 'AU-A1ZBR1GW' with cluster 'lightingColorCtrl' and type 'commandStepColorTemp' and data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}'
Zigbee2MQTT:debug 2020-12-03 00:03:38: Received Zigbee message from 'Single Rotary Dimmer Remote', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":24,"currentHue":24}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-03 00:03:38: No converter available for 'AU-A1ZBR1GW' with cluster 'lightingColorCtrl' and type 'attributeReport' and data '{"colorTemperature":24,"currentHue":24}'
Zigbee2MQTT:debug 2020-12-03 00:03:39: Received Zigbee message from 'Single Rotary Dimmer Remote', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-03 00:03:39: No converter available for 'AU-A1ZBR1GW' with cluster 'lightingColorCtrl' and type 'commandStepColorTemp' and data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}'
Zigbee2MQTT:debug 2020-12-03 00:03:39: Received Zigbee message from 'Ashish Smart Extension', type 'attributeReport', cluster 'genBasic', data '{"65506":24,"appVersion":66}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-03 00:03:39: Received Zigbee message from 'Single Rotary Dimmer Remote', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":42,"currentHue":42}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-03 00:03:39: No converter available for 'AU-A1ZBR1GW' with cluster 'lightingColorCtrl' and type 'attributeReport' and data '{"colorTemperature":42,"currentHue":42}'
Zigbee2MQTT:debug 2020-12-03 00:03:49: Received Zigbee message from 'Dining Spot', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 3 with groupID 0

Trying to get battery where to start

Koenkk, 2020-12-03:
Battery can take up to 24 hours to report, your config looks good already.

fawksie, 2020-12-21:
Hi @ashp8i, thanks for your work on this device so far. Have you done anything with the converters yet?

The device does not support the batteryPercentageRemaining attribute, you need to use batteryVoltage and convert it to a percentage. Battery will report correctly after one hour and every hour after that.

The firmware is not available in Zigbee-OTA currently and I can't find it online so I have commented the OTA: line.

I have also commented another line in the configure: block which seems to cause errors that I haven't investigated yet. 

So my devices.js looks like this:

```
    {
        zigbeeModel: ['1GBatteryDimmer50AU'],
        model: 'AU-A1ZBR1GW',
        vendor: 'Aurora Lighting',
        description: 'AOne One Gang Wireless Battery Rotary Dimmer',
        supports: 'action',
        fromZigbee: [fz.battery, fz.command_on, fz.command_off, fz.greenpower_on_off_switch, fz.command_step, fz.command_move_to
        toZigbee: [tz.light_brightness_step, tz.light_colortemp, tz.light_colortemp, tz.light_color_colortemp],
        exposes: [e.battery(), e.action(['on', 'off', 'brightness_step_up', 'brightness_step_down', 'color_temperature_move', 'c
    //  ota: ota.zigbeeOTA,
        meta: {configureKey: 1, battery: {voltageToPercentage: '3V_2100'}},
        endpoint: (device) => {
            return {default: 1, system: 242};
        },
        configure: async (device, coordinatorEndpoint) => {
            const endpoint1 = device.getEndpoint(1);
            await bind(endpoint1, coordinatorEndpoint, ['genBasic', 'genIdentify','genOnOff', 'genLevelCtrl', 'lightingColorCtrl
            await configureReporting.onOff(endpoint1);
            await configureReporting.batteryVoltage(endpoint1);
        //  await configureReporting.genLevelCtrl(endpoint1);
            const endpoint242 = device.getEndpoint(242);
            await bind(endpoint242, coordinatorEndpoint, ['greenPower']);
        },
    },
```

ashp8i, 2020-12-21:
> Hi @ashp8i, thanks for your work on this device so far. Have you done anything with the converters yet?
>
> The device does not support the batteryPercentageRemaining attribute, you need to use [...]

Hi fawksie,

Many Thanks for your help glad that you got the battery working good news. But upon close discussion with rest of the users at home we have decided this endevour is bit useless and not worth the effort:-

Let me highlight a few issues: -

this device is too noisey zigbee wise too many reporting attributes the major one being OTA that must be set to ignore

on / off works great on it’s own with an aurora hub but as soon as you turn the dial you have to wait before you can turn the light off

none of my other zigbee wireless remotes have this issue

also zigbee messaging is very sporadic and inconsistent too many attributes duplicated etc

If you could over come these issues and get the zigbee communication to be smooth I may chose to keep the remotes and not sell them on

also I’m looking to get these integrated with controllerx 
https://xaviml.github.io/controllerx/
but they are not as smooth as my ikea remote

let me know what you think also do you have the diff file so I can compare changes then there may be hope

just a heads up not a fan of Aurora Lighting seem to be very close source and not reliable and do not post anything about public ota updates.

long story short no ota update for the rgbw gu10 or the rotary remote

ashp8i, 2020-12-21:
The existing converters seem to be ok also sniffing signals from original aurora hub to figure how the colour temp / colourxy and scene controls are enabled

So this remote is set to use brightness_step color_tenperature_step colorxy and scene functions and ota then reporting on ota, battery, genlevel, color temp, colorxy and scene values

fawksie, 2020-12-24:
Hi @ashp8i, have just got the device fully set up in OpenHAB and can confirm your finding that it is utter scrap. When turning the dial, after about 10 clicks the device stops sending action messages for several seconds and sends announce messages instead, then eventually it catches up and spams the rest of the action messages that it should have sent before. It is absolutely unusable.

ashp8i, 2020-12-27:
at least we can agree the remotes are not that great if you compare it to let's say https://auroralighting.com/gb/ProductDetail/AU-A1ZB2WDM solid dimmer in my opinion, shame the wireless one is not that great, my issue could be signal related as this is a battery device it does not feature powerful zigbee radio and is not as reactive as my aqara or ikea remotes, If I could get my hands on some firmware updates I would be happy to disassemble the code to see what is going on. but that is highly unlikely so gonna try to return these

closed 2021-02-04 by github-actions[bot]