Koenkk/zigbee2mqtt #5177: HA config.json not match regular expression edge disappear on store

shirou93, 2020-12-03:
Zigbee2mqtt edge disappear on store.

Logs `WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/7ad98f9c/zigbee2mqtt/config.json: does not match regular expression for dictionary value @ data['schema']['frontend']['host']. Got 'string?'
20-12-03 05:15:49 INFO (MainThread) `

`WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/7ad98f9c/zigbee2mqtt-edge/config.json: does not match regular expression for dictionary value @ data['schema']['frontend']['host']. Got 'string?'`

LouiS22, 2020-12-03:
This has nothing to do with zigbee2mqtt, and it's resolved by now by the author of the plugin.

shirou93, 2020-12-03:
> This has nothing to do with zigbee2mqtt, and it's resolved by now by the author of the plugin.

Still not appear and warnings on loga.

Koenkk, 2020-12-03:
Addon issues should go here: https://github.com/danielwelch/hassio-zigbee2mqtt/issues

closed 2020-12-03 by Koenkk