Koenkk/zigbee2mqtt #5178: Silvercrest HG06338 not recognized

keteflips, 2020-12-03:
I have a Silvercrest HG06338 from LIDL.
My zigbee2mqtt exposes as Unsupported.
I have the last Z2M version.

[image]

Morphy99, 2020-12-07:
I have the same problem.
Using CC2530 coordinator on latest stable:
zigbee2mqtt version: 1.16.2 commit: 04c15f7
coordinator type: zStack12
coordinator revision: 20190619

The HG06338 is the -BS version as I'm in UK.

klumpler, 2020-12-07:
Same here on version 1.16.2.2 But it used to work before on version 1.16.1. even though it was recognized as a different manufacturer, not Silvercrest/Lidl, but TuYa.

Morphy99, 2020-12-07:
I beleive this is due to regional variants of the strip having a different fingerprint. I've opened a [PR ](https://github.com/xpavli44/zigbee-herdsman-converters/pull/1/commits)for mine (UK version).
If you look at the debug log when joining the "manufacturerName:" was "'_TZ3000_vmpbygs5"

closed 2021-01-15 by github-actions[bot]