Koenkk/zigbee2mqtt #5179: Silvercrest HG06335 not pairing...

njordan77, 2020-12-03:
updated recently to .16.2 and was hoping that the changelog would allow to add the motion sensor from Silvercrest/LIDL.

Did insert BATTERY, which starts pairing mode, config is set to TRUE....but nothing shows up in the Logs...
tried several times now.

Any idea?

I'm on the Jun'19 firmware with a standard stick in a Rpi4 machine/buster...tradfri, hue works fine

salopette, 2020-12-03:
I'm using the current dev branch, it works!

Fridolin34, 2020-12-03:
Not working for me either - added the following lines to the closed issue:
"Sorry, but this is still not working for me. Updated everything, tried stable and edge version, but it simply doesn't work.

I've bought a couple of Silvercrest products today and although the icons shown in the list/map is not making sense, all other devices have been nicely recognized and can be managed without any problems.

Just the motion sensor is not working."

njordan77, 2020-12-03:
hm, i guess i'm on stable - and at least i thought the changelog with new silvercrest devices was related to stable which i updated.
definitely nothing happens if you try to add this item.
Thanks
Norbert

chris1705, 2020-12-04:
Hey Norbert,

did you push the small pairing button on the side by using the provided tool? Inserting batteries wasn't enough in my case. It was necessary to press the pairing button for 2 seconds.

Regards
Chris

njordan77, 2020-12-04:
JESUS - i have seen this tiny hole yesterday but did not try as in the backside there is a visible button and i was sure that this one would be the one and it would not make sense to have another one. There was no manual included so i checked the internet. but there was no PDF available. After your input i tried a tiny nail and it feels like a button inside. did wait 2 secs and GREAT it works. Sorry so much and thanks for the help!!!

closed 2020-12-04 by njordan77

chris1705, 2020-12-04:
Great to hear it works!

When you take out everything out of the box, there is a manual, two screws and a tool for pressing the pairing button.

BTW the button on the backside is for tamper detection.

Fridolin34, 2020-12-04:
@chris1705 Thank you very much. I don't know how often I pressed the physical button and the blinking orange light tricked me to think it would try to find the coordinator. Now it works just fine.