Koenkk/zigbee2mqtt #5180: Livolo Zigbee Blind Motor

milobt, 2020-12-03:
## Information about the device + link
Livolo Zigbee version blind motor
[https://www.aliexpress.com/item/1005001702490864.html?spm=a2g0s.9042311.0.0.62474c4dXttyJL](https://www.aliexpress.com/item/1005001702490864.html?spm=a2g0s.9042311.0.0.62474c4dXttyJL)

Webpage says this device requires the new version of their zigbee gateway; whether this is true or they are trying to up-sell their gateway:
[https://www.aliexpress.com/item/4000349261585.html?spm=a2g0o.detail.1000023.1.3b9c57b31gKMbn](https://www.aliexpress.com/item/4000349261585.html?spm=a2g0o.detail.1000023.1.3b9c57b31gKMbn)

## data/database.db entry of the device
{"id":14,"type":"Unknown","ieeeAddr":"0x680ae2fffe373074","nwkAddr":5137,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":47}}},"binds":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1606851180760}

d-edge82, 2020-12-23:
Why is this closed? does this mean it's added? because it doesn't work for me..

milobt, 2020-12-23:
I opened and then closed it because I tried to pair these LIVOLO blind motors and realised that they weren’t being picked up by Z2M and that database entry was actually for some other device that tried to join my network.

Livolo state that you need to use their particular gateway, rather than the standard Tuya gateway, which leads me to think that they have altered the protocol in some way, limiting their use with their gateway only or that they may actually be using bluetooth after all - I’m no expert, just my assumption. 

As a result, I am returning them and buying Tuya branded ones instead to pair with the Tuya gateway (for now), as this is ~£16 from Amazon, rather than ~ £35+ with only Livolo compatibility.

Hopefully someone can get the Livolo ones working. 
I will comment again, once I’ve received my Tuya motors and tried pairing them with Z2M.

d-edge82, 2020-12-23:
OK this I understand, as I am having the same issue now.
Mine's from aliexpress so I cannot return it, It will just cost too much.

My question is, why close this issue then?
Other people have this issue, will it be handled when closed?

If it's something that I have missed, please excuse me..

milobt, 2020-12-23:
I bought mine from AliExpress also. 
They offer free returns on lots of items now. 
Check the item. Open a dispute and return with refund.
Gets sent back via AliExpress warehouse.

I closed it, as the database entry was not related and without being able to get it to be seen by Z2M, I’m not sure where to go with it. 

If you can get some where with the Livolo motor then post it here for others to help.

d-edge82, 2020-12-23:
I have it, already discovered by HA through Z2m.
My problem is that the Switches being created are not working.

It is being detected as this one:
https://www.zigbee2mqtt.io/devices/TI0001.html
and the switches do not work,

Should I just open a new request?

d-edge82, 2020-12-23:
Thanks! I see you reopened it.

I found it's info in the database.db file:

{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b002174ef0c","nwkAddr":39668,"manufId":0,"manufName":"LIVOLO\u0000\u0010TI0001  ","powerSource":"Mains (single phase)","modelId":"TI0001          ","epList":[6],"endpoints":{"6":{"profId":260,"epId":6,"devId":0,"inClusterList":[0,3],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"TI0001          ","manufacturerName":"LIVOLO\u0000\u0010TI0001  ","powerSource":1,"zclVersion":0,"hwVersion":0,"dateCode":"20060831        "}}},"binds":[],"meta":{}}},"hwVersion":0,"dateCode":"20060831        ","zclVersion":0,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1608745846935}

@Koenkk  please advise, how can this be added as a cover?

milobt, 2020-12-24:
What did you do to put yours in bind mode?
Instructions just say hold down and settings button for three seconds, until the led flashes blue/red. 
This happens, but don’t get a seuccessful interview and can’t see a database entry for Livolo device.

d-edge82, 2020-12-24:
Follow the notes on this page:
https://www.zigbee2mqtt.io/devices/TI0001.html

This allowed me to detect the device, but unfortunately it still doesn't work

milobt, 2021-01-06:
Have now received my Tuya branded blind motors - - and can confirm that they do pair successfully with Z2M albeit as an unrecognised variant of device: TS0601.

There's already a supported curtain motor from Moes using the same device id: TS0601, so this configuration could surely be used for the Tuya blind motor.

I am going to open a new issue for the Tuya blind motor variant.

Do you want this issue left open to pursue?
I'd advise just returning the Livolo ones and purchasing the Tuya variants instead - unless you want to use the Livolo hub or jump through all the hoops required to pair with Z2M.

milobt, 2021-01-11:
Closing as support has been added for Tuya branded version - Livolo version requires specific channel and network ID, making them incompatible with a standard Z2M setup.

See: [https://www.zigbee2mqtt.io/devices/TI0001.html](https://www.zigbee2mqtt.io/devices/TI0001.html)

closed 2021-01-11 by milobt

grinco, 2021-05-12:
Re-opened the issue here: https://github.com/Koenkk/zigbee2mqtt/issues/7444