Koenkk/zigbee2mqtt #5181: OSRAM AC03648 + zigbee2mqtt pairing failed

davisdmg, 2020-12-03:
Hi!

I wasnt unsure where to post this. I got this light that it is supposed that it is supported, however once I pair it, I get "unsupported".

Only thing that works is to use it as "router"

Zigbee Id is 0x7cb03eaa00ac7ec1.
Address on Zigbee Network is 0xDC47.
Link quality is unknown/255.
Model is unknown by unknown.
Hardware version is unknown.
Firmware version is unknown.
This device type is Router

I already reset it several times. I also tried to add it manually as the documentation says:
https://www.zigbee2mqtt.io/devices/AC03648.html

Zigbee2MQTT version: Zigbee2mqtt 1.16.2.2
Adapter hardware: CC2531
Adapter firmware version: Latest (installed 2 weeks ago)

Am I doing something wrong?

Thanks for taking your time with this :)

luchtgitaar, 2020-12-20:
I have the same problem.

Buva79, 2021-01-10:
Did you try updating everything to latest version?

davisdmg, 2021-01-10:
I did. Nothing changed :( (Updated everything today again).

pixeldoc2000, 2021-02-05:
Look at https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Post your database.db entry for the Device.

Does the database.db record for the Device have `"interviewCompleted":true` ?

If not try to bring the Device closer to your Coordinator or maybe try a Zigbee Repeater nearby.

closed 2021-03-14 by github-actions[bot]