Koenkk/zigbee2mqtt #5182: Sporadic freezes with CC2652

cody82, 2020-12-03:
Since I upgraded from CC2531 to ZZH my problem is that sometimes zigbee2mqtt completely stops working. It looks like the stick stops responding. To fix it, I have to unplug and replug it. Restarting zigbee2mqtt does not help as it will get a timeout when connecting to ZZH.

Here is a log file: https://gist.github.com/cody82/38b505a1e049a098f450c460b17c2ded

Versions:
```
info  2020-11-23 21:40:24: Starting zigbee2mqtt version 1.14.2 (commit #b91b456)
info  2020-11-23 21:40:24: Starting zigbee-herdsman...
info  2020-11-23 21:40:27: zigbee-herdsman started
info  2020-11-23 21:40:27: Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200805}}'
```

To me this looks like a ZZH firmware crash. It doesnt happen very often, maybe once in 2 weeks. Are there any known stability problems with zigbee2mqtt or ZZH?

Koenkk, 2020-12-03:
First update to the latest zigbee2mqtt version (1.16.2) and firmware (https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin)

cody82, 2020-12-11:
I upgraded firmware and zigbee2mqtt to the newest versions 5 days ago and have no problem so far.
But I have one more question:
The zigbee2mqtt log only shows messages it publishes to mqtt, so thats everything it receives over zigbee. Is it possible to also log everything it sends? I have some wierd behaviour where lights switch on during night without a reason. May be unrelated to the update though.

Koenkk, 2020-12-11:
@cody82 you can enable the debug logging which will show you the received MQTT messages. 

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Since the issue seems to be solved I will close this for now but feel free to re-open if it reappears.

closed 2020-12-11 by Koenkk