Koenkk/zigbee2mqtt #5185: Invalid system modes/presets in TuYa thermostatic valve

drzony, 2020-12-03:
Since #3821 became completely unreadable, I'm creating a new issue for this case.

Proposal of solution is in my fork https://github.com/drzony/zigbee-herdsman-converters/tree/tuya-thermostat-fixes

@Mickey016 started testing the changes and they seem fin. I'll rebase my changes on latest dev and make a PR

drzony, 2020-12-03:
@Mickey016 Can you give some status update about your testing?

drzony, 2020-12-03:
@slashbv @shirou93 Would it be possible for you to get zigbee-herdsman-converters from my fork and tell me whether the HA logs are gone?

akramarz, 2020-12-03:
Hey, I'm trying to connect to SEA802 valve and after after joining it to network I see a lot of `No converter available` messages, e.g.:
```
Zigbee2MQTT:debug 2020-12-03 21:51:36: Received Zigbee message from 'Radiator_Valve_Bedroom', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0,0,0,205],"type":"Buffer"},"dp":614,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
```
This one stands for the temperature read by valve's internal sensor. I think this PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/1742 changed the `dp` codes.

drzony, 2020-12-03:
@akramarz This issue is not about Saswell thermostats. You are using old version of zigbee-herdsman-converter. Please try latest dev version of zigbee2mqtt

Mickey016, 2020-12-04:
> @Mickey016 Can you give some status update about your testing?

Sorry, I've been busy lately so I didn't have time to test a lot.
Looks like the "time" problem is stil here for me. Maybe I need to update my Z2M with your latest changes @drzony ?
Otherwise, I did not encounter bugs with my (basic) usage.

drzony, 2020-12-04:
@Mickey016 can you try to replace `onEvent: tuya.setLocalTime,` with `onEvent: tuya.setTime,` in `zigbee-herdsman-converters/devices.js` (around line 1737 under `TS0601_thermostat`) and try again?

closed 2020-12-04 by Koenkk

slashbv, 2020-12-04:
@drzony  looks much better, thank you very much

Mickey016, 2020-12-04:
> @Mickey016 can you try to replace `onEvent: tuya.setLocalTime,` with `onEvent: tuya.setTime,` in `zigbee-herdsman-converters/devices.js` (around line 1737 under `TS0601_thermostat`) and try again?

I tried that and found something I didn't know. For my testings, I use TRVs that have already been paired with the add-on version of Z2M in HA. It appears that they need to be paired again with the new Z2M in order to solve the "time" problem. In fact, no need to change devices.js. Tested with two TRV and it works. Think the issue was on my end...

Thanks for the hard work @drzony

akramarz, 2020-12-04:
> @akramarz This issue is not about Saswell thermostats. You are using old version of zigbee-herdsman-converter. Please try latest dev version of zigbee2mqtt

omg, my bad, thank you 🙂