Koenkk/zigbee2mqtt #5186: Lonsonho dimmer not responding to setting the brightness_l2 property

itavero, 2020-12-03:
## What happened
Trying to control a [Lonsonho QS-Zigbee-D02-TRIAC-2C-LN](https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-2C-LN.html) by publishing a message to the `friendly_name/set` topic with `{"brightness_l2": 50}`.
Both lights controlled by the dimmer were off and stayed off after sending this command.
I expected one of them to turn on (dimmed to about 20%).

It does work if I publish `{"brightness":50}` to `friendly_name/l2/set`.

Also tested with `l1` instead of `l2`, providing similar results.

## What did you expect to happen
The light would turn on at a dimmed level with the first message mentioned, as this property (`brightness_l2`) is also the one mentioned in the `exposes` information:
```json
{
  "endpoint":"l2",
  "features":[
    {
      "access":7,
      "description":"On/off state of this light",
      "endpoint":"l2",
      "name":"state",
      "property":"state_l2",
      "type":"binary",
      "value_off":"OFF",
      "value_on":"ON",
      "value_toggle":"TOGGLE"
    },
    {
      "access":7,
      "description":"Brightness of this light",
      "endpoint":"l2",
      "name":"brightness",
      "property":"brightness_l2",
      "type":"numeric",
      "value_max":254,
      "value_min":0
    }
  ],
  "type":"light"
}
```

## How to reproduce it (minimal and precise)
* Send `{"brightness_l2": 50}` to `friendly_name/set`: *Fail*
* Send `{"brightness":50}` to `friendly_name/l2/set`: *Succeeds*

## Debug info
Zigbee2MQTT version: 1.16.1-dev (26841f5), but also noticed in previous official releases
Adapter hardware: LAUNCHXL-CC26X2R1
Adapter firmware version: 
```json
  "coordinator": {
      "meta": {
         "maintrel": 1,
         "majorrel": 2,
         "minorrel": 7,
         "product": 1,
         "revision": 20201026,
         "transportrev": 2
      },
      "type": "zStack3x0"
   }
```

Koenkk, 2020-12-04:
Fixed, thanks for reporting.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-04 by Koenkk

adipierro, 2022-08-11:
Hi! This still happens to [QS-Zigbee-D02-TRIAC-2C-L](https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-2C-L.html#lonsonho-qs-zigbee-d02-triac-2c-l).

yasapl, 2022-10-15:
I can also confirm there are issues with Lonsonho dimmer QS-Zigbee-D02-TRIAC-2C-L
Switching on from Z2M randromly works. It is usually possible to switch it off but rarely on. When switch it on in z2m, it will show on but lights will not come up. Refreshing switch status in z2m will flopp it back to off position. Then, turning light with the physical button will make the fully dimmed which in my case mean they don’t lit at all, only in z2m I can see they are on and brightness is set to 0. They worked fine for a couple of months and only recently started “playinig” with me. I have two of these dimmers and both are with the same issues.