Koenkk/zigbee2mqtt #5188: z2m will try to publish before MQTT is conencted

sjorge, 2020-12-03:
## What happened

```
Zigbee2MQTT:error 2020-12-03 21:08:11: Not connected to MQTT server!
Zigbee2MQTT:error 2020-12-03 21:08:11: Cannot send message: topic: 'zigbee2mqtt/entrance/motion', payload: '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":55,"occupancy":false,"temperature":18.91,
"update":{"state":"idle"}}
Zigbee2MQTT:error 2020-12-03 21:08:11: Not connected to MQTT server!
Zigbee2MQTT:error 2020-12-03 21:08:11: Cannot send message: topic: 'zigbee2mqtt/entrance/motion/availability', payload: 'online
Zigbee2MQTT:info  2020-12-03 21:08:11: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:info  2020-12-03 21:08:11: Connected to MQTT server
Zigbee2MQTT:info  2020-12-03 21:08:11: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
```

Restarted z2m and noticed this, could not re-trigger so I guess timing is important here

## What did you expect to happen
Either one of these
- z2m to ignore the message (and optionally log that it did)
- z2m to buffer the message and publish once MQTT is conencted

## How to reproduce it (minimal and precise)
??? luck I think, a device needs to send a message at just the right time.

## Debug info
Zigbee2MQTT version: 1.16.2-dev (commit #be61249)
Adapter hardware: CC2531
Adapter firmware version: n/a

Koenkk, 2020-12-04:
> z2m to ignore the message (and optionally log that it did)

It already does this right? (but maybe the logging message is confusing?)

We probably don't want to queue messages, if e.g. `{"action": "single"}` get's queued it can cause unexpected behaviour.

sjorge, 2020-12-04:
In this case the message was lost as z2m posted the data from state.json a few seconds later. 

This error is the edge case where we’re already receiving messages from the stick but have not yet setup mqtt/loaded the state.json (I believe, it very hard to trigger)

sjorge, 2020-12-04:
I guess the real fix here is make sure we have read state.json and (try) to publish that to mqtt before we start accepting messages from the stick at startup. That way even if we don’t have mqtt connection the state should end up in the next write to state.json atleast

Koenkk, 2020-12-04:
The messages that were not published due to "Not connected to MQTT server" still end up in the state so they will be published when the MQTT connection is established (if you don't disable it via `cache_state_send_on_startup`).

sjorge, 2020-12-04:
Hmm I have that enable but the state it published a few seconds laters was the old state.

Already out my log retention though, I just checked.

~ sjorge

closed 2021-01-11 by github-actions[bot]