Koenkk/zigbee2mqtt #5191: zigbee2mqtt/bridge/config/devices and .../groups lost

fnfid, 2020-12-04:
At the moment i do not get either device list nor group list the easy way by zigbee2mqtt/bridge/config/devices  or ...groups. This stopped working in master 1.16.2 (and in develeoper tree some week ago). This may be not a part of legacy to get rid of as its still in the documentation, and its a helpful thing.

How to reproduce it (minimal and precise)

send whatever payload to zigbee2mqtt/bridge/config/devices and listen to zigbee2mqtt/bridge/log with unfortunally no answer, .../groups the same

 Debug info
Zigbee2MQTT version: 1.16.2 (master) or latest dev same
Adapter hardware: CC1352P-2
Adapter firmware version: 20201026

Koenkk, 2020-12-04:
can you provide your configuration.yaml?

fnfid, 2020-12-04:
Shure, things haven't been changed here for a while: (sorry for pasting)

homeassistant: false
permit_join: false
experimental:
  new_api: true
frontend:
  port: 8080
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost:1883'
  include_device_information: true
serial:
  port: /dev/ttyACM0
advanced:
  pan_id: 6754
  ext_pan_id:
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
  channel: 11
  cache_state: true
  report: true
  availability_timeout: 0
  log_level: debug
  log_directory: /media/pi/SICHERUNG/Log/%TIMESTAMP%
  log_file: log.txt
devices:
  '0xec1bbdfffe19af4b':
    friendly_name: Wohn_Fern
  '0x000d6ffffe5b0bde':
    friendly_name: Master_Fern
  '0xec1bbdfffe0242a0':
    friendly_name: Schlaf_Fern
  '0xec1bbdfffe1a8073':
    friendly_name: Ess_Fern
  '0xec1bbdfffe181510':
    friendly_name: Oben_Fern
  '0xec1bbdfffea7c6c3':
    friendly_name: Treppe_Fern
  '0x680ae2fffe10c871':
    friendly_name: Wecker_Dim
  '0xec1bbdfffea17ea6':
    friendly_name: Kaffee_2
  '0xec1bbdfffe3dbe33':
    friendly_name: Motion_Keller_1
  '0x00178801089cfba0':
    friendly_name: Kelleru_Dose
  '0x000d6ffffe6366f5':
    friendly_name: Esslampe
  '0xccccccfffebcd62e':
    friendly_name: Kellerlampe
  '0xccccccfffe8d854a':
    friendly_name: Buerolampe1
  '0x14b457fffe399b4b':
    friendly_name: '0x14b457fffe399b4b'
  '0xccccccfffec8a8cf':
    friendly_name: Obenlampe1
  '0x000d6ffffe6163a5':
    friendly_name: Obenlampe2
  '0xccccccfffedf659c':
    friendly_name: Obenlampe3
  '0xec1bbdfffe36185d':
    friendly_name: Obenlampe4
  '0xccccccfffe3ef247':
[... 195 lines cut ...]
      - 0x680ae2fffeb14132/1
  '8':
    friendly_name: Schreibtisch
    devices:
      - 0x000d6ffffe12ed3b/1
  '9':
    friendly_name: Bado
    devices:
      - 0x680ae2fffebeb1aa/1
      - 0x680ae2fffebf4675/1
      - 0x680ae2fffea66ceb/1
      - 0x680ae2fffebf4c3f/1
      - 0x680ae2fffea66b83/1
  '10':
    friendly_name: Küchenunter
    devices:
      - 0x680ae2fffeacae19/1
  '11':
    friendly_name: Essunter
    devices:
      - 0x588e81fffe22be49/1
  '12':
    friendly_name: Mixiunter
    devices:
      - 0xec1bbdfffeb1d1de/1

Koenkk, 2020-12-04:
Can you provide the debug log of starting zigbee2mqtt?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

fnfid, 2020-12-04:
Here's the logfile from this morning, rest of the code is running extremely well:-)

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5643633/log.txt)

Koenkk, 2020-12-04:
To get the list your first have to publish to `zigbee2mqtt/bridge/config/devices/get` (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigdevicesget). I don't see this in your log

fnfid, 2020-12-04:
thats correct. Formerly i send an empty payload to zigbee/bridge/config/devices and get the answer from ../bridge/config, this does not work anymore.
Your solution works fine! Thanks a lot!
The old version is still documented in the mqtt_topics_and_message_structure.html , a few lines above, and may be removed as legacy not working anymore. Its in the zigbee/bridge/log section found near "list of all devices".
Thanks!

Koenkk, 2020-12-05:
> Formerly i send an empty payload to zigbee/bridge/config/devices and get the answer from ../bridge/config, this does not work anymore.

The documentation is wrong on this, this was never intended to work.

Anyway with the next release a new api will be released (already enabled by default in 1.16.2-dev)

fnfid, 2020-12-08:
Use this already:-)
Still for the records: Recent way to get device list is perfect, still groups list my be done same way similar to zigbee2mqtt/bridge/config/groups/get instead of empzy payload to zigbee2mqtt/bridge/config/groups (as this works so far this way only)
This can be closed, Thank for the support!

closed 2020-12-08 by fnfid