Koenkk/zigbee2mqtt #5192: BlitzWolf BW-SHP13 not pairing, interview_failed

diotta88, 2020-12-04:
Hi
I have a problem with my 2 BlitzWolf BW-SHP13
I can not pair them, and always failed interview.
the devices still appear to be registered to the network.
in fact it can repeat the signal, allowing the connection of distant devices.

when i try to join i have in log:
info 2020-12-02 23:20:26: Device 'smartPresaTuYa2' joined
info 2020-12-02 23:20:26: Starting interview of 'smartPresaTuYa2'
info 2020-12-02 23:20:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"smartPresaTuYa2"},"type":"device_connected"}'
info 2020-12-02 23:20:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"smartPresaTuYa2"},"type":"pairing"}'
error 2020-12-02 23:21:52: Failed to ping 'smartPresaTuYa2'
error 2020-12-02 23:22:44: Failed to interview 'smartPresaTuYa2', device has not successfully been paired
info 2020-12-02 23:22:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"smartPresaTuYa2"},"type":"pairing"}'
info 2020-12-02 23:22:44: MQTT publish: topic 'zigbee2mqtt/smartPresaTuYa2/availability', payload 'online'

when i reboot the service:
Dec 04 09:26:10 SRV-CASA bash[16970]: Zigbee2MQTT:info  2020-12-04 09:26:10: 0x588e81fffeff60ff (0x588e81fffeff60ff): Not supported (Unknown)

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC2531
Adapter firmware version: 20190608

sneerix, 2020-12-06:
I have the same issue with BW-SHP13. I was able to pair it once, but after that I was unable to pair again.

Zigbee2mqtt: 1.16.2.2 and latest (devel) as of 2020-12-06
Adapter hardware: zStack30x - cc2538+cc2592
Adapter firmware version: 20200211

closed 2021-01-13 by github-actions[bot]

raffalbednarz, 2021-03-07:
i have te same problem it was working 2 months then i want to relocate plug in other room  and i can't reconnect to HA :(

21-03-07 10:22:21 WARNING (MainThread) [supervisor.addons.options] Unknown option 'initialdelay' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:22:21 WARNING (MainThread) [supervisor.addons.options] Unknown option 'new_api' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:22:21 WARNING (MainThread) [supervisor.addons.options] Unknown option 'restartdelay' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:22:21 WARNING (MainThread) [supervisor.addons.options] Unknown option 'initialdelay' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:22:21 WARNING (MainThread) [supervisor.addons.options] Unknown option 'new_api' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:22:21 WARNING (MainThread) [supervisor.addons.options] Unknown option 'restartdelay' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:22:21 WARNING (MainThread) [supervisor.addons.options] Unknown option 'initialdelay' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:22:21 WARNING (MainThread) [supervisor.addons.options] Unknown option 'new_api' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:22:21 WARNING (MainThread) [supervisor.addons.options] Unknown option 'restartdelay' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:22:21 WARNING (MainThread) [supervisor.addons.options] Unknown option 'initialdelay' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:24:54 WARNING (MainThread) [supervisor.addons.options] Unknown option 'new_api' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:24:54 WARNING (MainThread) [supervisor.addons.options] Unknown option 'restartdelay' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:24:54 WARNING (MainThread) [supervisor.addons.options] Unknown option 'initialdelay' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:24:54 WARNING (MainThread) [supervisor.addons.options] Unknown option 'new_api' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:24:54 WARNING (MainThread) [supervisor.addons.options] Unknown option 'restartdelay' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:24:54 WARNING (MainThread) [supervisor.addons.options] Unknown option 'initialdelay' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:24:54 WARNING (MainThread) [supervisor.addons.options] Unknown option 'new_api' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)
21-03-07 10:24:54 WARNING (MainThread) [supervisor.addons.options] Unknown option 'restartdelay' for Zigbee2mqtt (7ad98f9c_zigbee2mqtt)

viannale, 2021-03-16:
I have the same problem.

raffalbednarz, 2021-03-16:
As the errors show, you have some options that got removed some time ago:

restartdelay, initialdelay and new_api

Please remove those from your config and z2m should start.

[https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/55](url)

mateuszpodlaski, 2021-08-15:
I have a similar problem with that device (BW-SHP13). I never was able to pair it properly. Tried multiple times and very close to the dongle. Openhab3.1. 

It always show: 
(0xcc86ecfffe4eaf2c): Not supported (Unknown)

Starting Zigbee2MQTT version 1.21.0 (commit #70891eec)
Starting zigbee-herdsman (0.13.124)

Any ideas/advice?
Thanks!