Koenkk/zigbee2mqtt #5193: Devices do not connecti to router

domosgabor, 2020-12-04:
Hello,

I have 50 devices in my network, 1 coordinator with  source routing firmware and 3 CC2530 routers. If you check the attached screenshot you can notice, that devices connects to the upstairs and downstairs router, and office router does not have any connected devices. I added the office router some days ago to the network and flashed the latest firmware. Can it be an issue that other routers/coordinator has different firmware version?
In the log, there are some entries from the office router, so I suppose that it's working correctly,
INFO: MQTT publish: topic 'zigbee2mqtt/router.office', payload '{"description":"54747/0xD0CF5EFFFEF91078","led":true,"linkquality":65,"rssi":26,"type":"PARENT"}'
INFO: MQTT publish: topic 'zigbee2mqtt/router.office', payload '{"description":"54747/0xD0CF5EFFFEF91078","led":true,"linkquality":65,"rssi":27,"type":"PARENT"}'
INFO: MQTT publish: topic 'zigbee2mqtt/router.office', payload '{"description":"10976/0xD0CF5EFFFEF2A5BF","led":true,"linkquality":65,"rssi":59,"type":"NEIGHBOR"}'

[image]

skumka, 2020-12-10:
I think this could be related to another router issue:
https://github.com/Koenkk/zigbee2mqtt/issues/5247

@Koenkk I am sure some of the recent changes all routers problems with discovery, joining devices in the network and random disconnection of devices from routers.

Koenkk, 2020-12-10:
Zigbee2MQTT has no influence on how devices connect to each other. Since it's not directly a bug in Zigbee2MQTT I will close this and recommend to start a discussion instead: https://github.com/Koenkk/zigbee2mqtt/discussions

closed 2020-12-10 by Koenkk