Koenkk/zigbee2mqtt #5194: Update OTA Philips hue dimmer to same version

Bastiencc, 2020-12-04:
## What happened
OTA is available for my philips hue dimmer switch (324131092621), but update to same version

```
Dec 04 09:49:54 raspberrypi npm[30737]: Zigbee2MQTT:info  2020-12-04 09:49:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'chambre_1_switch_mur' at 99.29%, +- 0 minutes remaining","meta":{"device":"chambre_1_switch_mur","progress":99.29,"status":"update_progress"},"type":"ota_update"}'

Dec 04 09:51:42 raspberrypi npm[30737]: Zigbee2MQTT:info  2020-12-04 09:51:42: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"battery":26.5,"linkquality":52,"update":{"progress":100,"remaining":19,"state":"updating"},"update_available":false}'
Dec 04 09:51:42 raspberrypi npm[30737]: Zigbee2MQTT:info  2020-12-04 09:51:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'chambre_1_switch_mur' at 100.00%","meta":{"device":"chambre_1_switch_mur","progress":100,"status":"update_progress"},"type":"ota_update"}'

Dec 04 09:51:43 raspberrypi npm[30737]: Zigbee2MQTT:info  2020-12-04 09:51:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"id":"chambre_1_switch_mur"},"meta":{"device":"chambre_1_switch_mur","from":{"date_code":"20160302","software_build_id":"5.45.1.17846"},"status":"update_succeeded","to":{"date_code":"20160302","software_build_id":"5.45.1.17846"}},"type":"ota_update"}' 

```

## What did you expect to happen
Update to other version

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC2531
Adapter firmware version: 20190608

kevincaradant, 2020-12-06:
Hi

I think it's a duplicate. 

it's already in discussion here :)
#4559

Bastiencc, 2020-12-06:
> Hi
> 
> I think it's a duplicate. 
> 
> it's already in discussion here :)
> #4559

Thanks but I'm not sure it's the same bug , because my switch is not already up-to-date

kevincaradant, 2020-12-06:
> > Hi
> > I think it's a duplicate.
> > it's already in discussion here :)
> > #4559
> 
> Thanks but I'm not sure it's the same bug , because my switch is not already up-to-date

Oh sorry I didn't understand that. OK my bad :)

chrissv, 2021-01-03:
The same situation happens with my Hue dimmer - including the same FW version numbers.

sjorge, 2021-01-03:
Have you tried removing and repairing the dimmer? I know for my ikea remote sometimes the device is already sleeping when z2m reads the version again after the update. And it 'stuck' on the old version in the database.

But the device did get updated, after a remove/pair it shows up correctly. Not sure the same applies to the hue dimmer. I have one but it is already up to date.

Sennevds, 2021-01-05:
I have exact the same issue. But I have had the problem for several months now:
```log
info  2021-01-05 12:31:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Hue dimmer' at 100.00%","meta":{"device":"Hue dimmer","progress":100,"status":"update_progress"},"type":"ota_update"}'
info  2021-01-05 12:31:48: Finished update of 'Hue dimmer', from '{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"}' to '{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"}'
```

Sennevds, 2021-01-05:
In the front end you can check it

grubyak, 2021-01-10:
i am having same issue as reporter

grubyak, 2021-01-10:
i tried to update my dimmer and it failed, apparently it turns into a deep sleep due to inactivity at one point so that no communication is possible:

```
Zigbee2MQTT:info  2021-01-10 14:22:38: Update of 'sensor/switch/hue-dimmer' at 90.36%, +- 4 minutes remaining
Zigbee2MQTT:info  2021-01-10 14:22:38: MQTT publish: topic 'home/sensor/switch/hue-dimmer', payload '{"battery":77,"brightness":211.1,"counter":1,"linkquality":147,"update":{"progress":90.36,"remaining":254,"state":"updating"},"update_available":false}'
Zigbee2MQTT:info  2021-01-10 14:22:38: MQTT publish: topic 'home/bridge/log', payload '{"message":"Update of 'sensor/switch/hue-dimmer' at 90.36%, +- 4 minutes remaining","meta":{"device":"sensor/switch/hue-dimmer","progress":90.36,"status":"update_progress"},"type":"ota_update"}'
(node:20428) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
(node:20428) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:20428) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
...
Zigbee2MQTT:error 2021-01-10 14:25:14: Update of 'sensor/switch/hue-dimmer' failed (Timeout: device did not request any image blocks)
(node:20428) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
(node:20428) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 2)
```

i will retry updating process and will press dimmer buttons every few minutes to prevent it from sleeping

grubyak, 2021-01-10:
when re-triggering update it jumped from 0% to 82% right away and because of that remaining update time estimation is wrong "0 minutes remaining", looks like it's delta is calculated once because further updates stay with "0 minutes" as well:

```
Zigbee2MQTT:info  2021-01-10 15:01:28: Update of 'sensor/switch/hue-dimmer' at 0.00%
Zigbee2MQTT:info  2021-01-10 15:01:28: MQTT publish: topic 'home/sensor/switch/hue-dimmer', payload '{"battery":77,"brightness":255,"counter":1,"linkquality":144,"update":{"progress":0,"state":"updating"},"update_available":false}'
Zigbee2MQTT:info  2021-01-10 15:01:28: MQTT publish: topic 'home/bridge/log', payload '{"message":"Update of 'sensor/switch/hue-dimmer' at 0.00%","meta":{"device":"sensor/switch/hue-dimmer","progress":0,"status":"update_progress"},"type":"ota_update"}'
Zigbee2MQTT:info  2021-01-10 15:01:54: MQTT publish: topic 'home/sensor/switch/hue-dimmer', payload '{"action":"up-press","battery":77,"brightness":255,"counter":1,"duration":0,"linkquality":153,"update":{"progress":0,"state":"updating"},"update_available":false}'
Zigbee2MQTT:info  2021-01-10 15:01:55: MQTT publish: topic 'home/living-room/square-light', payload '{"brightness":254,"color_temp":150,"linkquality":75,"state":"ON","update":{"state":"available"},"update_available":true}'
Zigbee2MQTT:info  2021-01-10 15:01:58: Update of 'sensor/switch/hue-dimmer' at 82.19%, +- 0 minutes remaining
...
Zigbee2MQTT:info  2021-01-10 15:08:33: MQTT publish: topic 'home/bridge/log', payload '{"message":"Update of 'sensor/switch/hue-dimmer' at 97.58%, +- 0 minutes remaining","meta":{"device":"sensor/switch/hue-dimmer","progress":97.58,"status":"update_progress"},"type":"ota_update"}'
Zigbee2MQTT:info  2021-01-10 15:09:04: Update of 'sensor/switch/hue-dimmer' at 98.77%, +- 0 minutes remaining
...
Zigbee2MQTT:info  2021-01-10 15:11:06: Update of 'sensor/switch/hue-dimmer' at 100.00%
Zigbee2MQTT:info  2021-01-10 15:11:06: MQTT publish: topic 'home/sensor/switch/hue-dimmer', payload '{"battery":77,"brightness":255,"counter":1,"linkquality":168,"update":{"progress":100,"remaining":0,"state":"updating"},"update_available":false}'
Zigbee2MQTT:info  2021-01-10 15:11:06: MQTT publish: topic 'home/bridge/log', payload '{"message":"Update of 'sensor/switch/hue-dimmer' at 100.00%","meta":{"device":"sensor/switch/hue-dimmer","progress":100,"status":"update_progress"},"type":"ota_update"}'
Zigbee2MQTT:info  2021-01-10 15:11:08: Finished update of 'sensor/switch/hue-dimmer', from '{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"}' to '{"dateCode":"20160302","softwareBuildID":"5.45.1.17846"}'
...
Zigbee2MQTT:info  2021-01-10 15:11:13: Update available for 'sensor/switch/hue-dimmer'
```

update from `5.45.1.17846` to `5.45.1.17846` finished, new update to same version is again available

cc: @Koenkk

closed 2021-02-18 by github-actions[bot]

Bastiencc, 2021-05-04:
more news about this issue?

Bastiencc, 2021-05-05:
Ok after update finished, just remove and repaire  the device and it seems that the software is actally updated (6.1.1.28573).

sjakub, 2021-06-19:
I am having the same issue, but I tried several times to update, remove (by clicking 'remove' button in the frontend) and re-pair the device, as well as resetting it. Nothing worked, the firmware version doesn't change... Is there anything else I can try?

sjakub, 2021-06-28:
I also tried replacing the battery and it didn't help. Finally I bought a Philips Hue Bridge, enabled automatic updates and in the morning the dimmer was updated. So it looks like there actually is an issue with the zigbee2mqtt update process...

SHxKM, 2021-07-02:
I found this thread while looking for a solution to what seems to be the [same issue](https://github.com/Koenkk/zigbee2mqtt/issues/7871). After the dimmer updated twice (I didn't take note of the versions it was updated from/to), I am now unable to update at all - and it seems stuck at firmware version 5.45.1.17846 (build date: 20160302). It seems that Zigbee2MQTT downgraded my version through the update mechanism because this was previously connected to a Hue bridge and was showing "up-to-date" status.

Cyriuz, 2022-01-01:
I have the same issue as you @SHxKM, did you figure out any way to solve it?

SHxKM, 2022-01-01:
> I have the same issue as you @SHxKM, did you figure out any way to solve it?

Not yet. In fact I’ve opened a separate  issue #10139 which better describes the issue.

hughb8on, 2022-02-04:
Hi,I'm adding this comment so I see any updates on this issue, and to let you know I have the same problem with my Hue v1 dimmers.

SHxKM, 2022-02-04:
> Hi,I'm adding this comment so I see any updates on this issue, and to let you know I have the same problem with my Hue v1 dimmers.

Same here. I’ve been documenting my failures to update in this issue[here](https://github.com/Koenkk/zigbee2mqtt/issues/10139#issuecomment-1029286926) .