Koenkk/zigbee2mqtt #5195: Home Assistant can't turn off lights after upgrading to 1.16.2.2

McGiverGim, 2020-12-04:
## What happened
After upgrading the Home Assistant addon from 1.16.1 to 1.16.2.2 some things stop working:
1. I can turn on the lights, but I can't turn them off again. The attributes of the light shows the switch as OFF, but the state is ON and the light remains ON. From the log:
```txt
info  2020-12-04 11:05:48: MQTT publish: topic 'zigbee2mqtt/rele_pasillo', payload '{"linkquality":73,"state_l1":"OFF"}'
```
This light is a relay that acts as light, I don't know if it matters:
```yaml
'0x60a423fffeab4e96':
  friendly_name: 'rele_pasillo'
  homeassistant:
    switch_l1:
      type: light
      object_id: light1
    switch_l2:
      type: light
      object_id: light2	  
```
2. I can't see any kind of error in the logs. 
3. I'm not using the new MQTT device trigger for Home Assistant, only the old change in the binary_sensor change.
4. After downgrading to 1.16.1, the problem seems to remain, at least partially so I reboot the raspberry and all seems to work ok again.
5. I have done the above process twice, with the same problem.

## What did you expect to happen
All continue working.

## How to reproduce it (minimal and precise)
In my case I only need to upgrade to the latest version, but I suppose there is something more because I don't see more reports.

## Debug info
Zigbee2MQTT version: 1.16.2 (Home Assistant 1.16.2.2)
Adapter hardware: CC2531
Adapter firmware version: zStack 12.20190608

Koenkk, 2020-12-04:
Can you provide the debug log when controlling the device via HA?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

McGiverGim, 2020-12-04:
Ayer reading my text, I added a complete reboot of the raspberry pi after updating and confirming the problem, and the reboot fixed it.
I don't know what the problem was, only rebooting the complete system fixed it. Usually this was not needed with other versions.
I close the issue.

closed 2020-12-04 by McGiverGim