Koenkk/zigbee2mqtt #5196: Aqara door and window sensor unresponsive

nikos1988, 2020-12-04:
## What happened
Aqara door and window sensor became unresponsive. Had to press the button for a short time on the sensor to make it work again. Aqara vibration and motion sensors in the same network kept working though.

## What did you expect to happen
Report contact true and false when in use, but nothing was registered.

## How to reproduce it (minimal and precise)
Used slaeshs CC2652RB Stick and restarted zigbee2mqtt a few times over the last months.

## Debug info
Zigbee2MQTT version: 1.16.2 and before that the first version compatible with slaeshs CC2652RB stick
Adapter hardware: slaeshs CC2652RB
Adapter firmware version: 20200805, now flashed 20201026 and see if it happens again

way2many, 2020-12-04:
Hi,
This is a known problem with Xiaomi Aqara devices. They tend to exit the network by their own. Temperature sensors have less issues as they send data regularly while contact sensors might not be used often.

I don't think a fix is possible as this is due to the end device itself.

kevincaradant, 2020-12-05:
Hi

Got the same issue recently (we are a lot of to have this issue with Z2M only so not sure it's completely the Xiaomi faults ?) 

Anyway, check that your wifi and neighbours wifi is not close or on the same channel that your Zigbee channel :)

And I don't know yet if it's useful or not but I added these parameters in `configuration.yaml` file:

```
 report: true
 availability_timeout: 60
```

And very recently I added this also:
```
# Optional: configure adapter concurrency (e.g. 2 for CC2531 or 16 for CC26X2R1) (default: null, uses recommended value)
  adapter_concurrent: 1
```

For the moment, everything is Okay. I will see in the next days / weeks :)

skumka, 2020-12-10:
+1 however problems exist also with Aqara vibration and motion sensors.
Motion Sensros are seriously affected by the recent Dev changes. They got broken every day, while before that update they were able to work more than 2 weeks and were available all the time.

kevincaradant, 2020-12-10:
For the moment, one week without any issue since I added these 3 parameters in my previous message. Maybe @skumka, you can try them and tell us if it's better.

Maybe not a final solution but at least a temporary workaround

nilsx, 2020-12-10:
@kevincaradant What does "adapter_concurrent: 1" do? And for cc2531it should be 2 according to the configuration page?

and..

report: true
 availability_timeout: 60

According to the configuration page:
# Only AC powered routers are checked for availability. (default: 0) . 

So, why would setting this option make a difference?

kevincaradant, 2020-12-10:
about `adapter_conccurency`. I prefer to link this answer which explain well what does this feature:
https://github.com/Koenkk/zigbee2mqtt/issues/5043#issuecomment-736806113
Maybe you could try with this feature. If it doesn't change anything try the 2 others. And otherwise, wait a better advice from someone else who could help you :)

About `availability_timeout: 60`, yes I thought exactly like you. Apparently it's only on AC but for a reason that I don't know. Maybe just a coincidence, but I had issues during 2 weeks (every day) ...
I have a idea, maybe wrong ..., but it could be possible that the availability_timeout wake up my AC routers and the end-devices connected on them are also wake up in consequence ? or they (end-devices) can more easily communicate with AC routers ?

`report: true` feature could help to keep wake up devices also but not sure which devices supports report at home. Maybe 0 maybe all, I don't know but I add this feature because I'm running on a enought poweful adapter so it's not an issue for me to enable (for nothing?) this feature.

It's just after to add these 3 options than everything works well but I don't know the exactly reason why.

nikos1988, 2021-01-08:
Hmm, i am not exactle sure what happened for me. I also had some trouble with my mosquitto broker and high cpu load (since version >= 2.0) and i rolled back to 1.6.9-1. It works now for at least 2 weeks without issues.

Is it possible that the mqtt broker could be the problem?

closed 2021-02-15 by github-actions[bot]