Koenkk/zigbee2mqtt #5198: Samsung SmartThings Plug (GP-WOU019BBDWG) is not reporting voltage or current

pedrolamas, 2020-12-04:
## What happened

I just got a brand-new Samsung SmartThings UK Plug (GP-WOU019BBDWG) that has energy monitoring capability, however it seems to only report power, but current and voltage are always 0.

## What did you expect to happen

To fully report power, voltage, and current values.

## How to reproduce it (minimal and precise)

I have a 50w Halogen lamp connected to this plug so I can get correct measurements, when I turn the plug on I get a power read of "50" as expected, but voltage and current are always "0".

## Debug info
Zigbee2MQTT version: 1.16.2 commit: 04c15f7
Adapter hardware: CC26X2R1
Adapter firmware version: 20201113

pedrolamas, 2020-12-04:
Full debug log (including herdsman debug): [log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5645948/log.txt)

pedrolamas, 2020-12-04:
I have a feeling this is somehow related to this: https://github.com/Koenkk/zigbee-herdsman-converters/pull/933#issuecomment-578406495

Koenkk, 2020-12-06:
Looks that the plug send new voltage/current measurements.

- Can you try to reconfigure via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure
- Are you sure that the plug supports this?

pedrolamas, 2020-12-06:
I've tried reconfigure a few times now, no errors were shown but nothing else changed.

I know that it correctly reports power every few seconds, but can't say anything about voltage or current, though it does show values on the multipliers!

Here's the current values on database.db:

```json
{
  "id": 30,
  "type": "Router",
  "ieeeAddr": "0xec1bbdfffe3dc3d7",
  "nwkAddr": 30575,
  "manufId": 4456,
  "manufName": "LDS",
  "powerSource": "Mains (single phase)",
  "modelId": "ZB-ONOFFPlug-D0005",
  "epList": [1, 242],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 81,
      "inClusterList": [0, 3, 4, 5, 6, 1794, 2820, 64642],
      "outClusterList": [3, 10, 25],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "ZB-ONOFFPlug-D0005",
            "manufacturerName": "LDS",
            "powerSource": 1,
            "zclVersion": 3,
            "appVersion": 1,
            "stackVersion": 2,
            "hwVersion": 1,
            "dateCode": "20190716-165",
            "swBuildId": "2.3"
          }
        },
        "haElectricalMeasurement": {
          "attributes": {
            "acVoltageMultiplier": 1,
            "acVoltageDivisor": 1,
            "acCurrentMultiplier": 1,
            "acCurrentDivisor": 1000,
            "acPowerMultiplier": 1,
            "acPowerDivisor": 1,
            "activePower": 0,
            "rmsCurrent": 0,
            "rmsVoltage": 0
          }
        },
        "seMetering": {
          "attributes": {
            "multiplier": 1,
            "divisor": 100,
            "currentSummDelivered": [0, 0]
          }
        },
        "genOnOff": { "attributes": { "onOff": 0 } }
      },
      "binds": [
        {
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b001ca16404",
          "endpointID": 1
        },
        {
          "cluster": 2820,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b001ca16404",
          "endpointID": 1
        },
        {
          "cluster": 1794,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b001ca16404",
          "endpointID": 1
        }
      ],
      "meta": {}
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [33],
      "clusters": {},
      "binds": [],
      "meta": {}
    }
  },
  "appVersion": 1,
  "stackVersion": 2,
  "hwVersion": 1,
  "dateCode": "20190716-165",
  "swBuildId": "2.3",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": { "reporting": 1, "configured": 3 },
  "lastSeen": 1607164074467
}
```

Maybe I should remove the entry on the converters devices.js entry and see what I get reported back?

Koenkk, 2020-12-06:
@pedrolamas I don't think that will help, you should see "Received zigbee message.. " with the voltage/current (as you see for the power) but I don't see it.

pedrolamas, 2020-12-07:
I see now that this issue was reported before (it's the same plug!): https://github.com/Koenkk/zigbee-herdsman-converters/issues/1587

pedrolamas, 2020-12-07:
I've found [this groovy script](https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/zigbee-metering-plug.src/zigbee-metering-plug.groovy) on the official Smartthings community repo, and I can see there is no reference to the voltage or current readings, so I guess that the plug might just not support it...

Koenkk, 2020-12-07:
@pedrolamas agree, can you make a PR? Also update the the `configure()`.

Koenkk, 2020-12-07:
Can be closed since https://github.com/Koenkk/zigbee-herdsman-converters/pull/1872 is merged.

closed 2020-12-07 by Koenkk