Koenkk/zigbee2mqtt #5201: Silvercrest HG06337: Failed to interview, device has not successfully been paired

ballwax, 2020-12-05:
# Bug Report
## What happened
I am trying to pair my new device Silvercrest HG06337, which is supported according to:
[https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.16.2](url)
[https://www.zigbee2mqtt.io/devices/HG06337.html](url)

But always I get following message:
"Failed to interview '0xec1bbdfffea14455', device has not successfully been paired"
and
"Interview - failed for device '0xec1bbdfffea14455' with error 'Error: Interview failed because can not get node descriptor ('0xec1bbdfffea14455')"

These hints, didn't help, I tried already:
[https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails](url)

Parts of bug report:
zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":41059,"extaddr":"0xec1bbdfffea14455","parentaddr":0} +50s
  zigbee-herdsman:controller:log Device '0xec1bbdfffea14455' joined +50s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:log Device '0xec1bbdfffea14455' accepted by handler +1ms
  zigbee-herdsman:controller:log Interview '0xec1bbdfffea14455' start +0ms
Zigbee2MQTT:info  2020-12-05 10:22:38: Starting interview of '0xec1bbdfffea14455'
  zigbee-herdsman:controller:device:log Interview - start device '0xec1bbdfffea14455' +10m
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":41059,"nwkaddrofinterest":41059} +50s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,99,160,99,160,35] +50s
Zigbee2MQTT:info  2020-12-05 10:22:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xec1bbdfffea14455"},"type":"pairing"}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +15ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +50s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '41059' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry +10m

...

zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0xec1bbdfffea14455', attempt 6 +23s
  zigbee-herdsman:controller:device:log Interview - failed for device '0xec1bbdfffea14455' with error 'Error: Interview failed because can not get node descriptor ('0xec1bbdfffea14455')
  zigbee-herdsman:controller:device:log     at Device.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:350:23)
  zigbee-herdsman:controller:device:log     at Generator.throw (<anonymous>)
  zigbee-herdsman:controller:device:log     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:25:65)' +1ms
  zigbee-herdsman:controller:error Interview failed for '0xec1bbdfffea14455 with error 'Error: Interview failed because can not get node descriptor ('0xec1bbdfffea14455')' +12m
Zigbee2MQTT:error 2020-12-05 10:24:57: Failed to interview '0xec1bbdfffea14455', device has not successfully been paired
Zigbee2MQTT:info  2020-12-05 10:24:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0xec1bbdfffea14455"},"type":"pairing"}'
Zigbee2MQTT:debug 2020-12-05 10:25:05: Saving state to file /opt/zigbee2mqtt/data/state.json

Thank you in advance,
Marcel

## Debug Info
Zigbee2MQTT version: V1.16.2
Adapter hardware: CC2531
Adapter firmware version: V1.2 CC2531_DEFAULT_20190608.zip

sroka222, 2020-12-05:
I also have a problem with this socket. My model is [HG06337-FR.]

info  2020-12-05 20:02:33: Device '0xbc33acfffe681daf' joined
info  2020-12-05 20:02:33: Starting interview of '0xbc33acfffe681daf'
info  2020-12-05 20:02:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe681daf","ieee_address":"0xbc33acfffe681daf"},"type":"device_joined"}'
info  2020-12-05 20:02:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe681daf","ieee_address":"0xbc33acfffe681daf","status":"started"},"type":"device_interview"}'
info  2020-12-05 20:02:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xbc33acfffe681daf"},"type":"device_connected"}'
info  2020-12-05 20:02:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xbc33acfffe681daf"},"type":"pairing"}'
info  2020-12-05 20:02:34: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe681daf","ieee_address":"0xbc33acfffe681daf"},"type":"device_announce"}'
info  2020-12-05 20:02:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xbc33acfffe681daf"},"type":"device_announced"}'
info  2020-12-05 20:02:37: Successfully interviewed '0xbc33acfffe681daf', device has successfully been paired
warn  2020-12-05 20:02:37: Device '0xbc33acfffe681daf' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_wamqdr3f' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Zigbee2MQTT version: V1.16.2
Adapter hardware: CC2531

kevinbischof, 2020-12-06:
Hey folks,

I faced the same issue like @ballwax, but with the motion sensor HG06335 I tried pairing multiple times, but the interview failed each time).

Here is a screenshot of the frontend:
[image]

Then I did the following:

1. Take battery out of motion sensor
2. Force delete entry of motion sensor
3. Restart zigbee2mqtt service
4. Wait some minutes before inserting battery
5. Reset device (holding the reset button (little hole on the side) for 5 seconds)

Then the motion sensor was recognized as Lidl | HG06335, but still saying pairing interview failed. The mqtt output is looking good:

`{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/lidlBewegungsmelder', payload '{"battery":100,"battery_low":false,"linkquality":68,"occupancy":true,"tamper":true}'"}`

And again a screenshot of the frontend:
[image]

Hope that helps.
Kevin

haggard71, 2020-12-07:
in my case there are no sub devices created....
It only reports model HG06337; Description Silvercrest smart plug (EU, CH, FR, BS)...

Domoticz log says:
(Zigbee2MQTT) Device 0x842e14fffef8f00a (0x842e14fffef8f00a) doesn't have "model" attribute, skipped

CyberClwn, 2020-12-07:
I have the same as haggard71. 

(I have reported it at the domoticz-plugin page of stas:
https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues/488

closed 2021-01-14 by github-actions[bot]

ballwax, 2021-01-16:
Hello all,

first of all, sorry for late response.
Only now, I was able to test a new device with the same result and actually I already gave up. 

But now I got it solved by trying different channels. I found one channel and with it, everything is working as expected.

Thanks.

cdrehner, 2021-02-21:
@ballwax which channel are you using? I cant get it working with channel 11...

marcgarciamarti, 2021-03-14:
@cdrehner did you manage to solve this?

@ballwax can I just simply change the channel in the configuration file and restart the add-on? will this be transparent for the already paired and working devices?

cdrehner, 2021-03-18:
Nope, not permanently.

Zephalon, 2022-01-17:
I have the same issue with the latest version (1.22.2). I can pair the device (if it's not to close too the coordinator) but the interview fails. After trying like 200 times I've had it reporting the occupancy once (!) and then failing again to do so.

The coordinator is a Sonoff Zigbee Adapter Plus with the latest Z-Stack.

rwijnhov, 2022-06-06:
I have the same issue.

26tajeen, 2022-06-15:
I have the same issue, @Koenkk would a log help ?

Koenkk, 2022-06-15:
Does pairing it close to the coordinator helps? If not provide the herdsman debug logging.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

26tajeen, 2022-06-15:
The device is already super close to the coordinator so don't think distance is the issue. It's within 5cm at the moment! :-)

> See [zigbee2mqtt.io/guide/usage/debug.html](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Thanks, will do. I tried logging in via SSH but got -bash: docker: command not found so I'm presuming you mean ssh login to port 2222 as described [here](https://developers.home-assistant.io/docs/operating-system/debugging/)? Not a complaint at all but that isn't clear from the debug page. Again, not a complain, super grateful for the support I'm getting here. I'll make a USB drive with my public key on it and try later on. Thank you!

26tajeen, 2022-06-15:
> Somehow it's channel 11 that's the problem with this device. Switch to 15 and it will start working.

Wow, really? It’s previously worked on this channel but I’m happy to try another one! Do I just do that in the zigbee2mqtt add on settings and all the devices will move over to the new channel? Thanks for the tip, hope it works! 🥰

26tajeen, 2022-06-15:
eek! `Changing the Zigbee channel requires repairing of all your devices!` Do not want!!!
Also...it's definitely previously worked on the channel I use.

adekloet, 2022-06-16:
Try pairing the device to another zigbee hub you might have. Can be any, I used a generic tuya one. After it pairs with that hub. Delete it there and re-add to your production home zigbee2mqtt system. I found it gave me a 90% succesrate in the first try.

26tajeen, 2022-06-16:
Wow, got it working. Didn't change anything about what I was doing but suddenly after the umpteenth reset it was happy and paired more or less instantly.
However it has had the unintended consequence of knocking offline an Xiaomi PIR sensor which has worked flawlessly for years. I'll open a separate issue about that to avoid muddying waters here.

adekloet, 2022-06-16:
Happy that it worked. I even bought the LIDL hub to see if these sensors 
had updates straight via the vendor hub because I suspect buggy Zigbee 
implementation in the device itself. It either goes to sleep to quickly 
to complete the pairing process or somethign else is the "Matter" but 
only with the Z2MQTT pairing.

The pairing with the vendor hub (LIDL, basicly also a Tuya like hub 
implementation) and the Tuya generic hub never failed once which 
offcourse is strange. Apperently the firmware of those hubs have 
something extra to make it succesfull nearly all the time.

What is also very confusing is the led sequences you see when resetting 
on the device. I also end up with the led in all kinds of sequences 
hoping that one of them is actually the pairing mode :)

If you have a possibility to sent Koen Debug logs that would be a great 
help in fingerpointing the exact problem with these devices.

Anyway thanks all and have a good Z2MQTT day!

Links:
------
[1] 
https://github.com/Koenkk/zigbee2mqtt/issues/5201#issuecomment-1157433207
[2] 
https://github.com/

noci2012, 2022-12-13:
I have a HG07310 (not the square device) and it has difficulty pairing..
First it was solid off-line after each pairing attempt (channel 11). After changing to channel 15  they did come online, and still unsupported.
Device id: 0xcc86ecfffe3d37a6 (0x4798) (move2)
Device id: 0xcc86ecfffe47b2f6 (0x49F8) (move3)

info  2022-12-13 23:38:37: MQTT publish: topic 'zigbee2mqtt/move3/availability', payload 'online'
info  2022-12-13 23:38:37: Starting interview of 'move3'
info  2022-12-13 23:38:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"move3","ieee_address":"0xcc86ecfffe47b2f6","status":"started"},"type":"device_interview"}'
info  2022-12-13 23:38:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"move3"},"type":"pairing"}'
error 2022-12-13 23:40:48: Failed to interview 'move3', device has not successfully been paired
info  2022-12-13 23:40:48: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"move3","ieee_address":"0xcc86ecfffe47b2f6","status":"failed"},"type":"device_interview"}'
info  2022-12-13 23:40:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"move3"},"type":"pairing"}'
info  2022-12-13 23:41:35: MQTT publish: topic 'zigbee2mqtt/move2/availability', payload 'online'
info  2022-12-13 23:info  2022-12-13 23:38:37: MQTT publish: topic 'zigbee2mqtt/move3/availability', payload 'online'
info  2022-12-13 23:38:37: Starting interview of 'move3'
info  2022-12-13 23:38:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"move3","ieee_address":"0xcc86ecfffe47b2f6","status":"started"},"type":"device_interview"}'
info  2022-12-13 23:38:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"move3"},"type":"pairing"}'
error 2022-12-13 23:40:48: Failed to interview 'move3', device has not successfully been paired
info  2022-12-13 23:40:48: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"move3","ieee_address":"0xcc86ecfffe47b2f6","status":"failed"},"type":"device_interview"}'
info  2022-12-13 23:40:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"move3"},"type":"pairing"}'
info  2022-12-13 23:41:35: MQTT publish: topic 'zigbee2mqtt/move2/availability', payload 'online'
info  2022-12-13 23:41:35: Starting interview of 'move2'
info  2022-12-13 23:41:35: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"move2","ieee_address":"0xcc86ecfffe3d37a6","status":"started"},"type":"device_interview"}'
info  2022-12-13 23:41:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"move2"},"type":"pairing"}'
error 2022-12-13 23:43:45: Failed to interview 'move2', device has not successfully been paired
info  2022-12-13 23:43:46: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"move2","ieee_address":"0xcc86ecfffe3d37a6","status":"failed"},"type":"device_interview"}'
info  2022-12-13 23:43:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"move2"},"type":"pairing"}'
41:35: Starting interview of 'move2'
info  2022-12-13 23:41:35: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"move2","ieee_address":"0xcc86ecfffe3d37a6","status":"started"},"type":"device_interview"}'
info  2022-12-13 23:41:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"move2"},"type":"pairing"}'
error 2022-12-13 23:43:45: Failed to interview 'move2', device has not successfully been paired
info  2022-12-13 23:43:46: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"move2","ieee_address":"0xcc86ecfffe3d37a6","status":"failed"},"type":"device_interview"}'
info  2022-12-13 23:43:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"move2"},"type":"pairing"}'

For an existing move (HG06335) control it seems that it is stuck in "room occupied" for ever.

noci2012, 2022-12-13:
ok, after several resets the device first announced itself as TUYA0202...., and after .
after several minutes it was presented as HG06335/HG07310.

```
info  2022-12-14 00:16:36: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"move2","ieee_address":"0xcc86ecfffe3d37a6","status":"failed"},"type":"device_interview"}'
info  2022-12-14 00:16:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"move2"},"type":"pairing"}'
info  2022-12-14 00:16:36: MQTT publish: topic 'homeassistant/binary_sensor/0xcc86ecfffe3d37a6/occupancy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/move2/availability"}],"availability_mode":"all","device":{"configuration_url":"http://192.168.6.28:8080/#/device/0xcc86ecfffe3d37a6/info","identifiers":["zigbee2mqtt_0xcc86ecfffe3d37a6"],"manufacturer":"Lidl","model":"Silvercrest smart motion sensor (HG06335/HG07310)","name":"move2"},"device_class":"motion","name":"move2 occupancy","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/move2","unique_id":"0xcc86ecfffe3d37a6_occupancy_zigbee2mqtt","value_template":"{{ value_json.occupancy }}"}'
info  2022-12-14 00:16:36: MQTT publish: topic 'homeassistant/binary_sensor/0xcc86ecfffe3d37a6/battery_low/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/move2/availability"}],"availability_mode":"all","device":{"configuration_url":"http://192.168.6.28:8080/#/device/0xcc86ecfffe3d37a6/info","identifiers":["zigbee2mqtt_0xcc86ecfffe3d37a6"],"manufacturer":"Lidl","model":"Silvercrest smart motion sensor (HG06335/HG07310)","name":"move2"},"device_class":"battery","entity_category":"diagnostic","name":"move2 battery low","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/move2","unique_id":"0xcc86ecfffe3d37a6_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low }}"}'
info  2022-12-14 00:16:36: MQTT publish: topic 'homeassistant/binary_sensor/0xcc86ecfffe3d37a6/tamper/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/move2/availability"}],"availability_mode":"all","device":{"configuration_url":"http://192.168.6.28:8080/#/device/0xcc86ecfffe3d37a6/info","identifiers":["zigbee2mqtt_0xcc86ecfffe3d37a6"],"manufacturer":"Lidl","model":"Silvercrest smart motion sensor (HG06335/HG07310)","name":"move2"},"device_class":"tamper","name":"move2 tamper","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/move2","unique_id":"0xcc86ecfffe3d37a6_tamper_zigbee2mqtt","value_template":"{{ value_json.tamper }}"}'
info  2022-12-14 00:16:36: MQTT publish: topic 'homeassistant/sensor/0xcc86ecfffe3d37a6/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/move2/availability"}],"availability_mode":"all","device":{"configuration_url":"http://192.168.6.28:8080/#/device/0xcc86ecfffe3d37a6/info","identifiers":["zigbee2mqtt_0xcc86ecfffe3d37a6"],"manufacturer":"Lidl","model":"Silvercrest smart motion sensor (HG06335/HG07310)","name":"move2"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"move2 battery","state_class":"measurement","state_topic":"zigbee2mqtt/move2","unique_id":"0xcc86ecfffe3d37a6_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info  2022-12-14 00:16:36: MQTT publish: topic 'homeassistant/sensor/0xcc86ecfffe3d37a6/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/move2/availability"}],"availability_mode":"all","device":{"configuration_url":"http://192.168.6.28:8080/#/device/0xcc86ecfffe3d37a6/info","identifiers":["zigbee2mqtt_0xcc86ecfffe3d37a6"],"manufacturer":"Lidl","model":"Silvercrest smart motion sensor (HG06335/HG07310)","name":"move2"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"move2 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/move2","unique_id":"0xcc86ecfffe3d37a6_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info  2022-12-14 00:17:56: MQTT publish: topic 'zigbee2mqtt/move1', payload '{"battery":100,"battery_low":false,"linkquality":111,"occupancy":true,"tamper":true,"voltage":3000}'
info  2022-12-14 00:19:19: MQTT publish: topic 'zigbee2mqtt/move1', payload '{"battery":100,"battery_low":false,"linkquality":255,"occupancy":true,"tamper":true,"voltage":3000}'
```