Koenkk/zigbee2mqtt #5202: Moes Smart light switch module (1 gang) (white-label of TuYa TS011F_switch) (switch); MS-104Z (new; not MS-104Z-1)

dlupsa, 2020-12-05:
## Information about the device + link
ZigBee+RF Switch Module MS-104Z
https://www.aliexpress.com/item/1005001605349292.html?spm=2114.12010612.8148356.66.352f2d9e0V9SXw

## data/database.db entry of the device
{"id":8,"type":"Router","ieeeAddr":"0x842e14fffe05b70d","nwkAddr":8003,"manufId":4098,"manufName":"_TZ3000_zmy1waw6","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_zmy1waw6","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0014dbc67f","endpointID":1}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1607169956305}

dlupsa, 2020-12-05:
devices.js (working):

	{
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_zmy1waw6'}],
        model: 'TS011F_switch',
        description: 'Smart light switch module (1 gang)',
        vendor: 'TuYa',
        whiteLabel: [
            {vendor: 'Moes', model: 'MS-104Z'},
        ],
        extend: generic.switch,
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            try {
                // Fails for some devices.
                // https://github.com/Koenkk/zigbee2mqtt/issues/4598
                await configureReporting.onOff(endpoint);
            } catch (e) {
                e;
            }
        },
    },

berezovskyi-oleksandr, 2020-12-06:
I made an external converter for this device based on the code above:

```
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');

const bind = async (endpoint, target, clusters) => {
    for (const cluster of clusters) {
        await endpoint.bind(cluster, target);
    }
};

const device = {
    fingerprint: [{ modelID: 'TS011F', manufacturerName: '_TZ3000_zmy1waw6' }],
    model: 'TS011F_switch',
    description: 'Smart light switch module (1 gang)',
    vendor: 'TuYa',
    whiteLabel: [
        { vendor: 'Moes', model: 'MS-104Z' },
    ],
    fromZigbee: [ zigbeeHerdsmanConverters.fromZigbeeConverters.on_off, zigbeeHerdsmanConverters.fromZigbeeConverters.ignore_basic_report ],
    toZigbee: [ zigbeeHerdsmanConverters.toZigbeeConverters.on_off ],
    exposes: [ zigbeeHerdsmanConverters.exposes.switch() ],
    meta: { configureKey: 1 },
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        try {
            // Fails for some devices.
            // https://github.com/Koenkk/zigbee2mqtt/issues/4598
            await configureReporting.onOff(endpoint);
        } catch (e) {
            e;
        }
    },
}

module.exports = device;
```

Koenkk, 2020-12-06:
Added, thanks!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-06 by Koenkk

way2many, 2020-12-13:
@dlupsa I’m very interested in those modules as regular and push button switches can be used but the RF part is worrying me. Is there a physical way to disable this?

I tried to find information but only generic stuff from AliExpress is available.

Thank you.

dlupsa, 2020-12-13:
@way2many Don't know what happens if you don't use/pair RF. You can pull out the antenna :) But I can confirm, that module works fine - tried even RF
[image]

way2many, 2020-12-13:
Thanks for the picture. I’m concerned that Zigbee 3.0 added a level of security in order to prevent unwanted snooping while this device accept RF which is known to have severe security issues. 

I suppose removing the antenna like you suggested might do the trick to “disable” RF.

ster1um, 2021-02-05:
> 
> 
> @dlupsa I’m very interested in those modules as regular and push button switches can be used but the RF part is worrying me. Is there a physical way to disable this?
> 
> I tried to find information but only generic stuff from AliExpress is available.
> 
> Thank you.

have you disabled the RF chip on this switch?

dlupsa, 2021-02-05:
> have you disabled the RF chip on this switch?

No, I left it as it is. I don't think that anyone will come to me and turn off my lights :)

joskohorvat, 2023-02-26:
I had an issue with ZigBee+RF Switch Module MS-104Z. It was too susceptible to RF interferences around. Since I live in country, this was a bit unacceptable. I simply removed the capacitor right before the antenna. Now, seems to be Working OK. Before it was like the attachment shows. Periodicaly ON/OFF in one minute or two time.
[image]

xabufr, 2023-09-07:
@joskohorvat which capacitor did you remove exactly?
Can you circle it in the image above?

Teunhen, 2023-09-10:
> 

I would like to know what to desolder. I have rf inference or something so its flipping all the time.

Thanks in advance

xabufr, 2023-09-11:
You can remove this antenna that receive RF433:

[image]

I did that and it's much better, even if I still sometimes have some troubles with one of my module (that's why I asked @joskohorvat which capacitor he removed).

jkleczkowski, 2023-10-15:
@xabufr Hello. Did you find out which capacitor to remove?

joskohorvat, 2023-10-15:
Uhm, the module is on the roof right now. There are 2 capacitors in use near antenna. One is connected to ground, the other is leading to the chip. The best way is to disconnect right next to the RX/TX pin of the chip. The another module, which had the same issue, I disconected this way. You have to be very accurate at this procedure. Nice propertie of the MS-104Z is ON/OFF log. I use it in conjunction of the rain sensor on my roof.
[image]

jkleczkowski, 2023-10-17:
Unfortunately, removing the capacitor and removing the antenna did not help. The module interferes with the transmitter from the weather station all the time

vukisz, 2023-10-26:
@jkleczkowski I just ordered 10+ of these and 3 of them already turning themself on and off.
Now I am reading manual again and can see that that there is a instruction for clearing RF codes:
2. How to clear up the RF code

2.1 For reset switch:Press the switch button for 5 times until you hear Di-Di(2 times),waiting for 5 seconds, press the switch button for 5 times again until you hear Di-Di-Di-Di(4 times). Then the RF code is now cleared up.

2.2 For rocker light switch Press the switch button for 10 times (ON/OFF cycle for 5times) ¦ until you hear Di-Di(2 times), waiting for 5 seconds, press the switch button for 10 times(ON/OFF cycle for 5 times) again until you hear Di-Di-Di-Di(4 times). Then the RF code is now cleared up.

2.3 For reset button:Long press the reset button on the switch module and release your finger until the beep sounds as Di-Di for 2 times. Then press the button after 5 seconds for 5 seconds until the beep sounds as Di-Di-Di-Di for 4 times.
Can you try it as I am not at a location with the issue at the moment

joskohorvat, 2023-10-27:
Uhm, I tried 100 times or more. It's up to spot where you live. I tried at my work, There it works totaly functional, no issue detectet. So I had to modify it at my home. There you can see this modyfied-disabled radio device, connectet as light triggered by radar sensor.. 
[image]

vukisz, 2023-10-27:
Thanks @joskohorvat.
Mine update: desoldered antena for 4 relays which were malfunctioning - turning on and off. No issues for the last 5 hours for all. Fingers crossed :-)

joskohorvat, 2023-10-29:
Happy, to hear that