Koenkk/zigbee2mqtt #5203: Support for Aqara high precision motion sensor (RTCGQ13LM; lumi.motion.agl04)

Otnow, 2020-12-05:
## Information about the device + link

Aqara high precision motion sensor (RTCGQ13LM; lumi.motion.agl04)

- https://www.reddit.com/r/Aqara/comments/k7aeqy/aqara_high_precision_motion_sensor_rtcgq13lm/
- https://aliexpress.com/item/1005001739628125.html
- https://item.jd.com/10024066313112.html

## data/database.db entry of the device

{"id":42,"type":"EndDevice","ieeeAddr":"0x04cf8cdf3c8a82e0","nwkAddr":53924,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.motion.agl04","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1030,3,1],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.motion.agl04","manufacturerName":"LUMI","powerSource":3,"zclVersion":3,"appVersion":25,"stackVersion":2,"hwVersion":1,"dateCode":"20200722","swBuildId":"2019\u0000www."}}},"binds":[],"meta":{}}},"appVersion":25,"stackVersion":2,"hwVersion":1,"dateCode":"20200722","swBuildId":"2019\u0000www.","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1607196412575}

## Pairing log

- Zigbee2MQTT:debug 2020-12-05 14:26:14: Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
- Zigbee2MQTT:info  2020-12-05 14:26:14: Zigbee: allowing new devices to join.
- Zigbee2MQTT:info  2020-12-05 14:26:16: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"1.16.2","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":104,"product":0,"revision":"0x26680700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.16.2"}'
- Zigbee2MQTT:info  2020-12-05 14:26:29: Device '0x04cf8cdf3c8a82e0' joined
- Zigbee2MQTT:info  2020-12-05 14:26:29: Starting interview of '0x04cf8cdf3c8a82e0'
- Zigbee2MQTT:info  2020-12-05 14:26:29: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c8a82e0","ieee_address":"0x04cf8cdf3c8a82e0"},"type":"device_joined"}'
- Zigbee2MQTT:info  2020-12-05 14:26:29: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c8a82e0","ieee_address":"0x04cf8cdf3c8a82e0","status":"started"},"type":"device_interview"}'
- Zigbee2MQTT:info  2020-12-05 14:26:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cf8cdf3c8a82e0"},"type":"device_connected"}'
- Zigbee2MQTT:info  2020-12-05 14:26:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x04cf8cdf3c8a82e0"},"type":"pairing"}'
- Zigbee2MQTT:debug 2020-12-05 14:26:35: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.motion.agl04"}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2020-12-05 14:26:35: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2020-12-05 14:26:36: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2020-12-05 14:26:36: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2020-12-05 14:26:37: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2020-12-05 14:26:37: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2020-12-05 14:26:39: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2020-12-05 14:26:39: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2020-12-05 14:26:40: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"appVersion":25}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2020-12-05 14:26:40: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2020-12-05 14:26:41: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2020-12-05 14:26:41: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2020-12-05 14:26:43: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2020-12-05 14:26:43: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2020-12-05 14:26:45: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20200722"}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2020-12-05 14:26:45: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:debug 2020-12-05 14:26:46: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2019\u0000www."}' from endpoint 1 with groupID null
- Zigbee2MQTT:debug 2020-12-05 14:26:46: Skipping message, definition is undefined and still interviewing
- Zigbee2MQTT:info  2020-12-05 14:26:46: Successfully interviewed '0x04cf8cdf3c8a82e0', device has successfully been paired
- Zigbee2MQTT:warn  2020-12-05 14:26:46: Device '0x04cf8cdf3c8a82e0' with Zigbee model 'lumi.motion.agl04' and manufacturer name 'LUMI' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
- Zigbee2MQTT:info  2020-12-05 14:26:46: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x04cf8cdf3c8a82e0","ieee_address":"0x04cf8cdf3c8a82e0","status":"successful","supported":false},"type":"device_interview"}'
- Zigbee2MQTT:info  2020-12-05 14:26:46: MQTT publish: topic 'zigbee2mqtt/bridge/devices', payload '**[{"date_code":"20200722","definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":["genBasic","msOccupancySensing","genIdentify","genPowerCfg"],"output":["genIdentify","genOta"]}}},"friendly_name":"0x04cf8cdf3c8a82e0","ieee_address":"0x04cf8cdf3c8a82e0","interview_completed":true,"interviewing":false,"network_address":37338,"power_source":"Battery","software_build_id":"2019\u0000www.","supported":false,"type":"EndDevice"}]**'
- Zigbee2MQTT:info  2020-12-05 14:26:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x04cf8cdf3c8a82e0","supported":false},"type":"pairing"}'

Jhonattan-Souza, 2021-01-30:
I'm not finding this device anywhere to buy

Otnow, 2021-02-03:
> I'm not finding this device anywhere to buy

Yes, unfortunately this is still the case. I bought this sensor myself by a miracle ([my purchase history](https://www.reddit.com/r/Aqara/comments/k7aeqy/aqara_high_precision_motion_sensor_rtcgq13lm/gexkm4g/)).

@Koenkk, please check. By analogy with `Aqara motion sensor T1 (RTCGQ12LM)` and `Philips Hue motion sensor (SML001)` full support of this sensor is [implemented](https://github.com/Koenkk/zigbee-herdsman-converters/pull/2199) (with changing sensitivity and motion detection period):

<details>
  <summary>Zigbee2mqtt log</summary>

```
- info  2021-02-03 18:44:42: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"2ef2c54","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190425,"transportrev":2},"type":"zStack30x"},"log_level":"debug","network":{"channel":25,"extendedPanID":"0x00124b0014d9f126","panID":22442},"permit_join":false,"version":"1.17.0-dev"}'
- debug 2021-02-03 18:45:50: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"transaction":"cz4aw-1","value":true}'
- info  2021-02-03 18:45:50: Zigbee: allowing new devices to join.
- info  2021-02-03 18:45:50: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":true},"status":"ok","transaction":"cz4aw-1"}'
- info  2021-02-03 18:46:04: Device '0x04cf8cdf3c8a82e0' joined
- info  2021-02-03 18:46:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c8a82e0","ieee_address":"0x04cf8cdf3c8a82e0"},"type":"device_joined"}'
- info  2021-02-03 18:46:04: Starting interview of '0x04cf8cdf3c8a82e0'
- info  2021-02-03 18:46:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c8a82e0","ieee_address":"0x04cf8cdf3c8a82e0","status":"started"},"type":"device_interview"}'
- info  2021-02-03 18:46:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x04cf8cdf3c8a82e0"},"type":"device_connected"}'
- info  2021-02-03 18:46:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x04cf8cdf3c8a82e0"},"type":"pairing"}'
- debug 2021-02-03 18:46:04: Device '0x04cf8cdf3c8a82e0' announced itself
- info  2021-02-03 18:46:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cf8cdf3c8a82e0","ieee_address":"0x04cf8cdf3c8a82e0"},"type":"device_announce"}'
- info  2021-02-03 18:46:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x04cf8cdf3c8a82e0"},"type":"device_announced"}'
- debug 2021-02-03 18:46:07: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.motion.agl04"}' from endpoint 1 with groupID 0
- debug 2021-02-03 18:46:07: No converter available for 'RTCGQ13LM' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"lumi.motion.agl04"}'
- debug 2021-02-03 18:46:08: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
- debug 2021-02-03 18:46:08: No converter available for 'RTCGQ13LM' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"LUMI"}'
- debug 2021-02-03 18:46:11: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
- debug 2021-02-03 18:46:11: No converter available for 'RTCGQ13LM' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
- debug 2021-02-03 18:46:12: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
- debug 2021-02-03 18:46:12: No converter available for 'RTCGQ13LM' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
- debug 2021-02-03 18:46:13: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"appVersion":26}' from endpoint 1 with groupID 0
- debug 2021-02-03 18:46:13: No converter available for 'RTCGQ13LM' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":26}'
- debug 2021-02-03 18:46:14: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
- debug 2021-02-03 18:46:14: No converter available for 'RTCGQ13LM' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":2}'
- debug 2021-02-03 18:46:15: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
- debug 2021-02-03 18:46:15: No converter available for 'RTCGQ13LM' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
- debug 2021-02-03 18:46:16: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20201123"}' from endpoint 1 with groupID 0
- debug 2021-02-03 18:46:16: No converter available for 'RTCGQ13LM' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20201123"}'
[... cut ...]
- debug 2021-02-03 18:55:16: Received Zigbee message from '0x04cf8cdf3c8a82e0', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 1 with groupID 0
- info  2021-02-03 18:55:16: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c8a82e0', payload '{"battery":100,"linkquality":126,"motion_sensitivity":"high","occupancy":false,"occupancy_timeout":0,"voltage":3200}'
```

</details>

**Environment:**

- Sensor firmware version: _0.0.0_0026_
- Ebyte E18-2G4U04B (CC2531) firmware version: _zStack30x (20190425)_
- Zigbee2mqtt version: _1.17.0-dev (2ef2c54)_

closed 2021-02-04 by Koenkk

avdept, 2021-06-02:
How to put it to pairing mode? The reset button seems to be not doing what it should

Otnow, 2021-06-02:
> How to put it to pairing mode? The reset button seems to be not doing what it should

To turn on the pairing mode, you need to hold down the button (~ 5 sec) until the blue LED inside the sensor starts to flash periodically.

stuntik, 2021-07-17:
I got this sensor recently (rare find that was) and I was under the impression that it works with zigbee2mqtt as it is listed. I successfully paired as supported, but no incoming messages are received. The occupancy state never changes. I am confused. Is support for this sensor actually implemented?

avdept, 2021-07-20:
> I got this sensor recently (rare find that was) and I was under the impression that it works with zigbee2mqtt as it is listed. I successfully paired it with HA and it was detected, interviewed. and shows as supported, but no incoming messages are received. The occupancy state never changes. I am confused. Is support for this sensor actually implemented?

I got this sensor too and it works just fine for me. Try to re-pair it.

stuntik, 2021-07-20:
Oh man, thanks for returning my faith into trying to fix that! And oh! fk, why is it such a hidden rare knowledge that I just uncovered by an accident that Aqara devices pair very strangely. It appeared that it's not enough to just hold the button on the sensor for 5 secs to turn on the pairing mode, you would need to keep doing short press every 2-3 seconds to pair it properly. I did it like 5 times until it started to work. Maybe that helps someone else.

Otnow, 2021-10-09:
> Hi guys, did the occupancy_timeout and the motion_sensitivity options work for you? For me is not working properly

Immediately after changing the settings in the web interface, you need to press the button on the sensor so that it wakes up and accepts the new parameter values.

pydog, 2021-11-28:
> > Hi guys, did the occupancy_timeout and the motion_sensitivity options work for you? For me is not working properly
> 
> Immediately after changing the settings in the web interface, you need to press the button on the sensor so that it wakes up and accepts the new parameter values.

Is it possible to let it wakes up without pressing the button. I just would like to update the occupancy_timeout and motion_sensitivity in different time of day, something like high sensitivity in the evening while low sensitivity during daytime. Thanks.

Otnow, 2021-11-28:
> Is it possible to let it wake up without pressing the button. I just would like to update the occupancy_timeout and motion_sensitivity in different time of day, something like high sensitivity in the evening while low sensitivity during daytime. Thanks.

This motion sensor is battery powered and most of the time it sleeps to save energy, so it cannot constantly listen to the radio air to quickly change its settings.
To test the feasibility of your idea, you need to experiment. I haven’t done this yet, but maybe I’ll try it somehow.

pydog, 2021-11-29:
> This motion sensor is battery powered and most of the time it sleeps to save energy, so it cannot constantly listen to the radio air to quickly change its settings. To test the feasibility of your [...]

cool man. BTW, I try with the method you had mentioned, but still get the following failure, do you have any idea?
Nov 29 09:23:21 smarthome npm[9087]: Zigbee2MQTT:error 2021-11-29 09:23:21: Publish 'set' 'occupancy_timeout' to 'primaryroom_washroom_motionsensor' failed: 'Error: Write 0x54ef4410002dc6a2/1 msOccupancySensing({"pirOToUDelay":30}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Nov 29 09:23:21 smarthome npm[9087]: Zigbee2MQTT:info  2021-11-29 09:23:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'occupancy_timeout' to 'primaryroom_washroom_motionsensor' failed: 'Error: Write 0x54ef4410002dc6a2/1 msOccupancySensing({\"pirOToUDelay\":30}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"primaryroom_washroom_motionsensor"},"type":"zigbee_publish_error"}'

Otnow, 2021-11-29:
> cool man. BTW, I try with the method you had mentioned, but still get the following failure, do you have any idea?

Unfortunately, I am not an expert in debugging, so I can only recommend re-pairing the sensor: remove it from the network and add it again.

pydog, 2021-11-29:
> > cool man. BTW, I try with the method you had mentioned, but still get the following failure, do you have any idea?
> 
> Unfortunately, I am not an expert in debugging, so I can only recommend re-pairing the sensor: remove it from the network and add it again.

Cool man. I re-paired the sensor and send the cmd as soon as z2m reported paird successfully. I it was set successfully.

serje73, 2021-12-24:
Hi guys.
Need help. Downgraded CC2531 firmware from 3.0.X to 1.2 for Aqara E1 Switch support. And now I re-paired two motion sensors. And they are don't change the occupancy state. I set up occupancy-timeout to 30 second (and pushed the button on the sensor, yes). State says "occupancy_timeout":30 . But occupancy is true all the time. I see in log when sensor detects motion. But there is nothing in log about changing occupancy to false. So dashboard shows "true".
Where to dig?
I re-paired them about 5 times. The same picture.

Otnow, 2021-12-25:
> Hi guys.
> Need help. Downgraded CC2531 firmware from 3.0.X to 1.2 for Aqara E1 Switch support. And now I re-paired two motion sensors. And they are don't change the occupancy state. I set up [...]

I think that when working with modified firmware, RTCGQ13LM changed its logic as well as [RTCGQ12LM](https://github.com/Koenkk/zigbee2mqtt/issues/5706#issuecomment-972472433) (_completely ceases to use standard clusters (msOccupancySensing no longer gives occupancy data and is occupied all the time)_).

Therefore, now it is required to write a new converter and I will try to do it in the near future.

serje73, 2021-12-25:
> I think that when working with modified firmware, RTCGQ13LM changed its logic as well as [RTCGQ12LM](https://github.com/Koenkk/zigbee2mqtt/issues/5706#issuecomment-972472433) (_completely ceases to [...]

Thanks a lot. Think the same way. But thanks for approving. And for your help.

z-x-f, 2021-12-25:
> I think that when working with modified firmware, RTCGQ13LM changed its logic as well as [RTCGQ12LM](https://github.com/Koenkk/zigbee2mqtt/issues/5706#issuecomment-972472433) (_completely ceases to [...]

FYI, my RTCGQ13LM looks like everything is normal, and the coordinator firmware is a modified firmware.
<pre>Debug Received Zigbee message from 'motion_sensor_washroom', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 1 with groupID 0</pre>

Zigbee2MQTT version: 1.22.1-dev commit: f9aa5250
Coordinator type: zStack3x0
Coordinator revision: 20211217

Otnow, 2021-12-25:
> FYI, my RTCGQ13LM looks like everything is normal, and the coordinator firmware is a modified firmware.

Thank you, I will try to check it with myself in the near future.

serje73, 2021-12-26:
> > > Hi guys.
> > > Need help. Downgraded CC2531 firmware from 3.0.X to 1.2 for Aqara E1 Switch support. And now I re-paired two motion sensors. And they are don't change the occupancy state. I set up occupancy-timeout to 30 second (and pushed the button on the sensor, yes). State says "occupancy_timeout":30 . But occupancy is true all the time. I see in log when sensor detects motion. But there is nothing in log about changing occupancy to false. So dashboard shows "true".
> > > Where to dig?
> > > I re-paired them about 5 times. The same picture.
> > 
> > 
> > I think that when working with modified firmware, RTCGQ13LM changed its logic as well as [RTCGQ12LM](https://github.com/Koenkk/zigbee2mqtt/issues/5706#issuecomment-972472433) (_completely ceases to use standard clusters (msOccupancySensing no longer gives occupancy data and is occupied all the time)_).
> > Therefore, now it is required to write a new converter and I will try to do it in the near future.
> 
> FYI, my RTCGQ13LM looks like everything is normal, and the coordinator firmware is a modified firmware.
> 
> Debug Received Zigbee message from 'motion_sensor_washroom', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 1 with groupID 0
> 
> Zigbee2MQTT version: 1.22.1-dev commit: [f9aa525](https://github.com/Koenkk/zigbee2mqtt/commit/f9aa5250487c58be7d1bfc32a150a2d9bab398a4) Coordinator type: zStack3x0 Coordinator revision: 20211217

I have another stick - 2531:

Zigbee2MQTT version 1.22.1 commit: unknown
Coordinator type zStack12
Coordinator revision 20211115

And I have no messages about occupancy off. I cannot move to 3.0.x because 3.0.x does not support another Aqara device - QBKG39LM.

I will try cc2652 a couple days later. But now have only 2531 with this result.

Otnow, 2022-01-06:
[RTCGQ13LM converter changed](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3631) for correct operation on coordinator firmware with support for native Aqara logic (is now available in the _[dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)_, and will appear in the _master branch_ at the beginning of next month), therefore, _for normal operation of the sensor with a new converter, it is [necessary to update the coordinator's firmware](https://www.zigbee2mqtt.io/devices/RTCGQ13LM.html#adapter-firmware)_.

On the coordinator's firmware without support for the native Aqara logic, as a workaround, it is proposed to use the previous version of the converter added in the configuration settings via the [`external_converters`](https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters) option:

- RTCGQ13LM_basic_mode.js

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.motion.agl04'],
    model: 'RTCGQ13LM',
    vendor: 'Xiaomi',
    description: 'Aqara high precision motion sensor',
    fromZigbee: [fz.occupancy, fz.occupancy_timeout, fz.aqara_opple, fz.battery],
    toZigbee: [tz.occupancy_timeout, tz.RTCGQ13LM_motion_sensitivity],
    exposes: [e.occupancy(), exposes.enum('motion_sensitivity', ea.ALL, ['low', 'medium', 'high']),
        exposes.numeric('occupancy_timeout', ea.ALL).withValueMin(2).withValueMax(65535).withUnit('s')
            .withDescription('Time in seconds till occupancy goes to false'), e.battery()],
    meta: {battery: {voltageToPercentage: '3V_2100'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'msOccupancySensing']);
        await reporting.occupancy(endpoint);
        await reporting.batteryVoltage(endpoint);
        await endpoint.read('msOccupancySensing', ['pirOToUDelay']);
        await endpoint.read('aqaraOpple', [0x010c], {manufacturerCode: 0x115f});
    },
};

module.exports = definition;
```

memme2k, 2022-01-27:
Hi,

I got the same Problem, the Motion Sensor RTCGQ13LM is occupany always true.
Setup:
SONOFF Zigbee 3.0 USB Dongle
CC1352P2_CC2652P_launchpad_coordinator_20211217.zip
Z-Stack_Home_3.x.0_20211217 
ioBroker (latest versions)

How to fix it? You updated just the Zigbee2MQTT files, but now the Z-Stack?

Thank you very much

Otnow, 2022-01-27:
> How to fix it? You updated just the Zigbee2MQTT files, but now the Z-Stack?

On the latest firmware of the coordinator (like yours), the sensor will work fine only on the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) until a new stable version is released (early next month).

memme2k, 2022-01-28:
> On the latest firmware of the coordinator (like yours), the sensor will work fine only on the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) until a new stable [...]
Will this dev version work? : https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20220103.zip

Or is it better to wait for the official next stable version?
Thanks

Otnow, 2022-01-28:
> Will this dev version work? : https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20220103.zip
> Or is it better to wait for the official next stable version?

I meant version _z2m_, not the firmware of the coordinator. I'm not familiar with _ioBroker_, so you better check on the profile forum how to install the developer version _z2m_ in it and if it's possible.

memme2k, 2022-01-28:
I dont use the Zigbee2MQTT, I use the direct zigbee adapter.
So how to fix it now?

Otnow, 2022-01-28:
> I dont use the Zigbee2MQTT, I use the direct zigbee adapter.
> So how to fix it now?

If you are not using _Zigbee2MQTT_ then I can't help you and you should contact `direct zigbee adapter` support accordingly.

memme2k, 2022-01-28:
But isn't this a firmware issue?

Otnow, 2022-01-28:
> But isn't this a firmware issue? I am using https://github.com/ioBroker/ioBroker.zigbee

Last fall, changes were made to the coordinator’s firmware, after which the new Aqara sensors consider that they are connected to their native gateway and change the logic of operation, therefore, for the normal operation of such devices, in addition to updating the coordinator’s firmware, updating the algorithms for working with them is required.

If the software you use does not support the modified Aqara sensor logic, then _you have to downgrade the coordinator firmware (for example, released last summer) or contact the developers of the software you use to add such support_.

Otnow, 2022-03-19:
[RTCGQ13LM converter modified](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4006) to solve [latency issue](https://github.com/Koenkk/zigbee2mqtt/issues/11843) when the timeout ends (`detection_interval`), occupancy is cleared even if in fact there is still movement.

To do this, option `occupancy_timeout` has been added, which allows you to set an arbitrary _delay_ after the end of `detection_interval`, so as not to clear the occupancy if motion is detected again during the _delay_:

`occupancy_timeout` = `detection_interval` + _delay_

_**Important**: default value for `occupancy_timeout` is set as `detection_interval` + 2 seconds, so you may need to adjust it for your needs as there will be an additional delay of 2 seconds after you update zigbee2mqtt (these changes is now available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html), and will appear in the master branch at the beginning of next month)._ 

Of the additional changes: added `no_occupancy_since` option and `temperature` expose.

In zigbee2mqtt web interface, `occupancy_timeout` and `no_occupancy_since` options is located on `Settings (specific)` tab of the sensor.

PaulMcMillan, 2022-03-24:
Note: In order to take advantage of the improved behavior, you may also need to reset and re-pair the device with z2m. To do this, hold down the back button for 10 seconds, release, press again, and make sure z2m is configured to permit joins.