Koenkk/zigbee2mqtt #5204: Silvercrest HG06337-FR: problem with configuring support new device

smarthomepch, 2020-12-05:
## Information about the device + link
I try to add new Silvercrest HG06337-FR (smart plug):

After first run I see:

```
warn  2020-12-05 20:41:30: Device '0xbc33acfffe4e7293' with Zigbee model 'undefined' and manufacturer name '_TZ3000_wamqdr3f' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
info  2020-12-05 20:41:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0xbc33acfffe4e7293","supported":false},"type":"pairing"}'
```
Based on the instruction I have added below entry to` /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js`:
```

    {
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_wamqdr3f'}],
        model: 'HG06337-FR',
        vendor: 'Silvercrest',
        description: 'Smart Plug',
        extend: generic.switch,
    },
```

After restart and 2nd try of pairing this I see:

```
info  2020-12-05 20:43:39: Device '0xbc33acfffe4e7293' joined
info  2020-12-05 20:43:39: Starting interview of '0xbc33acfffe4e7293'
info  2020-12-05 20:43:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xbc33acfffe4e7293"},"type":"device_connected"}'
info  2020-12-05 20:43:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xbc33acfffe4e7293"},"type":"pairing"}'
debug 2020-12-05 20:43:39: Device '0xbc33acfffe4e7293' announced itself
info  2020-12-05 20:43:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xbc33acfffe4e7293"},"type":"device_announced"}'
debug 2020-12-05 20:43:40: Received Zigbee message from '0xbc33acfffe4e7293', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS011F"}' from endpoint 11 with groupID 0
debug 2020-12-05 20:43:40: Skipping message, definition is undefined and still interviewing
debug 2020-12-05 20:43:40: Received Zigbee message from '0xbc33acfffe4e7293', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ3000_wamqdr3f"}' from endpoint 11 with groupID 0
debug 2020-12-05 20:43:40: Received Zigbee message from '0xbc33acfffe4e7293', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 11 with groupID 0
debug 2020-12-05 20:43:40: Received Zigbee message from '0xbc33acfffe4e7293', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 11 with groupID 0
debug 2020-12-05 20:43:41: Received Zigbee message from '0xbc33acfffe4e7293', type 'readResponse', cluster 'genBasic', data '{"appVersion":66}' from endpoint 11 with groupID 0
debug 2020-12-05 20:43:41: Received Zigbee message from '0xbc33acfffe4e7293', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 11 with groupID 0
debug 2020-12-05 20:43:41: Received Zigbee message from '0xbc33acfffe4e7293', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 11 with groupID 0
debug 2020-12-05 20:43:41: Received Zigbee message from '0xbc33acfffe4e7293', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 11 with groupID 0
debug 2020-12-05 20:43:41: Received Zigbee message from '0xbc33acfffe4e7293', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 11 with groupID 0
info  2020-12-05 20:43:41: Successfully interviewed '0xbc33acfffe4e7293', device has successfully been paired
info  2020-12-05 20:43:41: Device '0xbc33acfffe4e7293' is supported, identified as: Silvercrest Smart Plug (HG06337-FR)
info  2020-12-05 20:43:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Smart Plug","friendly_name":"0xbc33acfffe4e7293","model":"HG06337-FR","supported":true,"vendor":"Silvercrest"},"type":"pairing"}'
```

And nothing more happen. I don't see any messages from this device :(

P.S. I have updated zigbee2mqtt before I started

## data/database.db entry of the device

`{"id":22,"type":"Router","ieeeAddr":"0xbc33acfffe4e7293","nwkAddr":4407,"manufId":4098,"manufName":"_TZ3000_wamqdr3f","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_wamqdr3f","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1607197421464}`

Thanks for help in advance!

sroka222, 2020-12-05:
I also have a problem with this socket HG06337-FR

info 2020-12-05 20:02:33: Device '0xbc33acfffe681daf' joined
info 2020-12-05 20:02:33: Starting interview of '0xbc33acfffe681daf'
info 2020-12-05 20:02:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe681daf","ieee_address":"0xbc33acfffe681daf"},"type":"device_joined"}'
info 2020-12-05 20:02:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe681daf","ieee_address":"0xbc33acfffe681daf","status":"started"},"type":"device_interview"}'
info 2020-12-05 20:02:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xbc33acfffe681daf"},"type":"device_connected"}'
info 2020-12-05 20:02:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xbc33acfffe681daf"},"type":"pairing"}'
info 2020-12-05 20:02:34: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe681daf","ieee_address":"0xbc33acfffe681daf"},"type":"device_announce"}'
info 2020-12-05 20:02:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xbc33acfffe681daf"},"type":"device_announced"}'
info 2020-12-05 20:02:37: Successfully interviewed '0xbc33acfffe681daf', device has successfully been paired
warn 2020-12-05 20:02:37: Device '0xbc33acfffe681daf' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_wamqdr3f' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Zigbee2MQTT version: V1.16.2
Adapter hardware: CC2531

kadiusone, 2020-12-06:
It was working ok on 1.16.1 and it got Unsupported after update to 1.16.2

bverlinden, 2020-12-06:
I was able to succesfully pair the device but it doesn't show up in my devices in domoticz.
This is what I added in devices.js:
    {
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_wamqdr3f'}],
        model: 'HG06337-FR',
        vendor: 'Silvercrest',
        description: 'Smart Plug',
        extend: generic.switch,
    },
Log after pairing succesfully:
zigbee2mqtt/bridge/log {'message': 'interview_successful', 'meta': {'description': 'Smart Plug', 'friendly_name': '0xbc33acfffe56a260', 'model': 'HG06337-FR', 'supported': True, 'vendor': 'Silvercrest'}, 'type': 'pairing'}

Log after switching on (which works):
This plugin does not support zigbee device with model "HG06337-FR" yet
If you would like plugin to support this device, please create ticket by this link: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues/new?labels=new+device&template=new-device-support.md

szporwolik, 2020-12-07:
Was working in 1.16.1 and lost support in 1.16.2

Version:
https://i.imgur.com/i6imMzk.png

Device:
https://i.imgur.com/d8uPbXu.png

French plug version:
https://www.lidl-sklep.pl/p/silvercrest-przejsciowka-do-gniazdek-zigbee-smart-home/p100306992

Koenkk, 2020-12-08:
@szporwolik can you try with the latest dev? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

szporwolik, 2020-12-08:
@Koenkk sure, will dev docker container be OK as well ?

kadiusone, 2020-12-08:
@Koenkk sorry to jump in. I have just tested on Edge version of the plugin and work ok in my case.

smarthomepch, 2020-12-08:
I tried the latest dev and seems nothing changed for me or.. I am doing something wrong.
Item is properly discovered and paired:
```

info  2020-12-08 11:10:32: Device '0xbc33acfffe4e7293' joined
info  2020-12-08 11:10:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe4e7293","ieee_address":"0xbc33acfffe4e7293"},"type":"device_joined"}'
info  2020-12-08 11:10:32: Starting interview of '0xbc33acfffe4e7293'
info  2020-12-08 11:10:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe4e7293","ieee_address":"0xbc33acfffe4e7293","status":"started"},"type":"device_interview"}'
info  2020-12-08 11:10:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xbc33acfffe4e7293"},"type":"device_connected"}'
info  2020-12-08 11:10:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xbc33acfffe4e7293"},"type":"pairing"}'
info  2020-12-08 11:10:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe4e7293","ieee_address":"0xbc33acfffe4e7293"},"type":"device_announce"}'
info  2020-12-08 11:10:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xbc33acfffe4e7293"},"type":"device_announced"}'
info  2020-12-08 11:10:44: Successfully interviewed '0xbc33acfffe4e7293', device has successfully been paired
info  2020-12-08 11:10:44: Device '0xbc33acfffe4e7293' is supported, identified as: Lidl Silvercrest smart plug (EU, CH, FR, BS) (HG06337)
info  2020-12-08 11:10:44: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Silvercrest smart plug (EU, CH, FR, BS)","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"HG06337","vendor":"Lidl"},"friendly_name":"0xbc33acfffe4e7293","ieee_address":"0xbc33acfffe4e7293","status":"successful","supported":true},"type":"device_interview"}'
```
entry in **database.db**:
`
{"id":22,"type":"Router","ieeeAddr":"0xbc33acfffe4e7293","nwkAddr":30158,"manufId":4098,"manufName":"_TZ3000_wamqdr3f","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_wamqdr3f","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1607422244318}`

but when I run `mosquitto_sub -d -t zigbee2mqtt/#` and push the button on the device I see nothing (no status changes On/Off)

szporwolik, 2020-12-08:
@Koenkk with the latest docker image:
- lidl sockets - not displayed in the GUI/not available
- lidl led strip - unsupported device

Koenkk, 2020-12-09:
@szporwolik can you provide the data/database.db entries for both?

szporwolik, 2020-12-09:
@Koenkk sure, no problem. 

Both devices purchased at Lild in Poland (French plugs).

Led strip:
`{"id":29,"type":"Router","ieeeAddr":"0x588e81fffeec9110","nwkAddr":62126,"manufId":4098,"manufName":"_TZ3000_riwp3k79","powerSource":"Mains (single phase)","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"manufacturerName":"_TZ3000_riwp3k79","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1607372714511}`

Outlet:
`{"id":13,"type":"Router","ieeeAddr":"0xbc33acfffe697be2","nwkAddr":42295,"manufId":4098,"manufName":"_TZ3000_wamqdr3f","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_wamqdr3f","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1607372251170}`

Koenkk, 2020-12-09:
@szporwolik should be fixed now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

smarthomepch, 2020-12-09:
@Koenkk does the latest dev branch solve both issues (with lidl sockets and lidl led strip) or only with lidl led strip?

szporwolik, 2020-12-09:
@Koenkk confirmed to be supported with the latest docker dev image:

Light strip:
https://i.imgur.com/igLdgmS.png

Outlet
https://i.imgur.com/rkWYWih.png

Thank you Sir! 👍🏻

Koenkk, 2020-12-10:
@smarthomepch should work in latest dev indeed.

brozikcz, 2020-12-11:
I can confirm too that the `Silvercrest smart plug (EU, CH, FR, BS) (HG06337)` works in the dev branch.

The device doesn't send feedback / zigbee message when manually press the button on the device is that right?

Koenkk, 2020-12-11:
@brozikcz can you try changing the definition in devices.js to

```js
    {
        fingerprint: [
            {modelID: 'TS011F', manufacturerName: '_TZ3000_kdi2o9m6'}, // EU
            {modelID: 'TS011F', manufacturerName: '_TZ3000_plyvnuf5'}, // CH
            {modelID: 'TS011F', manufacturerName: '_TZ3000_wamqdr3f'}, // FR
            {modelID: 'TS011F', manufacturerName: '_TZ3000_00mk2xzy'}, // BS
        ],
        model: 'HG06337',
        vendor: 'Lidl',
        description: 'Silvercrest smart plug (EU, CH, FR, BS)',
        extend: generic.switch,
    },
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint);
        },
```

and see if that fixes the issue?

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

brozikcz, 2020-12-11:
I have changed to:

```
    {                                                                            
        fingerprint: [                                                           
            {modelID: 'TS011F', manufacturerName: '_TZ3000_kdi2o9m6'}, // EU     
            {modelID: 'TS011F', manufacturerName: '_TZ3000_plyvnuf5'}, // CH     
            {modelID: 'TS011F', manufacturerName: '_TZ3000_wamqdr3f'}, // FR     
            {modelID: 'TS011F', manufacturerName: '_TZ3000_00mk2xzy'}, // BS     
        ],                                                                                    
        model: 'HG06337',                                                        
        vendor: 'Lidl',                                                          
        description: 'Silvercrest smart plug (EU, CH, FR, BS)',                  
        extend: generic.switch,                                                  
        meta: {configureKey: 1},                                                 
        configure: async (device, coordinatorEndpoint) => {                      
            const endpoint = device.getEndpoint(1);                              
            await bind(endpoint, coordinatorEndpoint, ['genOnOff']);             
            await configureReporting.onOff(endpoint);                            
        },                                                                              
    },    
```
but I have got:

```
Zigbee2MQTT:error 2020-12-11 17:45:02: Failed to configure '0x0017880103e58494', attempt 1 (Error: Bind 0x0017880103e58494/1 genOnOff from '0x00124b0018e8718a/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
Zigbee2MQTT:info  2020-12-11 17:45:02: Configuring 'switch'
Zigbee2MQTT:error 2020-12-11 17:45:02: Failed to configure 'switch', attempt 1 (TypeError: Cannot read property 'bind' of undefined
    at bind (/app/node_modules/zigbee-herdsman-converters/devices.js:49:24)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices.js:15452:19)
    at Configure.configure (/app/lib/extension/configure.js:131:45)
    at Configure.onZigbeeStarted (/app/lib/extension/configure.js:99:28)
    at async Controller.callExtensionMethod (/app/lib/controller.js:381:21))
```

Truddel, 2020-12-11:
I have tried it as well and have the same feedback like @brozikcz

Truddel, 2020-12-11:
Okay I have tried to remove and pair the devices again after the changes you mentioned. 
Now I do not get the error message anymore but the state still does not change automatically after pressing the hardware button on the socket itself. 

Debug Log when pressing on State refresh on expose tab in UI:

> 2020-12-11T18:37:32.470005074Z Zigbee2MQTT:debug 2020-12-11 19:37:32: Received MQTT message on 'zigbee2mqtt/Silvercrest01/get' with data '{"state":""}'
> 
> 2020-12-11T18:37:32.474109434Z Zigbee2MQTT:debug 2020-12-11 19:37:32: Publishing get 'get' 'state' to 'Silvercrest01'
> 
> 2020-12-11T18:37:32.476125518Z   zigbee-herdsman:controller:endpoint Read 0xec1bbdfffea1319b/11 genOnOff(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +18s
> 
> 2020-12-11T18:37:32.477020531Z   zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xec1bbdfffea1319b:6767/11 (0,0,1) +18s
> 
> 2020-12-11T18:37:32.478178271Z   zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":6767,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":39,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,40,0,0,0]}} +18s
> 
> 2020-12-11T18:37:32.479649628Z   zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,111,26,11,1,6,0,39,0,30,5,16,40,0,0,0,87] +18s
> 
> 2020-12-11T18:37:32.492031412Z   zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +16s
> 
> 2020-12-11T18:37:32.492700991Z   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
> 
> 2020-12-11T18:37:32.493061278Z   zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +1ms
> 
> 2020-12-11T18:37:32.493513707Z   zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +18s
> 
> 2020-12-11T18:37:32.493616995Z   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
> 
> 2020-12-11T18:37:32.499784839Z   zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,39,225] +6ms
> 
> 2020-12-11T18:37:32.500114915Z   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,39,225] +1ms
> 
> 2020-12-11T18:37:32.500349515Z   zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,39] - 225 +0ms
> 
> 2020-12-11T18:37:32.501167491Z   zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":39} +16s
> 
> 2020-12-11T18:37:32.501961351Z   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
> 
> 2020-12-11T18:37:32.508541516Z   zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,111,26,0,247] +6ms
> 
> 2020-12-11T18:37:32.509066137Z   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,111,26,0,247] +1ms
> 
> 2020-12-11T18:37:32.509834995Z   zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [111,26,0] - 247 +0ms
> 
> 2020-12-11T18:37:32.510292372Z   zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":6767,"relaycount":0,"relaylist":[]} +9ms
> 
> 2020-12-11T18:37:32.510403943Z   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
> 
> 2020-12-11T18:37:32.557652058Z   zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,6,0,111,26,11,1,0,15,0,132,60,169,0,0,8,24,40,1,0,0,0,16,0,111,26,29,255] +47ms
> 
> 2020-12-11T18:37:32.558310700Z   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,6,0,111,26,11,1,0,15,0,132,60,169,0,0,8,24,40,1,0,0,0,16,0,111,26,29,255] +1ms
> 
> 2020-12-11T18:37:32.558442585Z   zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,6,0,111,26,11,1,0,15,0,132,60,169,0,0,8,24,40,1,0,0,0,16,0,111,26,29] - 255 +0ms
> 
> 2020-12-11T18:37:32.559565635Z   zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":6767,"srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"linkquality":15,"securityuse":0,"timestamp":11091076,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,40,1,0,0,0,16,0]}} +49ms
> 
> 2020-12-11T18:37:32.561671569Z   zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":40,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":0,"status":0,"dataType":16,"attrData":0}]},"address":6767,"endpoint":11,"linkquality":15,"groupID":0}' +16s
> 
> 2020-12-11T18:37:32.564108100Z Zigbee2MQTT:debug 2020-12-11 19:37:32: Received Zigbee message from 'Silvercrest01', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 11 with groupID 0
> 
> 2020-12-11T18:37:32.565964016Z   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +8ms
> 
> 2020-12-11T18:37:32.568874071Z Zigbee2MQTT:info  2020-12-11 19:37:32: MQTT publish: topic 'zigbee2mqtt/Silvercrest01', payload '{"linkquality":15,"state":"OFF"}'
>

Koenkk, 2020-12-11:
And with:

```js
    {
        fingerprint: [
            {modelID: 'TS011F', manufacturerName: '_TZ3000_kdi2o9m6'}, // EU
            {modelID: 'TS011F', manufacturerName: '_TZ3000_plyvnuf5'}, // CH
            {modelID: 'TS011F', manufacturerName: '_TZ3000_wamqdr3f'}, // FR
            {modelID: 'TS011F', manufacturerName: '_TZ3000_00mk2xzy'}, // BS
        ],
        model: 'HG06337',
        vendor: 'Lidl',
        description: 'Silvercrest smart plug (EU, CH, FR, BS)',
        extend: generic.switch,
    },
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(11);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint);
        },
```

Truddel, 2020-12-11:
@Koenkk Works like a charm. Thanks!

Needed to change the brakets a little bit but in general it works.
```
{
        fingerprint: [
            {modelID: 'TS011F', manufacturerName: '_TZ3000_kdi2o9m6'}, // EU
            {modelID: 'TS011F', manufacturerName: '_TZ3000_plyvnuf5'}, // CH
            {modelID: 'TS011F', manufacturerName: '_TZ3000_wamqdr3f'}, // FR
            {modelID: 'TS011F', manufacturerName: '_TZ3000_00mk2xzy'}, // BS
        ],
        model: 'HG06337',
        vendor: 'Lidl',
        description: 'Silvercrest smart plug (EU, CH, FR, BS)',
        extend: generic.switch,
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(11);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint);
        },
        },

```

Debug log fyi:

```
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,111,26,1,41,145,72] +2m

  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,111,26,1,41,145,72] +1ms

  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [111,26,1,41,145] - 72 +0ms

  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":6767,"relaycount":1,"relaylist":[37161]} +2m

  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms

  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,111,26,11,1,0,23,0,121,31,110,0,0,7,24,35,10,0,0,16,1,41,145,28,59] +46ms

  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,111,26,11,1,0,23,0,121,31,110,0,0,7,24,35,10,0,0,16,1,41,145,28,59] +0ms

  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,111,26,11,1,0,23,0,121,31,110,0,0,7,24,35,10,0,0,16,1,41,145,28] - 59 +1ms

  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":6767,"srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"linkquality":23,"securityuse":0,"timestamp":7217017,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,35,10,0,0,16,1]}} +48ms

  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":35,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":1}]},"address":6767,"endpoint":11,"linkquality":23,"groupID":0}' +2m

Zigbee2MQTT:debug 2020-12-11 20:46:00: Received Zigbee message from 'Silvercrest01', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID 0

  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +5ms

Zigbee2MQTT:info  2020-12-11 20:46:00: MQTT publish: topic 'zigbee2mqtt/Silvercrest01', payload '{"linkquality":23,"state":"ON"}'

```

brozikcz, 2020-12-11:
It works for me too.

@Koenkk can you please explain why does the changing of endpoint number work?

Truddel, 2020-12-12:
@brozikcz I think this is that simple: 
[image]

HG06337 is using Endpoint ID 11 instead of 1.

Koenkk, 2020-12-13:
@Truddel is right, it has no endpoint 1 but 11.

Integrated this now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-13 by Koenkk