Koenkk/zigbee2mqtt #5207: Xiaomi switch status message is reset immediately, no switching possible

salopette, 2020-12-06:
Hello, since the last update, the status signals are only transmitted very briefly, so that switching is not possible.

In the frontend it is only displayed briefly and is immediately empty again.
It does not stop at the click or dooble click.

[image]
[image]
[image]
[image]

salopette, 2020-12-06:
I discovered the bug
it only happens when I activate home assistant support.

When it is turned off, it works normally.

[image]

Koenkk, 2020-12-06:
This is expected behaviour, for Home Assistant to trigger the action property needs to be cleared again, otherwise automations don't trigger. See https://github.com/Koenkk/zigbee2mqtt/issues/1155#issuecomment-467094619

closed 2020-12-06 by Koenkk