Koenkk/zigbee2mqtt #5208: Silvercrest HG06337 Smart Plug (switch) UK version

nottledim, 2020-12-06:
I added this to fingerprint array in devices.js for Silvercrest HG06337.  I think it's different because it's UK version first sold 6-Dec-2020 
{modelID: 'TS011F', manufacturerName: '_TZ3000_00mk2xzy'}

It seems to work

rock-lobster, 2020-12-06:
Ahha, someone beat me to it. Picked up a few this afternoon and yep, different manufacturer name (feels wrong, should be a custom model but hey ho).

Zigbee model 'TS011F' and manufacturer name '_TZ3000_00mk2xzy' is NOT supported

Koenkk, 2020-12-06:
Just supported.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-06 by Koenkk

haggard71, 2020-12-07:
I see Silvercrest smart plug (EU, CH, FR, BS) now, but no sub devices are created.....

johnhamelink, 2020-12-07:
I've just moved to zigbee2mqtt edge in Home Assistant and my switches now pair nicely!

LifeBandit666, 2020-12-14:
Still not supported for me using HA addon Z2M Edge. Do I need to do anything extra (it should be auto updating and I have no updates).

LifeBandit666, 2020-12-14:
its an HG06337-bs

rock-lobster, 2020-12-14:
are you running the latest edge? It doesn't (AFAIK) auto update on edge - the readme says:

**Updating the Edge add-on**
To update the edge version of the add-on, you will need to uninstall and re-install the add-on.

so if you've not updated for a while, it might be worth backing up the conf, uninstalling and reinstalling and then trying again after resinstalling latest edge. This is what I now see (and the version)

[image]

LifeBandit666, 2020-12-14:
That did it, turns out it was a PICNIC (Problem In Chair Not In Computer) error again. Thanks rock-lobster