Koenkk/zigbee2mqtt #5210: Tuya Zigbee smart LED light not supported

ROEEROEEROEE, 2020-12-06:
## Information about the device + link
the bulb is a Tuya Zigbee smart LED light bulb RGB+CCT E27 220V 
link :+1: 
https://www.aliexpress.com/item/1005001368916827.html?spm=a2g0o.productlist.0.0.dee056084IEHpW&algo_pvid=09db11e5-8f2f-46e3-8b75-8bcdcbade1f0&algo_expid=09db11e5-8f2f-46e3-8b75-8bcdcbade1f0-1&btsid=2100bdd016072847296746148e2960&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_
..
i've try to add new device, yet, the most i could see is the lqi 
[image]

## data/database.db entry of the device
..INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0022414bae","ieee_address":"0x00124b0022414bae"},"type":"device_announce"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b0022414bae"},"type":"device_announced"}'

i have also succeed to see that the manufactor is 
manufacturerName: "JunYu"}],
 zigbeeModel: ['ZB-RGBCW'] 

thanks in advanced !

Koenkk, 2020-12-06:
This device is already supported in the dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-06 by Koenkk

ROEEROEEROEE, 2020-12-06:
hi, thanks, 
any idea when it would be available within the regular addon?

Koenkk, 2020-12-07:
1 January (next z2m release)