Koenkk/zigbee2mqtt #5211: Supporting the HOMA1001

TorainX, 2020-12-06:
This device was already discussed in https://github.com/Koenkk/zigbee2mqtt/issues/3594
Unfortunately it stucked.

I'm using latest version of Z2M (1.16.2.2) on home assistant Raspberry.
followed the instructions for adding new devices, but now stuck at the same point as https://github.com/kalsan who opened the former ticket.

## Information about the device + link
https://www.isolicht.com/led-beleuchtung/installationsmaterial/dimmer-schalter-schutz/zigbee-zll-4in1-4-kanal-controller-dimmer-fur-led-flexbander-und-led-leuchtmitte

..

## data/database.db entry of the device
..{"id":6,"type":"Router","ieeeAddr":"0x00124b001eb0821f","nwkAddr":64989,"manufId":0,"manufName":"ShenZhen_Homa","powerSource":"Mains (single phase)","modelId":"HOMA1001","epList":[10,11,13],"endpoints":{"10":{"profId":49246,"epId":10,"devId":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"HOMA1001","manufacturerName":"ShenZhen_Homa","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.2"}}},"binds":[],"meta":{}},"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{},"binds":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.2","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1607253767573}

If you need further information logs etc. let me know.
Thanks in advance!

TorainX, 2020-12-11:
Would be really great if anybody at least could guide me, what and where could be the next step to investigate what kind of changes/implementation has to be done.
Thanks in advance!

Koenkk, 2020-12-11:
I will reopen #3594 and continue there.

closed 2020-12-11 by Koenkk