Koenkk/zigbee2mqtt #5212: Interview fails for Orvibo SM10ZW

erikman, 2020-12-06:
## What happened
I try to pair my Orvibo SM10ZW window/door sensor https://www.zigbee2mqtt.io/devices/SM10ZW.html but the interview fails.

## What did you expect to happen
Interview should succeed

## How to reproduce it (minimal and precise)
Try to pair Orvibo SM10ZN device, have failed in same manner 10 times in a row. Interview succeeds for other devices I have tried with.

## Debug info
Zigbee2MQTT version: Zigbee2MQTT version 1.16.2 (commit `#04c15f7`), zigbee-herdsman (0.13.37)
Adapter hardware: CC2652R1 Electrolama zig-a-zig-ah! (zzh!) 
Adapter firmware version: 20201026 from https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

Herdsman logs for pairing: https://pastebin.com/X27UFmaA (device is `0x00124b000945a6e3` with friendly_name `front_door`)

Koenkk, 2020-12-07:
Besides the interview failing, does the device work correctly?

erikman, 2020-12-08:
No, there is nothing published when I trigger the sensor

Koenkk, 2020-12-08:
Can you try changing the definition to below (does not fix interview failure but might fix no events being received).

```js
{
    zigbeeModel: ['fdd76effa0e146b4bdafa0c203a37192', 'c670e231d1374dbc9e3c6a9fffbd0ae6'],
    model: 'SM10ZW',
    vendor: 'ORVIBO',
    description: 'Door or window contact switch',
    fromZigbee: [fz.ias_contact_alarm_1, fz.battery],
    toZigbee: [],
    exposes: [e.contact(), e.battery_low(), e.tamper(), e.battery()],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.write('ssIasZone', {'iasCieAddr': coordinatorEndpoint.getDevice().ieeeAddr});
        await endpoint.command('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, {disableDefaultResponse: true});
    },
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

erikman, 2020-12-09:
Thank you for your support!
I still can't see any events. Here is a new pairing log with the updated `devices.js` file: https://pastebin.com/w76ucdjW

The way I interpret the logs is that something goes wrong when reading `["iasCieAddr","zoneState"],` which in the latest log is initiated on row 297, then on line 318 it looks like it receives the response but it is ignored according to line 319, and then the read fails, starting on line 321. Do you agree? Is it a bug that the response is ignored on line 319?

Koenkk, 2020-12-10:
The device is acting very strange, it stays the zoneState is an unsupported_attribute but later it responds with it (looks like a bug in the device firmware). We will fix the failed to interview later. 

Can you provide the log when waking up the device by clicking a button on it and directly after that configure it via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure ?

erikman, 2020-12-11:
doesn't really give any log, I just get this output:

```
zigbee2mqtt_1    | Zigbee2MQTT:warn  2020-12-11 22:07:48: Skipping configure of 'front_door', device does not require this.
```

This is the entry I have in `database.db`, the device is never announced to home-assistant:

```json
{
  "id": 2,
  "type": "EndDevice",
  "ieeeAddr": "0x00124b000945a6e3",
  "nwkAddr": 58307,
  "manufId": 0,
  "manufName": "欧瑞",
  "powerSource": "Battery",
  "modelId": "75a4bfe8ef9c4350830a25d13e3ab068",
  "epList": [
    1
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 1026,
      "inClusterList": [
        0,
        1,
        3,
        1280
      ],
      "outClusterList": [
        3
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "75a4bfe8ef9c4350830a25d13e3ab068",
            "manufacturerName": "欧瑞",
            "powerSource": 3,
            "zclVersion": 1,
            "appVersion": "v1.12.151128             ",
            "hwVersion": 1,
            "dateCode": "20151128        "
          }
        },
        "ssIasZone": {
          "attributes": {
            "zoneState": 0
          }
        }
      },
      "binds": [],
      "meta": {}
    }
  },
  "appVersion": "v1.12.151128             ",
  "hwVersion": 1,
  "dateCode": "20151128        ",
  "zclVersion": 1,
  "interviewCompleted": false,
  "meta": {},
  "lastSeen": 1607720851802
}
```

Koenkk, 2020-12-13:
@erikman that means you didn't correctly update devices.js according to https://github.com/Koenkk/zigbee2mqtt/issues/5212#issuecomment-740808796

erikman, 2020-12-14:
I think I found why, my device has a `modelId` that isn't listed in `devices.js`. I've updated and now I can see activity from the device when triggering it (writing `iasCieAddr` returns a failure-code, but it still seems like it was applied, like you said it looks like really buggy firmware). I'll add logs later.

erikman, 2020-12-20:
Ok, so I'm back at being lost :)
I tried adding following hack into `device.ts` in zigbee-herdsman, in `interviewInternal` before `Enroll IAS device` (sorry about the spelling in the log messages):
```ts
        // Goodbyte hack
        for (const endpoint of this.endpoints.filter((e): boolean => e.supportsInputCluster('ssIasZone'))) {
            const coordinator = Device.byType('Coordinator')[0];

            debug.log('Executig configure for SM10ZW');
            // const endpoint = device.getEndpoint(1);

            await Wait(500);
            debug.log('Executig configure for SM10ZW, coordinator: ', coordinator.ieeeAddr);
            await endpoint.write('ssIasZone', {'iasCieAddr': coordinator.ieeeAddr});

            debug.log('Executig configure for SM10ZW, enrollRsp:');
            await Wait(500);
            await endpoint.command('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, {disableDefaultResponse: true});
        }
```

this results in a working device (pairing log here: https://pastebin.com/WyqFvUDb) meaning I can see it in home-assistant and triggering the sensor results in this output (correctly reflected in ha):
```
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,29,68,129,0,0,0,5,131,183,1,10,0,144,0,138,45,194,0,0,9,25,1,0,32,1,0,23,0,0,131,183,29,25] +58s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,29,68,129,0,0,0,5,131,183,1,10,0,144,0,138,45,194,0,0,9,25,1,0,32,1,0,23,0,0,131,183,29,25] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 29 - 2 - 4 - 129 - [0,0,0,5,131,183,1,10,0,144,0,138,45,194,0,0,9,25,1,0,32,1,0,23,0,0,131,183,29] - 25 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1280,"srcaddr":46979,"srcendpoint":1,"dstendpoint":10,"wasbroadcast":0,"linkquality":144,"securityuse":0,"timestamp":12725642,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[25,1,0,32,1,0,23,0,0]}} +58s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":1,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"zonestatus":288,"extendedstatus":0}},"address":46979,"endpoint":1,"linkquality":144,"groupID":0}' +59s
Zigbee2MQTT:info  2020-12-20 22:40:49: MQTT publish: topic 'zigbee2mqtt/front_door', payload '{"battery_low":false,"contact":true,"device":{"applicationVersion":"v1.12.151128             ","dateCode":"20151128        ","friendlyName":"front_door","hardwareVersion":1,"ieeeAddr":"0x00124b000945a6e3","manufacturerID":0,"manufacturerName":"欧瑞","model":"SM10ZW","networkAddress":46979,"powerSource":"Battery","type":"EndDevice","zclVersion":1},"linkquality":144,"tamper":false}'
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +6ms
```

but when I try with only your modification to `devices.js` in `zigbee-herdsman-converters` then I don't get events when the device is triggered. (I've tried adding delays and catch the write errors to make sure all steps are executed even if the device reports errors like this):

```js
    {
        zigbeeModel: ['fdd76effa0e146b4bdafa0c203a37192', 'c670e231d1374dbc9e3c6a9fffbd0ae6', '75a4bfe8ef9c4350830a25d13e3ab068'],
        model: 'SM10ZW',
        vendor: 'ORVIBO',
        description: 'Door or window contact switch',
        fromZigbee: [fz.ias_contact_alarm_1, fz.battery],
        toZigbee: [],
        exposes: [e.contact(), e.battery_low(), e.tamper(), e.battery()],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);

            try {
                await endpoint.write('ssIasZone', {'iasCieAddr': coordinatorEndpoint.getDevice().ieeeAddr});
            }
            catch(err) {
                console.log('write ssIasZone:iasCieAddr failed', err.message);
            }

            await new Promise(resolve => setTimeout(() => resolve(), 500))

            try {
                await endpoint.command('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, {disableDefaultResponse: true});
            }
            catch(err) {
                console.log('send ssIasZone:enrollRsp command failed', err.message);
            }
        },
    },
```
pairigin log for this (without any `devices.ts` changes) is here: https://pastebin.com/pNCYdAzy

from what I understand the only difference is the order it works when I just write `iasCieAddr` (and then abort since it will result in failure), but the device don't work if I first read `iasCieAddr` (which fails) and then try to write the `iasCieAddr`. Do you agree, and do you have any suggestions on how to get it working without my `devices.ts` hack (which I would like to avoid)?

Koenkk, 2020-12-21:
The only difference I see is 1x write via configure vs 2x with herdsman so I suggest trying with:

```js
    {
        zigbeeModel: ['fdd76effa0e146b4bdafa0c203a37192', 'c670e231d1374dbc9e3c6a9fffbd0ae6', '75a4bfe8ef9c4350830a25d13e3ab068'],
        model: 'SM10ZW',
        vendor: 'ORVIBO',
        description: 'Door or window contact switch',
        fromZigbee: [fz.ias_contact_alarm_1, fz.battery],
        toZigbee: [],
        exposes: [e.contact(), e.battery_low(), e.tamper(), e.battery()],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);

            try {
                await endpoint.write('ssIasZone', {'iasCieAddr': coordinatorEndpoint.getDevice().ieeeAddr});
            }
            catch(err) {
                console.log('write ssIasZone:iasCieAddr failed', err.message);
            }
 await new Promise(resolve => setTimeout(() => resolve(), 300))

 try {
                await endpoint.write('ssIasZone', {'iasCieAddr': coordinatorEndpoint.getDevice().ieeeAddr});
            }
            catch(err) {
                console.log('write ssIasZone:iasCieAddr failed', err.message);
            }

            await new Promise(resolve => setTimeout(() => resolve(), 500))

            try {
                await endpoint.command('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, {disableDefaultResponse: true});
            }
            catch(err) {
                console.log('send ssIasZone:enrollRsp command failed', err.message);
            }
        },
    },
```

erikman, 2020-12-28:
Thanks for your support :smile: 
I've been digging more in this and it turns out I was looking in the wrong place. I've been using slightly different versions while testing (`zigbee2mqtt from docker:1.16.2`, vs. `zigbee2mqtt:master` with `zigbee-herdsman:master` and `zigbee-herdsman-converters:master`), and it turns out there is some change in `zigbee-herdsman` between `v0.13.37` and `master` that makes the device publish changes when triggered (and then there is also the buggy firmware making `iasCieAddr` not being written as expected).

So I'll wait for a version bump on `zigbee-herdsman`, and prepare a PR for `zigbee-herdsman-converters` with the added `zigbeeModel`, this is what I currently use:

```javascript
    {
        zigbeeModel: ['fdd76effa0e146b4bdafa0c203a37192', 'c670e231d1374dbc9e3c6a9fffbd0ae6', '75a4bfe8ef9c4350830a25d13e3ab068'],
        model: 'SM10ZW',
        vendor: 'ORVIBO',
        description: 'Door or window contact switch',
        fromZigbee: [fz.ias_contact_alarm_1, fz.battery],
        toZigbee: [],
        exposes: [e.contact(), e.battery_low(), e.tamper(), e.battery()],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);

            // We check for buggy firmware where the interview fails when we try to read iasCieAddr.
            try {
                await endpoint.read('ssIasZone', ['iasCieAddr', 'zoneState']);
            }
            catch(err) {
                // The interview has failed and we haven't written iasCieAddr, as a workaround we therefore write iasCieAddr,
                // which will report a failure but still update the device.
                console.log('Reading ssIasZone:iasCieAddr, zoneState failed, writing iasCieAddr as workaround for buggy firmware, message:', err.message);

                try {
                    await endpoint.write('ssIasZone', {'iasCieAddr': coordinatorEndpoint.getDevice().ieeeAddr});
                }
                catch(err) {
                    console.log('Expected failure to write ssIasZone:iasCieAddr due to buggy firmware (it still updates the device), message:', err.message);
                }

                await new Promise(resolve => setTimeout(() => resolve(), 500));
                await endpoint.command('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, {disableDefaultResponse: true});
            }
        },
    },
```

du you have any preferences on how the PR should look, for instance should it include `console.log` or do you think the whole piece is to hacky so I should just add the new `zigbeeModel`?

Koenkk, 2020-12-28:
In the latest dev branch I've added a logger to configure, so signature now becomes: `configure: async (device, coordinatorEndpoint,logger) => {`. You can call `logger.info('hello world')` to log. Rest looks good so please create a PR. I guess the interview still fails? If so please also create a PR to update the docs:  https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/LVS-SM10ZW.md

closed 2021-02-05 by github-actions[bot]