Koenkk/zigbee2mqtt #5213: Ikea Tradfri E1702 Plug Power On behavior

AndyZuerich, 2020-12-06:
Hi

zigbee2mqtt seems to support the power on behavior (on/off/remember state) feature of Philips Hue and some other devices. However, for the Ikea smart plugs, this is not (yet) supported. I used deconz until I bricked my Conbee stick with a firmware update. In Deconz, setting the power on behavior of the Ikea plug was available. Since I did rely on this feature in my home applications (and Smart-Home-Wife-compatibility), I would be glad if this will be supported, soon.

I am software developer and EE, so in case you need some more information or even help with developing / testing, I could contribute.

Koenkk, 2020-12-07:
Implemented! You can set it via the frontend by going to the device -> exposes -> power on behaviour. (https://www.zigbee2mqtt.io/information/frontend.html)

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-07 by Koenkk

AndyZuerich, 2020-12-13:
I just had the time to test the new power-on behavior setting for the Trafri plug. Well, yes, basically the feature does work now. However, the setting that I would need is to have it power "on" at startup. In the web interface as well as via mosquitto_pub, only the two options "previous" and "off" are available, so the third possible state "on" is still missing.

I am sorry to bother you with this detail. Especially in case you have a (old fashioned) lamp attached to the plug, the third option would be important.

AndyZuerich, 2020-12-13:
Hello Koen

I commented on the power-on feature of the ikea outlets. I did fix this 
now myself just by adding the state "on": 1 within toZigbee.js , and it 
is working :-)

Up to now, I did never did work in JavaScript before, and I was 
surprised when npm ci overwrote my changes until I realized that the .js 
files are the output of npm call (where is the original code then coming 
from?) I am more in Java / Scala / Haskell / Ada than JavaScript . 
Nevertheless: It would be nice if you could add the "on" state for the 
power-on-behavior to your branch, too.

Kind regards and stay healthy!

Andreas

Koenkk, 2020-12-14:
Just fixed by @Kryzek  in https://github.com/Koenkk/zigbee-herdsman-converters/pull/1907 , thanks @Kryzek