Koenkk/zigbee2mqtt #5214: Blitzwolf door sensor not recognized

rodrigorsdev, 2020-12-07:
The BlitzWolf Rechargeable Zigbee contact sensor is available on compatible devices, but is not recognized.

Here it is the database:

{"id":10,"type":"EndDevice","ieeeAddr":"0xec1bbdfffe6f3096","nwkAddr":59051,"manufId":4098,"epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,10,1,1280],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0013\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0013\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0013\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0013","65506":21,"appVersion":66}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":22}}},"binds":[],"meta":{}}},"appVersion":66,"interviewCompleted":false,"meta":{},"lastSeen":1607299036695}

thanks

Koenkk, 2020-12-08:
What is the model of this device?

rodrigorsdev, 2020-12-09:
BlitzWolf Rechargeable Zigbee contact sensor, BW-IS2

Koenkk, 2020-12-09:
Added.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-09 by Koenkk