Koenkk/zigbee2mqtt #5218: Devices suddenly leave the network

salopette, 2020-12-07:
Hello, I have had the problem since yesterday that some devices say goodbye to the network.
Are simply set to Unsuported.
Up to now 2 Hue motion detectors and 2 Osram Smart lamps

zigbee2mqtt version
1.16.2-dev commit: 92b942e
coordinator type
zStack3x0
coordinator revision
20201026
frontend version
0.2.100

At Hue, these are motion detectors

[image]

kevincaradant, 2020-12-07:
Hi

What's the logs are saying about this issue?

When you said motion and bulbs become unavailable, is it because motion sensors (end-devices) are (was) connected on the bulb which is a router and by consequence, these end devices become unavailable because the router left the network?

Not sure it could help but if this issue happens often for you, maybe you could try to add the option `availability_timeout: 60` in the `configuration.yaml` and see if it's better :). 

Don't forget to enable the debug mode of  Z2M. That will be more easy maybe for developers here to see what's happens.

salopette, 2020-12-08:
> Hallo
> 
> Was sagen die Protokolle zu diesem Problem?
> 
> Wenn Sie sagten, dass Bewegung und Lampen nicht mehr verfügbar sind, liegt dies daran, dass Bewegungssensoren (Endgeräte) an der Lampe angeschlossen sind (waren), die ein Router ist, und diese Endgeräte folglich nicht mehr verfügbar sind, weil der Router das Netzwerk verlassen hat?
> 
> Ich bin mir nicht sicher, ob es helfen könnte, aber wenn dieses Problem bei Ihnen häufig auftritt, könnten Sie vielleicht versuchen, die Option `availability_timeout: 60`im hinzuzufügen `configuration.yaml`und zu sehen, ob es besser ist :).
> 
> Vergessen Sie nicht, den Debug-Modus von Z2M zu aktivieren. Das wird für Entwickler hier vielleicht einfacher zu sehen sein, was passiert.

> Hi
>
> What's the logs are saying about this issue?
>
> When you said motion and bulbs become unavailable, is it because motion sensors (end-devices) are (was) connected on the bulb which is a router and [...]

Hello, no the motion sensor was not connected to the lamp. He was the first to leave the network and later the lamp.
I don't think it has anything to do with that.

Anjerlaan, 2020-12-09:
Did you recently replaced the coordinator (by a e.g ZZH)...and were the devices paired with the old coordinator?

salopette, 2020-12-09:
no I don't

Anjerlaan, 2020-12-09:
Ok..I had this leaving network issue (with end point devices) when I changed the coordinator.
Did you change to newer firmware then, but devices paired with old firmware?

salopette, 2020-12-09:
No, I haven't made any changes, everything has been running smoothly for a month or more!

skumka, 2020-12-10:
https://github.com/Koenkk/zigbee2mqtt/issues/5247

I am sure some of the recent changes all routers problems with discovery, joining devices in the network and random disconnection of devices from routers.

closed 2021-01-17 by github-actions[bot]