Koenkk/zigbee2mqtt #5220: Unable to connect Saswell SEA801-DF1

lukasdietz, 2020-12-07:
Hey there, 
I am new to zigbee2mqtt but read the documentation to get started. 
However I am unable to connect my Saswell SEA801-DF1 to my Bridge. I am using a usb extension cable. 

# Bug Report
## What happened
Tried to connect SEA801 thermostat but it did not succeed. 

## What did you expect to happen
Start zigbee2mqtt and get device in pairing Mode. Should connect automaticly. 

## How to reproduce it (minimal and precise)
Open Logs and press Auto and + on Thermostat until ---- is shown in the Display. It should connect but did not. 

## Debug Info
Zigbee2MQTT version: 1.16.2
Adapter hardware: Cc2531 
Adapter firmware version: default_20190607

I hope some of you can help me out. 

With best regards 
Lukas

LouiS22, 2020-12-08:
DF-01 is wifi based!

For Zigbee, you need SEA801-Z01 (where Z is for Zigbee).

lukasdietz, 2020-12-08:
Oh man thanks for your answer. I did not now about multiple versions. Do you know if it is possible to use this Version without a gateway?

LouiS22, 2020-12-09:
I doubt. Better get a replacement, if you can.

lukasdietz, 2020-12-09:
Thanks for your support! So I will purchase a Z01 one.

closed 2020-12-09 by lukasdietz