Koenkk/zigbee2mqtt #5221: Support for Lidl - Malinera "Smart LED String Lights"

olialb, 2020-12-07:
Hey Guys,

Christmas is coming and I ordered the new Zigbee compatible "Smart LED String Light" from Lidl...
I was able to add the device (Tuya TS0601). But if I press any button nothing is seen in the debug log. Only if I reset the device its gives some info where I would appreciate your help... Its first tiome for me

The device has the following finctionality (from product description)
- Set color
- Set temperature
- Switch between 16 effects

I read the this, but unfortunately I am bit lost....

## Information about the device + link
Lidl Malinera "Smart LED String Lights"
https://www.lidl.de/de/melinera-lichterkette-zigbee-smart-home/p360021

## data/database.db entry of the device
`{"id":86,"type":"Router","ieeeAddr":"0xbc33acfffe733e84","nwkAddr":30823,"manufId":4098,"manufName":"_TZE200_s8gkrkxk","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,4,5,3,6,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_s8gkrkxk","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":258,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1607372615664}`

I added this to decices.js:

`const devices = [
    // Lidl (Malinera)
    {
        fingerprint: [ // Since a lot of Tuya devices use the same modelID, but use different data points
                       // it's usually necessary to provide a fingerprint
            {
                modelID: 'TS0601', // The model ID from: Device with modelID 'TS0601' is not supported.
                                   // You may need to add \u0000 at the end of the name in some cases
                manufacturerName: '_TZE200_s8gkrkxk' // The manufacturer name from: Device with modelID 'TS0601' is not supporte$            },
        ],
        model: 'SmartLEDStringLights',
        vendor: 'Lidl',
        description: 'Smart LED String Light',
        supports: 'color, temperature',
        fromZigbee: [
           fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
           fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
        ],
        toZigbee: [
            tz.tuya_data_point_test, // Another debug converter
        ],
        onEvent: tuya.setTime, // Add this if you are getting no converter for 'commandSetTimeRequest'
        meta: {
            configureKey: 1,
        },
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genBasic']);
        },
        exposes: [
            // Here you should put all functionality that your device exposes
        ],
    },
`

Now I get this messages in the debug log when I reset the device:

`info  2020-12-07 21:46:16: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"ed8b4e5","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":false,"version":"1.15.0"}'
debug 2020-12-07 21:46:45: Received Zigbee message from '0xbc33acfffe733e84', type 'attributeReport', cluster 'genBasic', data '{"appVersion":70}' from endpoint 1 with groupID 0
error 2020-12-07 21:46:45: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at resolvedEntity.definition.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/receive.js:120:42)
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:119:65)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:371:44)
    at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:260:14)
    at Zigbee.emit (events.js:198:13)
    at Controller.herdsman.on (/opt/zigbee2mqtt/lib/zigbee.js:68:52)
    at Controller.emit (events.js:198:13)
    at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:544:22)
    at Generator.next (<anonymous>))
debug 2020-12-07 21:46:48: Received Zigbee message from '0xbc33acfffe733e84', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[1],"type":"Buffer"},"dp":257,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID 0
error 2020-12-07 21:46:48: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at resolvedEntity.definition.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/receive.js:120:42)
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:119:65)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:371:44)
    at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:260:14)
    at Zigbee.emit (events.js:198:13)
    at Controller.herdsman.on (/opt/zigbee2mqtt/lib/zigbee.js:68:52)
    at Controller.emit (events.js:198:13)
    at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:544:22)
    at Generator.next (<anonymous>))
debug 2020-12-07 21:46:48: Received Zigbee message from '0xbc33acfffe733e84', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0,0,0,0],"type":"Buffer"},"dp":519,"fn":0,"status":0,"transid":5}' from endpoint 1 with groupID 0
error 2020-12-07 21:46:48: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at resolvedEntity.definition.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/receive.js:120:42)
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:119:65)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:371:44)
    at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:260:14)
    at Zigbee.emit (events.js:198:13)
    at Controller.herdsman.on (/opt/zigbee2mqtt/lib/zigbee.js:68:52)
    at Controller.emit (events.js:198:13)
    at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:544:22)
    at Generator.next (<anonymous>))
debug 2020-12-07 21:46:48: Received Zigbee message from '0xbc33acfffe733e84', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[1],"type":"Buffer"},"dp":257,"fn":0,"status":0,"transid":3}' from endpoint 1 with groupID 0
error 2020-12-07 21:46:48: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at resolvedEntity.definition.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/receive.js:120:42)
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:119:65)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:371:44)
    at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:260:14)
    at Zigbee.emit (events.js:198:13)
    at Controller.herdsman.on (/opt/zigbee2mqtt/lib/zigbee.js:68:52)
    at Controller.emit (events.js:198:13)
    at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:544:22)
    at Generator.next (<anonymous>))
`

How to continue from this point!? 
Have a peaceful pre christmas time!

Koenkk, 2020-12-08:
Already supported in 1.16.2: https://www.zigbee2mqtt.io/devices/HG06467.html

closed 2020-12-08 by Koenkk

olialb, 2020-12-08:
Cool Thanks!