Koenkk/zigbee2mqtt #5223: Error adding reporting for device (PTVO)

sdhawade, 2020-12-08:
Pairing my custom device based on PTVO firmware results in error. I have configured my firmware without input 1, do we need all inputs configured ?

## What did you expect to happen
Succesful pairing with result reporting for all items

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:  1.16.2
Adapter hardware: Conbee II
Adapter firmware version: 0x26660700

Zigbee2MQTT:info  2020-12-08 10:46:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b001fc64555"},"type":"pairing"}'
Zigbee2MQTT:info  2020-12-08 10:46:21: MQTT publish: topic 'zigbee2mqtt/0x00124b001fc64555/availability', payload 'online'
Zigbee2MQTT:info  2020-12-08 10:47:29: Successfully interviewed '0x00124b001fc64555', device has successfully been paired
Zigbee2MQTT:info  2020-12-08 10:47:29: Device '0x00124b001fc64555' is supported, identified as: Custom devices (DiY) [Multi-channel relay switch](https://ptvo.info/zigbee-switch-configurable-firmware-router-199/) (ptvo.switch)
Zigbee2MQTT:info  2020-12-08 10:47:30: Successfully setup reporting for '0x00124b001fc64555' - 2 - genOnOff
Zigbee2MQTT:info  2020-12-08 10:47:31: Successfully setup reporting for '0x00124b001fc64555' - 3 - genOnOff
Zigbee2MQTT:info  2020-12-08 10:47:32: Successfully setup reporting for '0x00124b001fc64555' - 4 - genOnOff
Zigbee2MQTT:info  2020-12-08 10:47:33: Successfully setup reporting for '0x00124b001fc64555' - 5 - genOnOff
Zigbee2MQTT:error 2020-12-08 10:47:34: Failed to setup reporting for '0x00124b001fc64555' - Error: Bind 0x00124b001fc64555/6 genOnOff from '0x00212effff054345/1' failed (Error: status: 140)
    at DeconzAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:660:23)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:24:58)
Zigbee2MQTT:info  2020-12-08 10:47:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"[Multi-channel relay switch](https://ptvo.info/zigbee-switch-configurable-firmware-router-199/)","friendly_name":"0x00124b001fc64555","model":"ptvo.switch","supported":true,"vendor":"Custom devices (DiY)"},"type":"pairing"}'

Koenkk, 2020-12-08:
@ptvoinfo can you check this?

ptvoinfo, 2020-12-08:
@sdhawade, Could you please show your configuration in my firmware? I'll check it against Zigbee2MQTT.

sdhawade, 2020-12-08:
> @sdhawade, Could you please show your configuration in my firmware? I'll check it against Zigbee2MQTT.
Thanks for the response, Here it is : Using the v2.5 November build
Board type: CC2530 + CC2591
Device type: Router
Update firmware's timestamp : 2020-12-06
Power saving mode (PSM): No
Default reporting interval (s): 60

Output pins:
P10: Output 2, GPIO, Pull-up, Remember state
P12: Output 3, GPIO, Pull-up, Remember state
P13: Output 4, GPIO, Pull-up, Remember state
P15: Output 5, GPIO, Pull-up, Remember state
P16: Output 6, GPIO, Pull-up, Remember state
P17: Output 7, GPIO, Pull-up, Remember state
P20: Output 8, GPIO, Pull-up, Remember state

Input pins:
P00: Input 2, GPIO, Pull-up, Switch, Toggle, Link to out 2
P01: Input 3, GPIO, Pull-up, Switch, Toggle, Link to out 3
P02: Input 4, GPIO, Pull-up, Switch, Toggle, Link to out 4
P03: Input 5, GPIO, Pull-up, Switch, Toggle, Link to out 5
P04: Input 6, GPIO, Pull-up, Switch, Toggle, Link to out 6
P05: Input 7, GPIO, Pull-up, Switch, Toggle, Link to out 7
P06: Input 8, GPIO, Pull-up, Switch, Toggle, Link to out 8

ptvoinfo, 2020-12-08:
@sdhawade 
I cannot repeat this problem with the latest version of the firmware and the latest dev build of Zigbee2MQTT. But I don't have Conbee II.

sdhawade, 2020-12-08:
@Koenkk  Is this a Conbee II issue ? Do I need to enable any flags to get more debug info for this ?

ptvoinfo, 2020-12-09:
@shades684 Try the "adapter_concurrent: 2" option in your configuration for Conbee II.
https://www.zigbee2mqtt.io/information/configuration.html

sdhawade, 2020-12-10:
> @shades684 Try the "adapter_concurrent: 2" option in your configuration for Conbee II.
> https://www.zigbee2mqtt.io/information/configuration.html
Thanks for the suggestion, there is co change in the behaviour. It fails in the same way

fi-sch, 2020-12-18:
@sdhawade Did you manage to get it work? Did you have problem with this one configuration?
I am curious if firmware with simple outputs would work (ConBee + zigbee2mqtt) as I use a couple of those and plan to move from deCONZ. I have some crucial devices controlled by CC2530+CC2591 + PTVO firmware (e.g. garage door), so it wouldn't be if I break my setup during migration :(

sdhawade, 2020-12-26:
> @sdhawade Did you manage to get it work? Did you have problem with this one configuration?
> I am curious if firmware with simple outputs would work (ConBee + zigbee2mqtt) as I use a couple of those [...]

It is still failing consistently. I plan to try this with the updated release of the Deconz adapter. Will check with the latest release and update.

closed 2021-02-03 by github-actions[bot]