Koenkk/zigbee2mqtt #5224: Lonsonho dimmer delay or no response

thve25, 2020-12-08:
# Bug Report

How to solve this problem? Is it the cc2531 stick? Should i buy a better zigbee stick?

## What happened
Dimmer doesn't turn on/off the light or with very long delay (sometimes 30s)

Dimmer: Lonsonho QS-Zigbee-D02-TRIAC-LN

## How to reproduce it (minimal and precise)
It happens always withing 12 hours after a reboot, or when toggling frequently.

## Debug Info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608 

log file when it did not wanted to turn on:
https://pastebin.com/raw/4wuUdPmE

log file when it turned off after a 10 second delay:
https://pastebin.com/raw/YP7Us2hu

Koenkk, 2020-12-08:
Can you try with the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing  ?

thve25, 2020-12-13:
I flashed the stick with the source routing firmware and now the problem seems to be fixed. The dimmer switches immediately. 

I now have some more devices and connected them via routers. Still no issues. 

Is there any other downside to  the source routing firmware, other than the forced use of routers?

Koenkk, 2020-12-13:
The only downside is that it can handle less direct children (5 instead of 20) so more routers are needed, but it will give you better performance and stability.

Closing this as issue seems to be fixed.

closed 2020-12-13 by Koenkk