Koenkk/zigbee2mqtt #5225: State change of Group not being acted on by group member devices

LucidityCrash, 2020-12-08:
## What happened
Physical State of switches not changed of devices in Group

Debug Logs here : https://hastebin.com/ixokidegam.yaml

## What did you expect to happen
I expected 2 switches to turn on.  My understanding was that this is carried out using Zigbee messaging and not the actual MQTT component.

## How to reproduce it (minimal and precise)
If I send {"state":"ON"} to the groups set topic eg zigbee2mqtt/out_back/set then the value of the devices topic is updated to include "state":"ON" but the actual physical switch remains off.

## Debug info
Zigbee2MQTT version: 1.16.2 commit: 04c15f7
Adapter hardware:  CC26X2R1 (zzh)
Adapter firmware version:     zStack3x0 20201026

devices.yaml : 
```
'0x5c0272fffece4eb8':
  friendly_name: 'Back Door'
'0x5c0272fffece6d93':
  friendly_name: 'Patio Door'
```

groups.yaml
```
'1':
  friendly_name: out_back
  devices:
    - '0x5c0272fffece4eb8'
    - '0x5c0272fffece6d93'
```

configuration.yaml
```
homeassistant: false
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: XXXXXXXXXX
serial:
  port: /dev/ttyUSB0
advanced:
  pan_id: XXXXXXXXXX
  ext_pan_id: XXXXXXXXXX
  network_key: XXXXXXXXXX
  rtscts: false
  channel: 26
  cache_state: true
  cache_state_persistent: true
  last_seen: epoch
  log_level: debug
  report: true
devices: devices.yaml
groups: groups.yaml
frontend:
  port: 8880
experimental:
  new_api: true
```

I'm assuming this isn't really an issue more a case of me getting it wrong but some assistance would be great.

Koenkk, 2020-12-08:
Can you provide the data/database.db entries of these devices? (0x5c0272fffece4eb8/0x5c0272fffece6d93)

LucidityCrash, 2020-12-08:
```json
{"id":3,"type":"EndDevice","ieeeAddr":"0x5c0272fffece6d93","nwkAddr":2509,"manufId":4098,"manufName":"_TYZB01_qeqvmvti","powerSource":"Battery","modelId":"TS0011","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65506":19,"modelId":"TS0011","manufacturerName":"_TYZB01_qeqvmvti","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1607456562205}
{"id":4,"type":"EndDevice","ieeeAddr":"0x5c0272fffece4eb8","nwkAddr":44496,"manufId":4098,"manufName":"_TYZB01_qeqvmvti","powerSource":"Battery","modelId":"TS0011","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65506":19,"modelId":"TS0011","manufacturerName":"_TYZB01_qeqvmvti","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1607456562140}
```

and the group is 

```json
{"id":2,"type":"Group","groupID":1,"members":[{"deviceIeeeAddr":"0x5c0272fffece4eb8","endpointID":1},{"deviceIeeeAddr":"0x5c0272fffece6d93","endpointID":1}],"meta":{}}
```

One thing I'm noticing that is wrong with the switches as I paste that is that the devices are light switches connected to the mains yet the power source is listed as battery :)

Koenkk, 2020-12-09:
@LucidityCrash does controlling them individually work? It might be that the parent router of the devices does not hold the messages long enough for the switches.

LucidityCrash, 2020-12-09:
Yes sending {"state":"ON"}  or {"state":"ON"} to zigbee2mqtt/Back Door/set and zigbee2mqtt/Patio Door/set works as expected
I'm using a zzh usb stick

Koenkk, 2020-12-09:
Can you provide your networkmap?

LucidityCrash, 2020-12-09:
https://hastebin.com/ugayoducix.json

Just 2 end devices connected to the router :) 

Just to double check my expectations are correct with groups if I send a message to the group/set topic in mqtt then all devices in the group will get that message. And if I physically turn on one of the devices the other devices in that group should also turn on ?

Koenkk, 2020-12-10:
@LucidityCrash To dive deeper here we would need to sniff the traffic when sending the command. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

> Just to double check my expectations are correct with groups if I send a message to the group/set topic in mqtt then all devices in the group will get that message. 

Yes

>  And if I physically turn on one of the devices the other devices in that group should also turn on ?

No

LucidityCrash, 2020-12-11:
> > And if I physically turn on one of the devices the other devices in that group should also turn on ?
> 
> No

That would be Binding wouldn't it :)

LucidityCrash, 2020-12-11:
> @LucidityCrash To dive deeper here we would need to sniff the traffic when sending the command. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Just to double check ... the USB stick used for Sniffing would need to be seprate to the one I'm currently using as a Coordinator ?

Koenkk, 2020-12-11:
Yes a separate stick is needed for sniffing.

LucidityCrash, 2020-12-11:
OK ... I'll need to aquire one, I'll report back when I can.

LucidityCrash, 2021-01-11:
Just touching this to keep it open ... I have just got sniffer but not had a chance to install firmware and run tests

LucidityCrash, 2021-01-11:
OK pulled my finger out and done a capture, file pcapng file attached
[zigbee2mqtt_group.zip](https://github.com/Koenkk/zigbee2mqtt/files/5798598/zigbee2mqtt_group.zip)

I sent
```
{
  "state": "ON"
}
```
and

```
{
  "state": "OFF"
}
```
to the topic "zigbee2mqtt/out_back/set"

Koenkk, 2021-01-12:
Can you also provide your network key and data/database.db?

LucidityCrash, 2021-01-13:
[key_db.zip](https://github.com/Koenkk/zigbee2mqtt/files/5809117/key_db.zip)

Koenkk, 2021-01-13:
You can see that Zigbee2MQTT sends the commands correctly (packet 41 and 256). If controlling the devices directly works (`zigbee2mqtt/0x5c0272fffece6d93/set` payload `ON`) works I can only conclude that these devices don't support groups (or there is a bug in the firmware of them). Since these are TuYa devices it's unfortunately not uncommon to behave strange.

LucidityCrash, 2021-01-14:
Fair enough ... Binding doesn't appear to work on them either.

Koenkk, 2021-01-15:
Updated the docs. Assuming this can be closed now.

closed 2021-01-15 by Koenkk