Koenkk/zigbee2mqtt #5229: Separately identify HG06492C

soesa01, 2020-12-08:
**Information about the device + link**
There are 3 tunable LED (non-RGB) models sold by Lidl under the brand LivarnoLux currently:
 Bulb E27: Model is HG06492C (TS0502A) by _TZ3000_49qchf10 
 Candle E14: Model is HG06492B  (TS0502A) by _TZ3000_oborybow 
 Spot GU10:  Model is HG06492A (TS0502A) by _TZ3000_el5kt5im  

The GU10 model (HG06492A) is already in devices.js, added through: https://github.com/Koenkk/zigbee-herdsman-converters/commit/cef37de43d0dbdd340ee8a8db7afa3af1eccbe88

I only have the E27 model, so this request is for  HG06492C.
Link to manufacturer:
https://www.lidl-service.com/cps/rde/SID-EE373822-3A6B9064/lsp/hs.xsl/product.html?id=5019433851&rdeLocaleAttr=en&title=Smart%20LED%20Light%20Bulb

**data/database.db entry of the device**
{"id":2,"type":"Router","ieeeAddr":"0x588e81fffeb6ae93","nwkAddr":24033,"manufId":4098,"manufName":"_TZ3000_49qchf10","powerSource":"Mains (single phase)","modelId":"TS0502A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0502A","manufacturerName":"_TZ3000_49qchf10","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1606674522802}

Koenkk, 2020-12-09:
Done!

closed 2020-12-09 by Koenkk

soesa01, 2020-12-09:
Thanks for the quick response!