Koenkk/zigbee2mqtt #5231: Network Map is generating few hours or never

Mr-Groch, 2020-12-09:
## What happened
My network map is generating few hours (2-5 hours) or even sometimes it won't generate (at least when I'm waiting for it over 6 hours). This behaviour is always repeatable. My network contains 40 devices and I don't have any problems with responsivity or stability of my network. At least I haven't noticed any problems with setting or getting repsonse from any device when I wanted too.

This my network map looks like (default view):
[image]

And view with all route types on:
[image]

Here is full debug log - network scan is started at 2020-12-08 22:26:29 and ended at 2020-12-09 02:34:18:
http://groch.cloud/misc/z2m_debug_log_20201208.zip

## What did you expect to happen
Network map should generate maximum 2 minutes, as described in frontend.

## How to reproduce it (minimal and precise)
Jus start network map from frontend or from zigbee2mqtt Node-Red addon

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware:  CC26X2R1 zzh!
Adapter firmware version: 20201113

Koenkk, 2020-12-09:
Can you provide the herdsman debug log of this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Mr-Groch, 2020-12-09:
Yes I can. Lucky, this time, after enabling herdsman debug logging and zigbee2mqtt restart, generating network map took only about 13 minutes (I have observed that generating network map takes longer as z2m uptime is longer). But so quick network map is an exception rather than the rule (only after fresh z2m start). 

http://groch.cloud/misc/z2m_herdsman_debug_log_20201209.zip

Koenkk, 2020-12-10:
The delay is increasing because the queue is getting bigger and bigger. I see many read request towards 0x842e14fffe186b71 and 0x842e14fffe126280, what devices are this?

Mr-Groch, 2020-12-10:
Both are BlitzWolf BW-SHP13 plugs (https://www.zigbee2mqtt.io/devices/TS0121_plug.html). I don't have any noticable problems with them. Both are responding quick and power measurements are continuously reported.

Koenkk, 2020-12-11:
@Mr-Groch  already had a feeling it were these. Can you set the `measurement_poll_interval` for both to `60` in your `configuration.yaml`:

Example:
```yaml
devices:
  '0x12345678':
    measurement_poll_interval: 60
```

Mr-Groch, 2020-12-12:
I've changed `measurement_poll_interval` to 60 as you requested, and I can say that after 24 hours of z2m uptime, my network map is generation in less than 2 minutes! I have tried many times and always it will end in less than 2 minutes!

Are there any drabacks in measurement precision now? I'm using power meters in my statistics and automations...

Koenkk, 2020-12-13:
@Mr-Groch it will only update once every 60 seconds instead of 10 now.

Mr-Groch, 2020-12-13:
Next 24 hours uptime, and still network map generating in less than 2 minutes. Maybe I'll try to set` measurement_poll_interval: 30` for test if that is still acceptable interval for zigbee network health?

Koenkk, 2020-12-13:
@Mr-Groch sounsd good

Mr-Groch, 2020-12-15:
`measurement_poll_interval: 30` is also acceptable interval. My network map is generating 1-3 minutes after 2 days z2m uptime.

Koenkk, 2020-12-17:
Thanks, I've changed the default interval to 60 seconds now (to be on the safe side) but of course people can also reduce this with `measurement_poll_interval`.

I assume this can be closed now.

closed 2020-12-17 by Koenkk

septillion-git, 2021-01-03:
Glad I read this (in release notes of 1.17). I also noticed the negative impact of this plug. This plug is out for any power measuring in my setup then.

grasponcrypto, 2023-08-12:
> The delay is increasing because the queue is getting bigger and bigger. I see many read request towards 0x842e14fffe186b71 and 0x842e14fffe126280, what devices are this?

can i ask what you saw in the log (logfile no longer available) to indicate this behavior? I believe I am seeing a similar issue, but not sure what to look for.

titom43, 2023-11-21:
Hi all,

Same thing here... And same request that @grasponcrypto...

Thank you in advance.
Tom