Koenkk/zigbee2mqtt #5233: TuYa TS0601_thermostat - several properties read only

s4nf4n, 2020-12-09:
## What happened

I cannot set following properties via MQTT by using `zigbee2mqtt/[FRIENDLY_NAME]/set`:

- comfort_temperature
- eco_temperature
- boost_time

Might be more.

Additionally several properties/states are not exposed or exposed with "state"-name to Zigbee2Mqtt GUI and thus cannot be changed. 
Trying to change a state of "state" results in `No converter available for 'state' (OFF)` Toast.
See Screenshot:

[image]

## What did you expect to happen

- for values to be successfully changed on TuYa TS0601_thermostat
- GUI exposing properties correctly

For reference GUI from 1.16 from https://github.com/Koenkk/zigbee2mqtt/issues/4879

[image]

## How to reproduce it (minimal and precise)

e.g. publish `zigbee2mqtt/[FRIENDLY_NAME]/set` for TS0601_thermostat with payload `{"comfort_temperature":23}`

## Debug info
Zigbee2MQTT version: 1.16.2
zigbee-herdsman version: 0.13.37
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20201026
zigbee2mqtt log: https://hastebin.com/ukodukagib.apache

groenmarsmannetje, 2020-12-13:
I can confirm that temperatures can now be set in dev version. Boost time is still not possible to set.

s4nf4n, 2020-12-13:
Thank you @drzony! I can now confirm as well, that I can set `comfort_temperature` and `eco_temperature` with dev branch. `boost_time` still doesn't work.

The Zigbee2mqtt GUI is still missing those temps and the true/false states are not working from UI.

drzony, 2020-12-13:
@s4nf4n Devices do not accept floats, so z2m cannot do it.

drzony, 2020-12-13:
@s4nf4n I don;t have this TRV, so I cannot check it, there seem to be no errors in `boost_time` have you tried setting higher/lower (i.e. 100, 200, 250, 300, 350, 400) values? Maybe there is some limit on the device?

s4nf4n, 2020-12-13:
I have tried to publish i.e. `{"boost_time":240}` in steps of 60 mostly. State of `boost_time` doesn't get updated as when i set other values. I haven't tried to actually look at how long the device actually boosts in boost mode. I might be able to try tomorrow.

s4nf4n, 2020-12-14:
> I have tried to publish i.e. `{"boost_time":240}` in steps of 60 mostly. State of `boost_time` doesn't get updated as when i set other values. I haven't tried to actually look at how long the device actually boosts in boost mode. I might be able to try tomorrow.

Ok. Boost time really doesn't change.

drzony, 2020-12-14:
@s4nf4n Try sending low values like 1,2,3,4. As I said I do not have such TRV so it's hard for me to test this. If you feel comfortable with setting up a development version of zigbee-herdsman-converters, then I would be able to help further. If not, then I'm out of options, but maybe someone else can help here.

groenmarsmannetje, 2020-12-14:
I tried setting Boost time with latest dev build. And now it is changing.

When I publish zigbee2mqtt/kantoor_thermostaat/set/boost_time with payload 1, it is set to 0
When I publish zigbee2mqtt/kantoor_thermostaat/set/boost_time with payload 60, it is set to 0
When I publish zigbee2mqtt/kantoor_thermostaat/set/boost_time with payload 100, it is set to 100
When I publish zigbee2mqtt/kantoor_thermostaat/set/boost_time with payload 200, it is set to 200
When I publish zigbee2mqtt/kantoor_thermostaat/set/boost_time with payload 240, it is set to 200
When I publish zigbee2mqtt/kantoor_thermostaat/set/boost_time with payload 300, it is set to 300
When I publish zigbee2mqtt/kantoor_thermostaat/set/boost_time with payload 400, it is set to 400

So it is working now, but be aware that it rounds the values to 100.

s4nf4n, 2020-12-14:
Thank you @groenmarsmannetje . I will try 5o confirm tomorrow. 

@drzony can I actually debug herdsmen-converters?

s4nf4n, 2020-12-15:
> I tried setting Boost time with latest dev build. And now it is changing.
> 
> When I publish zigbee2mqtt/kantoor_thermostaat/set/boost_time with payload 1, it is set to 0
> When I publish zigbee2mqtt/kantoor_thermostaat/set/boost_time with payload 60, it is set to 0
> When I publish zigbee2mqtt/kantoor_thermostaat/set/boost_time with payload 100, it is set to 100
> When I publish zigbee2mqtt/kantoor_thermostaat/set/boost_time with payload 200, it is set to 200
> When I publish zigbee2mqtt/kantoor_thermostaat/set/boost_time with payload 240, it is set to 200
> When I publish zigbee2mqtt/kantoor_thermostaat/set/boost_time with payload 260, it is set to 200
> When I publish zigbee2mqtt/kantoor_thermostaat/set/boost_time with payload 300, it is set to 300
> When I publish zigbee2mqtt/kantoor_thermostaat/set/boost_time with payload 400, it is set to 400
> 
> So it is working now, but be aware that it rounds the values down to multiple of 100.

With a freshly pulled dev build I can set boost_time now as well. It behaves exactly like @groenmarsmannetje said.

drzony, 2020-12-15:
@s4nf4n Can you close this issue then? Exposes in web interface are still experimental, so those problems should either be in #4266 or a separate issue.

s4nf4n, 2020-12-15:
Ok. Thanks for helping.

`comfort_temperature`, `eco_temperature`, `boost_time` can be set successfully now.

closed 2020-12-15 by s4nf4n